areaDetector/ADCore #58: XML problem with HDF5 plugin hdf5_layout_*.xml files

prjemian, 2015-02-19:
in ADCore/iocs/simDetectorIOC/iocBoot/iocSimDetector/
the files hdf5_layout_demo.xml and hdf5_layout_nexus.xml are not well-formed XML files

The problem is the root element "xml" in these files violates the XML Naming Rule: "Element names cannot start with the letters xml (or XML, or Xml, etc)"

see, for example: 
http://www.w3schools.com/xml/xml_elements.asp
http://stackoverflow.com/questions/2519845/how-to-check-if-string-is-a-valid-xml-element-name

The consequence of this problem is that other software that handles these files may have problems handling / validating these files.  Such as a possible GUI to assist in editing and synchronizing the attributes and layout files.

MarkRivers, 2015-02-19:
Can that file be modified to make it valid XML?  If so, what else if anything needs to be changed?

Mark

http://www.w3schools.com/xml/xml_elements.asp
http://stackoverflow.com/questions/2519845/how-to-check-if-string-is-a-valid-xml-element-name

prjemian, 2015-02-19:
Change the root tag to something that I'd not "xml" in any mix of upper or
lower case.  One obvious choice is "hdf5".  The c++ might need to have this
change, to expect a different root tag.  Then any other layout files in
ADCore would need the change.  And the users would then need to change.

ulrikpedersen, 2015-02-20:
Oh, this is news to me. We've been running xmllint on the files to ensure valid xml and I've not noticed this as a problem before. But rules are rules and they're quite clear actually, so we'd better fix it sooner rather than later - I don't expect it to be a big change.

As I'm about to go on holiday :smile: I'll forward this on to @ajgdls (Alan Greer) who implemented this in the first place. He is at DLS again on Monday.

prjemian, 2015-02-20:
For the new root tag, a better suggestion is
<hdf5_layout>
</hdf5_layout>

which then marks this XML document with its intended use, matching the documentation terms.

prjemian, 2015-02-24:
fixed by #62

closed 2015-02-24 by prjemian