areaDetector/ADCore #63: NDFileHDF5: Requested hard links are not created when the dataset is from an EPICS_PV ndattribute.

MarkRivers, 2015-03-02:
The problem is that HDF5::createHardLinks() is being called too early in HDF5::openFile (indirectly via the call to HDF5::createFileLayout()).

MarkRivers, 2015-03-02:
Problem was originally fixed on hdf5-bugfix branch by moving the call to HDF5::createHardLinks() to the HDF5::closeFile() function.  However, this is not desirable because the file layout should be completely fixed at the end of HDF5::openFile().
Changed the fix to move the call to HDF5::createHardLinks() to the very end of HDF5::openFile().  This creates all of the hardlinks in the test1_attributes.xml, test1_layout.xml, test2_attributes.xml and test2_layout.xml files without any errors.

ulrikpedersen, 2015-03-03:
For reference the commits with the fix are: d3625c9 and 775e5a8

Thanks for changing that. SWMR is just around the corner! :smiley: 

Looks like we can close this one as well then. Agree?

closed 2015-03-03 by MarkRivers