areaDetector/ADCore #64: NDFileHDF5: Requested datasets whose values come from NDAttributes of type NDAttrString result in HDF5 attributes rather than HDF5 datasets

MarkRivers, 2015-03-02:
This was a design decision made in the original design of the HDF5 plugin.  However, it means that string attributes that change from one NDArray to the next cannot be stored in the file.  It also means that hardlinks cannot be created to datasets of this type, because they are not really datasets, they are attributes.  
The original restriction was because of the problem with string lengths varying from one NDArray to another.  In the netCDF plugin this is handled by storing string attributes as 2-D datasets, [MAX_ATTRIBUTE_STRING_SIZE, numArrays], i.e. using a fixed string size.  It is proposed to use a similar scheme in the HDF5 plugin.

ulrikpedersen, 2015-03-03:
Mark commented by email:

"I think it would be good to create a new parameter in asynNDArrayDriver which is NDAttrMaxStringSize.  It would replace MAX_ATTRIBUTE_STRING_SIZE and would allow the user to control the maximum allowed size of NDAttributeStrings in certain places, i.e. in the paramAttribute.cpp class, and the TIFF, netCDF, and HDF5 file plugins.  The default value of this parameter would be 256 for backwards compatibility."

ulrikpedersen, 2015-03-03:
It would not be difficult to create such a NDAttrMaxStringSize parameter, but I have a concern about how to deal with users changing this value (in particularly if it happens dynamically during acquisition). If this is allowed then we really have to support any-length strings (which we could probably do with STL strings anyway!)

For HDF5 datasets we would probably want it to be constant during acquisition - in fact we would only read it once at open-time in order to determine the max. string-length of string type NDAttributes. If it changes during acquisition that would be ignored by the file writer.

BTW, I'm listed as assigned, but really Alan Greer @ajgdls  will work on this - I just can't assign him because he is not part of this github organization...

MarkRivers, 2015-03-04:
I don't think it is a problem with it changing dynamically during acquisition.  As you said it would only be read during openFile(), not after that.  I'm happy postponing this and just using the current MAX_ATTRIBUTE_STRING_SIZE for now.

ulrikpedersen, 2015-03-05:
OK, I am most comfortable to stick with the MAX_ATTRIBUTE_STRING_SIZE for now as well.

We will at some point later have to work on multi-dimension NDAttributes - and variable length strings may fall naturally under that work.

ulrikpedersen, 2015-03-13:
Fix in hdf5-issues branch PR #85

commit 0224d176 (2015-03-11): Merge pull request #78 from areaDetector/issue-64

Added dataset support for String type NDAttributes.

Fixes #64
closed 2015-03-13 by ulrikpedersen