areaDetector/ADCore #66: NDFileHDF5: If the xml layout file does not define a group with “ndattr_default=true” it crashes the application.

MarkRivers, 2015-03-02:
If the NDFileHDF5 plugin is using a user-specific XML layout file, and if that file does not define a group with the ndattr_default=true attribute then it will crash the IOC application.

aglowacki, 2015-03-02:
What should the writer do if this field is missing. I believe I found the two spots in the code that we need to check for a NULL pointer. I can return an asynError but then these attributes are not saved to the hdf5 file. Should we put them in the root of the hdf5 if it doesn't exist?

prjemian, 2015-03-02:
Tells me that the layout author did not want them in the HDF5 file.

ulrikpedersen, 2015-03-03:
The trouble is when a user simply forgets to tag a group with ndattr_default=true and ends up missing a bunch of NDAttributes from the file... Do we want to support this unfortunate user (who don't know about or forgets setting ndattr_default=true) or the experienced user who deliberately leaves our ndattr_default=true because she does not want to store all NDAttributes?

I personally tend to agree with @prjemian on this one: if the user did not specify a ndattr_default group then it's probably because they're not required to be stored. It's a corner-case of course, because why would you define the NDAttributes in the first place if you don't wish to store them?

MarkRivers, 2015-03-04:
I would argue that this should always be created.  Users are much less happy with missing data in their files than with extra data.  If it is decided that the new datasets for UniqueID, timeStamp, etc. (see issue #68) should go in this group then I would strongly argue that is must always be created even if it was not present in the XML file.

prjemian, 2015-03-05:
One way to resolve this is to make writing of this group be optional 
with a default decision to always write.  An optional attribute on the 
root element of the layout XML file could be added to disable the 
automatic write if that is truly intended.  Here's an example:

<hdf5_layout auto_ndattr_default="false">
...
</hdf5_layout>

ulrikpedersen, 2015-03-05:
@prjemian yes that auto_ndattr_default=false would solve it: by default storing the NDAttributes in the root unless the user specifically disable this behaviour - or has tagged another group with the ndattr_default=true flag.

ulrikpedersen, 2015-03-10:
In response to @aglowacki's email:

> As I understand it we want the logic to be
> 
> 1) ndattr_default=true > // it saves the default to this group
> 2) if no ndattr_default, // save to root group
> 3) if new tag hdf5_layout auto_ndattr_default=“false // we don’t save default attr anywhere
> 
> Is this correct?

Yes, I think this is pretty much the correct logic.

prjemian, 2015-03-10:
Agreed

ulrikpedersen, 2015-03-13:
Fix in hdf5-issues branch PR #85

closed 2015-03-13 by ulrikpedersen

prjemian, 2015-03-19:
This is not fixed yet.
Just received this from the IOC console:
8257 Segmentation fault      ../../bin/linux-x86_64/simDetectorApp st.cmd.linux

OS: linux mint v17.1
base: 3.14.12.4
synApps: SVN trunk
ADCore: git master branch cloned ~6 hours ago
IOC: <synApps>/areaDetector/ADCore/iocs/simDetectorIOC/iocBoot/iocHDF5Test

Part of the file was written:
-rw-r--r-- 1 mintadmin  2691 Mar 18 22:23 test_005.h5

but the content of the file cannot be understood by the HDF5 support (HDFview reports a java error when trying to open the file):
mintadmin@mintadmin-vbox /tmp $ h5dump test_005.h5
h5dump error: unable to open file "test_005.h5"
mintadmin@mintadmin-vbox /tmp $ file test_005.h5
test_005.h5: data
mintadmin@mintadmin-vbox /tmp $ h5toText test_005.h5
Traceback (most recent call last):
  File "/home/mintadmin/Apps/anaconda/bin/h5toText", line 9, in <module>
    load_entry_point('spec2nexus==2015.214.0', 'console_scripts', 'h5toText')()
  File "/home/mintadmin/Apps/anaconda/lib/python2.7/site-packages/spec2nexus/h5toText.py", line 372, in main
    do_filelist(cmd_args.infile, limit, cmd_args.show_attributes)
  File "/home/mintadmin/Apps/anaconda/lib/python2.7/site-packages/spec2nexus/h5toText.py", line 333, in do_filelist
    print '\n'.join(mc.report(show_attributes) or '')
  File "/home/mintadmin/Apps/anaconda/lib/python2.7/site-packages/spec2nexus/h5toText.py", line 45, in report
    f = h5py.File(self.filename, 'r')
  File "/home/mintadmin/Apps/anaconda/lib/python2.7/site-packages/h5py/_hl/files.py", line 231, in __init__
    fid = make_fid(name, mode, userblock_size, fapl)
  File "/home/mintadmin/Apps/anaconda/lib/python2.7/site-packages/h5py/_hl/files.py", line 78, in make_fid
    fid = h5f.open(name, h5f.ACC_RDONLY, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.4.0/h5py/_objects.c:2405)
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.4.0/h5py/_objects.c:2362)
  File "h5py/h5f.pyx", line 72, in h5py.h5f.open (-------src-dir-------/h5py-2.4.0/h5py/h5f.c:1738)
IOError: Unable to open file (File signature not found)

attributes file: <IOC>/test2_attributes.xml
layout file: <IOC>/test2_layout_no_ndattr_default.xml

MarkRivers, 2015-03-19:
I'm confused.  You said you are running the master branch of ADCore, and you are in the iocHDF5Test directory.

However, your message said you were using the following XML layout file:

layout file: /test2_layout_no_ndattr_default.xml

But that file no longer exists in the master branch of ADCore.  Where did you get that file?

That file has been replaced with file test3_layout.xml, which is the same as test2_layout.xml but without ndattr_default="true".

I just tested using test3.sh, which uses test3_layout.xml, and it worked fine.  No segfault and it created the following 2 files:

-rw-rw-r-- 1 epics epics 140800 Mar 18 23:02 test3_001.h5
-rw-rw-r-- 1 epics epics 310525 Mar 18 23:02 test3_001.txt

The .txt file is the output of h5dump, which ran with no errors.

IOC: /areaDetector/ADCore/iocs/simDetectorIOC/iocBoot/iocHDF5Test

attributes file: /test2_attributes.xml
layout file: /test2_layout_no_ndattr_default.xml

�

prjemian, 2015-03-19:
did these steps today:
cd /usr/local/epics
wget https://subversion.xray.aps.anl.gov/synApps/support/trunk/checkout.csh
chmod +x ./checkout.csh
./checkout.csh
cd synApps/SVN
/bin/rm -rf areaDetector
git clone --recursive https://github.com/areaDetector/areaDetector.git

then configured and built synApps, including AD

prjemian, 2015-03-19:
mintadmin@mintadmin-vbox ~/.../simDetectorIOC/iocBoot/iocHDF5Test $ git branch -a
* (detached from 143797f)
  master
  remotes/origin/CreateDir
  remotes/origin/HEAD -> origin/master
  remotes/origin/ROI_fix
  remotes/origin/TempSuffix
  remotes/origin/circ_buff_trigger
  remotes/origin/db-includes
mintadmin@mintadmin-vbox ~/.../simDetectorIOC/iocBoot/iocHDF5Test $ ll *.xml
-rwxr-xr-x 1 mintadmin  5329 Mar 18 13:16 NexusTemplate.xml*
-rw-r--r-- 1 mintadmin  1447 Mar 18 13:16 test1_attributes.xml
-rw-r--r-- 1 mintadmin  2020 Mar 18 13:16 test1_layout.xml
-rwxr-xr-x 1 mintadmin 13054 Mar 18 13:16 test2_attributes.xml*
-rwxr-xr-x 1 mintadmin 12763 Mar 18 13:16 test2_layout_no_ndattr_default.xml*
-rwxr-xr-x 1 mintadmin 13059 Mar 18 13:16 test2_layout.xml*

  remotes/origin/diamond
  remotes/origin/dls_plugin_fixes
  remotes/origin/hdf5-bugfix
  remotes/origin/hdf5-issues
  remotes/origin/hdf5-update
  remotes/origin/issue-64
  remotes/origin/issue-65
  remotes/origin/issue-66
  remotes/origin/issue-68
  remotes/origin/issue-72
  remotes/origin/issue-74
  remotes/origin/issue-86
  remotes/origin/master
  remotes/origin/plugin_reset
  remotes/origin/roistat_update
  remotes/origin/unittest-framework

prjemian, 2015-03-19:
need to add this to the above installation procedure, after the git clone:

cd ADCore
git checkout master

then rebuild ADCore

done, and now this (using test3_attributes.xml and test3_layout.xml):

A call to 'assert(hdfdim >= 0)'
    by thread 'SimDetTask' failed in ../NDFileHDF5.cpp line 2494.
EPICS Release EPICS R3.14.12.4 $Date: Mon 2013-12-16 15:51:45 -0600$.
Local time is 2015-03-18 23:48:17.855503204 CDT
Please E-mail this message to the EPICS areaDetector collaboration (https://github.com/areaDetector/ADCore/issues) or to tech-talk@aps.anl.gov
Calling epicsThreadSuspendSelf()

prjemian, 2015-03-19:
also a minor point, since I'm using caQtDM:
NDFileBase.adl needs to be converted to NDFileBase.ui

ulrikpedersen, 2015-03-19:
OK this has got a bit tangled up by now :smiley: 

In summary: this issue #66 is solved and @prjemian just happened to build an old revision of ADCore (through the areaDetector top-level repo) which did not have the fix. The commands to get the very latest revision of ADCore are correct
    
    cd ADCore
    # optional; if you haven't cloned in a while do: git fetch origin 
    git checkout master

The new issue (assert(hdfdim >= 0) is related to the recent issue #88 (or in fact the fix). @ajgdls and I are looking into this. Please post comments related to this problem over on the issue #88 thread.

MarkRivers, 2015-03-19:
One does not always want to checkout master.  For example when R2-2 is released and master may not be stable then one should do:
cd ADCore
git checkout R2-2

Please E-mail this message to the EPICS areaDetector collaboration (https://github.com/areaDetector/ADCore/issues) or to tech-talk@aps.anl.gov<mailto:tech-talk@aps.anl.gov>

commit 3b712805 (2015-03-11): Save attribute datasets to the root location as a default (Resolve #66).

Attribute datasets are now saved to the root location if no default dataset
group is specified.  This can be overridden by setting a tag to false
(auto_ndattr_default) in the hdf5_layout root element.
Resolves issue #66.
commit 54298de8 (2015-03-11): Merge pull request #79 from areaDetector/issue-66

 Save attribute datasets to the root location as a default. Tested and works for me.

Fixes #66