areaDetector/ADCore #67: NDFileHDF5: Automatic HDF5 attributes for NDAttribute datasets are prone to name conflicts

MarkRivers, 2015-03-02:
The HDF5 plugin automatically adds two additional attributes for NDAttribute datasets, "description" and "source".

There are 2 problems with this.
- The names "description" and "source" are too generic, and are likely to conflict with attribute names that the user would like to add for compatibility with specific file conventions, etc.

- There are additional attributes that should also be present: the "name" of the NDAttribute, and the NDAttrSourceType of the NDAttribute (DRIVER, PARAM, EPICS_PV, etc.)

MarkRivers, 2015-03-02:
This problem has been fixed.  "description" and "source" have been renamed to NDAttrDescription and NDAttrSource.  Two additional automatic attributes have been added, NDAttrSourceType and NDAttrName.

ulrikpedersen, 2015-03-03:
@MarkRivers committed a fix: 2ef740f

I agree with this and "NDAttr" prefix is as good as any. We can probably close this ticket then?

Unrelated to Marks fix: we should probably use a [std::map](http://en.cppreference.com/w/cpp/container/map) instead of the two equal-length arrays... A rainy-day kind of issue...

closed 2015-03-03 by MarkRivers