areaDetector/ADCore #68: NDFileHDF5: Important default attributes are lacking

MarkRivers, 2015-03-02:
All NDArrays have certain important attributes, including UniqueID, timeStamp, epicsTS, and the dimension information for each dimension.  This critical information should be stored for all NDArrays in the HDF5 file even if it is not specifically requested in the XML layout file.

The following is an example of the output of ndump on a netCDF file that was written using no attributes.xml file for the driver, so there were no user-specific NDAttributes on the NDArray.

corvette:simDetectorIOC/iocBoot/iocHDF5Test>ncdump test_netcdf_002.nc
netcdf test_netcdf_002 {
dimensions:
        numArrays = 1 ;
        dim0 = 10 ;
        dim1 = 10 ;
        attrStringSize = 256 ;
variables:
        int uniqueId(numArrays) ;
        double timeStamp(numArrays) ;
        int epicsTSSec(numArrays) ;
        int epicsTSNsec(numArrays) ;
        byte array_data(numArrays, dim0, dim1) ;
        int Attr_ColorMode(numArrays) ;

// global attributes:
                :dataType = 1 ;
                :NDNetCDFFileVersion = 3. ;
                :numArrayDims = 2 ;
                :dimSize = 10, 10 ;
                :dimOffset = 0, 0 ;
                :dimBinning = 1, 1 ;
                :dimReverse = 0, 0 ;
                :Attr_ColorMode_DataType = "Int32" ;
                :Attr_ColorMode_Description = "Color mode" ;
                :Attr_ColorMode_Source = "Driver" ;
                :Attr_ColorMode_SourceType = "NDAttrSourceDriver" ;
data:

 uniqueId = 9 ;

 timeStamp = 794167395.448596 ;

 epicsTSSec = 794167395 ;

 epicsTSNsec = 449894139 ;

 array_data =
  8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
  9, 10, 11, 12, 13, 14, 15, 16, 17, 18,
  10, 11, 12, 13, 14, 15, 16, 17, 18, 19,
  11, 12, 13, 14, 15, 16, 17, 18, 19, 20,
  12, 13, 14, 15, 16, 17, 18, 19, 20, 21,
  13, 14, 15, 16, 17, 18, 19, 20, 21, 22,
  14, 15, 16, 17, 18, 19, 20, 21, 22, 23,
  15, 16, 17, 18, 19, 20, 21, 22, 23, 24,
  16, 17, 18, 19, 20, 21, 22, 23, 24, 25,
  17, 18, 19, 20, 21, 22, 23, 24, 25, 26 ;

 Attr_ColorMode = 0 ;
}

So for every NDArray it always stores 
        int uniqueId(numArrays) ;
        double timeStamp(numArrays) ;
        int epicsTSSec(numArrays) ;
        int epicsTSNsec(numArrays) ;
        byte array_data(numArrays, dim0, dim1) ;
        int Attr_ColorMode(numArrays) ;

It also defines global (file-wide) attributes for
                :dataType = 1 ;
                :NDNetCDFFileVersion = 3. ;
                :numArrayDims = 2 ;
                :dimSize = 10, 10 ;
                :dimOffset = 0, 0 ;
                :dimBinning = 1, 1 ;
                :dimReverse = 0, 0 ;
                :Attr_ColorMode_DataType = "Int32" ;
                :Attr_ColorMode_Description = "Color mode" ;
                :Attr_ColorMode_Source = "Driver" ;
                :Attr_ColorMode_SourceType = "NDAttrSourceDriver" ;

Note that it always stores the uniqueID, timestamps, color mode, of each array.  It also stores the full characteristics of every array dimension (e.g. offset, binning, reverse, size) as global attributes.

ulrikpedersen, 2015-03-03:
A few of the data details are implicitly stored in the HDF5 file as a description of the dataset. In particular: datatype and the length of each dimension. Reader applications always have to query these parameters before reading out a dataset anyway. I don't think we need to repeat those as additional attributes.

I agree that it is a good idea to include the NDArray descriptive data in the HDF5 file and we just need to agree on two things:
* List of exactly what parameters to store and how (dataset or attribute depending on the parameter)
* How to define where these parameters are to be stored - do we hardcode it or do we let the user decide?

## Proposed parameter list ##

From your list above I think I can distil the first proposal of parameters to store. Please review and comment:

Datasets, with an entry for each captured frame:
* uniqueId - int32 
* timeStamp - double
* epicsTSSec - int
* epicsTSNsec - int 
* ColorMode - int

Attributes, attached to the relevant detector dataset:
* numArrayDims
 * i.e. 2 for 2D images
 * Single integer value
* dimOffset
 * the offset of the frame if it was originally part of a larger image before passing through ROI plugin (or HW ROI).
 * 1D array with N (image dimension) scalar integers
* dimBinning
 * Number of pixels binned from original source (SW or HW)
 * 1D array with N (image dimension) scalar integers
* dimReverse
 * 0 = dimension not reversed, 1 = dimension has been reversed.
 * 1D array with N (image dimension) scalar integers

Notice that I'm missing these on my list, because I'm not entirely sure of their meaning...:

> :Attr_ColorMode_DataType = "Int32" ;
> :Attr_ColorMode_Description = "Color mode" ;
> :Attr_ColorMode_Source = "Driver" ;
> :Attr_ColorMode_SourceType = "NDAttrSourceDriver" ;

## Alternative plugin idea ##

An easy solution which would not break backwards compatibility would simply be to create a plugin which pull out all of these data elements from the NDArray and attach them as NDAttributes... That way the user can decide what and where to store this. 

The reason that a plugin may not be a good idea is that it forces an extra copy of the data (as we're not supposed to append to the attribute list without making a copy of the NDArray)...

MarkRivers, 2015-03-04:
I don't think that ColorMode needs any new treatment.  ColorMode is just a regular NDAttribute, and that is already handled by other mechanisms in the HDF5 plugin.  Other than that I agree with your proposed parameter list.
I don't think we should create a new plugin to convert the intrinsic array properties (uniqueID, timeStamp, etc. to attributes.  That is wasted overhead since the values are already part of the NDArray.  However, you already have a private copy of the NDAttributes from the NDArray plus the attributes defined for this plugin in this->pFileAttributes.  It is created in openFile.  So if it makes the implementation simpler you could add uniqueID, timeStamp,etc. to that list with pAttributeList->add().  That is entirely internal to this plugin.

MarkRivers, 2015-03-04:
One question is where these new datasets should be placed.  Should they be placed in the same group that contains the detector data, or should they be placed in the group defined with the ndattr_default tag.  If the latter is chosen then this impacts issue #66 because I would then argue that the ndattr_default tag group must be created so these datasets are always present.  Note also that the names of these datasets should have a prefix list NDAttr or NDDataset to avoid conflicts with user-defined datasets.

ulrikpedersen, 2015-03-05:
OK I think that it's best to let the user have a choice in where these dataset go. So if we use this->pAttributelist->add() to add these data-items (with a suitable prefix) then the user can chose where the datasets will go.

With @prjemian suggestion in [issue 68](https://github.com/areaDetector/ADCore/issues/66#issuecomment-77301005) to have an auto_ndattr_default=false flag on the root group we can ensure that if a user forgets (or don't know how) to define a default NDAttribute group, these datasets will fall back to be created in the root.

That way we have flexibility to define appropriate locations for the data - and a reasonable guarantee that we don't accidentally lose data.

MarkRivers, 2015-03-06:
I have completed the first part of this task, automatically adding NDAttributes for the following NDArray properties: 
uniqueId - int32
timeStamp - double
epicsTSSec - int
epicsTSNsec - int

These are then getting written to the HDF5 file for each NDArray:
            DATASET "NDArrayEpicsTSSec" {
               DATATYPE  H5T_STD_I32LE
               DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
               DATA {
               (0): 794509916, 794509916, 794509916, 794509916, 794509916,
               (5): 794509917, 794509917, 794509917, 794509917, 794509917
               }

            DATASET "NDArrayEpicsTSnSec" {
               DATATYPE  H5T_STD_I32LE
               DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
               DATA {
               (0): 539289909, 639422612, 739533018, 839676662, 939702920,
               (5): 39776392, 139886662, 239949022, 340142072, 440216791
               }

            DATASET "NDArrayTimeStamp" {
               DATATYPE  H5T_IEEE_F64LE
               DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
               DATA {
               (0): 7.9451e+08, 7.9451e+08, 7.9451e+08, 7.9451e+08,
               (4): 7.9451e+08, 7.9451e+08, 7.9451e+08, 7.9451e+08,
               (8): 7.9451e+08, 7.9451e+08

            DATASET "NDArrayUniqueId" {
               DATATYPE  H5T_STD_I32LE
               DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
               DATA {
               (0): 89, 90, 91, 92, 93, 94, 95, 96, 97, 98

I have not done the second part of this task, adding attributes for the detector dataset for numArrayDims, and dimOffset, dimBinning, and dimReverse for each dimension.  I'll let Ulrik tackle that, I am not sure how to do it.

ulrikpedersen, 2015-03-09:
Ah, I did the same on Friday. I didn't push because I hadn't completed the ticket.

@MarkRivers could you reference the relevant commit(s) SHA number in the ticket here, please? At a quick glance I didn't find the commit...

I'll continue on the remainder of the work today.

MarkRivers, 2015-03-09:
This is the commit where I did it:
commit de929796a2f06c1e0385d8ce8b45da3ee6148bbb
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Fri Mar 6 11:44:48 2015 -0600

    Create NDAttributes for NDArray properties uniqueId, timeStamp, and epicsTS

ulrikpedersen, 2015-03-13:
Fix in hdf5-issues branch PR #85

closed 2015-03-13 by ulrikpedersen

MarkRivers, 2015-03-15:
The default attributes (NDArrayUniqueId, NDArrayTimeStamp, etc.) were not being written to the file.  The problem was that NDFileHDF5::addDefaultAttributes() cannot just be called from writeFile, it must also be called from openFile(), because all datasets must be defined by the end of openFile().  This must not have been tested before this issue was closed.  This has now been fixed in commit 5f4c7e8a128e931380cb746a253d43c930145d6c to hdf-issues branch.

ulrikpedersen, 2015-03-16:
Damn! Sorry about that - I'm sure I tested at the time (it was the first fix I did in the series), but it doesn't work for me now so I must have slipped up. Hope there won't be any more of those!

MarkRivers, 2015-03-16:
No problem.  The good news is that I have done the following:

- New test3 which is the same as test2, but without the ndattr_default="true".

- New test4 which is the same as test2, but has auto_ndattr_default="false" in the root element.

- New test5 which is the same as test2, but test5.sh sets NumCapture=0.

I created test1.sh, etc. which are simple shell scripts to do the tests in a reproducible way.  I run these as follows:

- rm autosave/auto_settings.sav*
- Run the IOC, ../../bin/linux-x86_64/simDetectorApp st.cmd.linux
- In another window run one of the shell scripts, e.g.
- ./test1.sh
- This produces test1_001.txt, which is an h5dump of the HDF5 file.

I have run all 5 of the shell scripts and the resulting .txt files all appear to be correct.  I don't see any of the bugs that are described in https://github.com/areaDetector/ADCore/blob/master/RELEASE.md.

Mark

commit 7f682f38 (2015-03-11): HDF5 plugin: store default NDArray data as dataset attributes

Storing NDArray.dims.[offset,binning,reverse] as dataset attributes in file.

This fixes #68