areaDetector/ADCore #72: HDF5: Starting file saving in Stream mode crashes IOC if NumCapture=0

MarkRivers, 2015-03-05:
The HDF5 plugin crashes if FileWriteMode=Stream, NumCapture=0, and Capture is set to 1.

The behavior of Stream mode when NumCapture=0 is not documented.  The netCDF plugin and the single-file writing plugins (TIFF, JPEG, etc.) treat NumCapture=0 as meaning infinite, i.e. continue streaming forever until it is stopped by setting Capture=0.  Perhaps we should document this as the "correct" behavior?

ulrikpedersen, 2015-03-11:
I agree that setting NumCapture=0 is a sensible API for allowing infinite number of frames to be written. It is easy to prevent this mode of operation from crashing the ioc, but we must be a bit careful as NumCapture is used in a number of places throughout the HDF5 plugin. For example to calculate the size of the internal hdf5 chunk index (a binary search tree). 

Also @ajgdls latest fix to #74 uses NumCapture to find a sensible default chunk size for NDAttribute datasets. 

I think I have some ideas on how to get around this and chose sensible (i.e. well performing) default values without knowing the final dataset size...

ulrikpedersen, 2015-03-12:
OK, I have had a go at writing support for NumCapture=0 (i.e. infinite number of frames). The feature basically works - but I have not yet tested it with large acquisitions, nor have I tested for performance on large file systems yet...

ulrikpedersen, 2015-03-13:
Fix in hdf5-issues branch PR #85

closed 2015-03-13 by ulrikpedersen