areaDetector/ADCore #92: Tasks for R2-2

MarkRivers, 2015-03-20:
I think we are very close to releasing ADCore R2-2.  The tasks I know of are:
- Finish updating NDFileHDF5.html to reflect the current operation.
- Finish updating RELEASE.md for all changes in R2-2.
  - Add note about how all detector st.cmd files need to be changed, adding environment variable EPICS_DB_INCLUDE_PATH, and adding NDARRAY_PORT=$(PORT) to dbLoadRecords for NDStdArrays.template.

Any other items before release?

ulrikpedersen, 2015-03-20:
I think that about sums it up.
I will take a look at the NDFileHDF5.html on Monday (I'm unsure what state it's in) - but anyone else please feel free to pitch in!

Happy weekend! :smiley:

MarkRivers, 2015-03-23:
I've updated NDFileHDF5.html.  Please check.  The HDF5View png image definitely needs to be updated, but I don't have that Java application installed.  Perhaps it should use the simDetectorAttributes.html and the default HDF5 layout, i.e. no layout XML file?

ulrikpedersen, 2015-03-23:
Thanks for doing that Mark - it looks complete to me :+1: 

I have updated the HDFView screenshot using simDetectorAttributes.html and the default HDF5 layout as you suggested.

I also added a note about file readers not being able to open/read/monitor the file as it is being written (this sometimes surprises people).

MarkRivers, 2015-03-24:
R2-2 is released.

closed 2015-03-24 by MarkRivers

ulrikpedersen, 2015-03-24:
Wooohooooo!!!! I'm having a small celebration dance here  :smile: