areaDetector/ADCore #101: HDF5 files saved in single mode with default settings are very large

MarkRivers, 2015-04-27:


ulrikpedersen, 2015-04-27:
Hmm... in the past when files have been artificially blown up like that it has been related to inappropriate chunking or use of a large alignment parameter. Possibly when storing NDAttribute datasets. What is the file size if you switch off storing of NDAttributes?

I can also check this out on Wednesday and Thursday with Alan who is in fact working on this area (improving NDAttribute HDF5 datasets) for implementing SWMR support in the AD hdf5 plugin.

MarkRivers, 2015-04-28:
I have learned more about this issue.  This is the screen shot of the HDF5 plugin when it was performing very poorly and producing 23 MB files when they should be 1.2 MB.
[image]

I then restarted the IOC and the HDF5 plugin no longer had this problem.  It ran with 271 dropped frames (~ 7%) rather than 3665 (~ 92%).  The files were now 1.2 MB, not 23 MB.

This is the medm screen after restarting the IOC, when the problem was gone.  I cannot see any difference from the previous screen shot  that would cause this problem.

I suspect it is some interaction between the Nexus plugin and the HDF5 plugin.  I can now reproducibly make the Nexus plugin have very bad performance by running it after the HDF5 plugin is run.  If the Nexus plugin is run before the HDF5 plugin it drops 3% of the frames, while if it is run after the HDF5 plugin has been run it drops 81% of the frames.  I strongly suspect this is because the HDF5 plugin has set some value in the HDF5 library that results in poor performance for the Nexus plugin.  The Nexus files are not any larger than they should be, it is just a performance loss.

I also suspect that when the HDF5 plugin had bad performance and very bloated file size it may have been due to the Nexus plugin setting something in the HDF5 library.  However, I have not yet been able to reproduce this problem.

MarkRivers, 2015-04-28:
Github stopped letting me upload images for some reason, so the second screen shot did not load.  I have created a Google doc with the 2 screen shots here:
https://docs.google.com/document/d/1g-RWst_N7kPpWpA8ockBZNfddD1aVK4mrAfGeaMtGAw/edit?usp=sharing

The updated performance table is here.  I think the previous link I sent was not the shared link.
https://docs.google.com/spreadsheets/d/1ZapYxjNoCuVq_aWcRlDZYSZnR85FBQhaWxUIa9_Yvf4/edit?usp=sharing

ulrikpedersen, 2015-04-29:
That is a quite strange behaviour with the Nexus plugin. Everything in the HDF5 library is operating through 'handles' (hid_t) to specific objects in specific files - so it would surprise me if this is really the case.

However, I have spotted one difference in the screenshots on google docs: the "# Capture" is 0 in one and 100 in another... I know that we use "# Capture" to work out a few parameters for the HDF5 file when we create it - but that should only matter in "Multiple" (or  "Capture") mode. Perhaps something is spilling over into "Single" mode.

I will discuss with @ajgdls today as his work with SWMR touches on this aspect as well.

ajgdls, 2015-04-29:
I have investigated this issue and understand what is happening.  There is a chunking calculation made for attribute datasets that uses the number of capture frames to work out a suitable chunking value.  This calculation does two things that result in our problem here:
1) It does not check for the single mode parameter.
2) It assumes 0 capture frames means unlimited (which is potentially true but not in the single mode case).

The plugin then decides that for a potentially unlimited dataset we need some suitable chunking parameter (16384).  This results in large chunks for every attribute dataset even in the single frame case.

This is fixed by checking for single mode in the calculation and if the plugin is in this mode then set the chunking to 1.  I'll make the relevant changes.

ulrikpedersen, 2015-04-29:
Thanks Alan! That sounds about right to me. Please create an "issue-101" branch and commit your fix there - even better with a reference to this ticket (i.e. put #101 somewhere in the commit msg).

MarkRivers, 2015-04-29:
Make the new branch from the file_locks branch?

ajgdls, 2015-04-29:
Hi Mark,
I don't see that this is related to the file_locks branch, this problem looks present in all branches.  I re-created it on my own personal copy and also the ADCore master branch.  Do you still want the branch created from the file_locks branch?

MarkRivers, 2015-04-29:
I just meant to avoid merge conflicts.

ulrikpedersen, 2015-04-29:
I think issue #101 is really separate - it just happened to be discovered in working on the file locking. So I think it's best to create it off master (as Alan already did in fact). However, you can then really easily pull that into the 'file_locks' branch once the fix is up. That won't cause any problems when eventually merge into master as the history is the same (both branch off master at the same point).

MarkRivers, 2015-04-29:
OK, I was just worried that we would be making edits in the same part of the file and would have to manually resolve conflicts.

ulrikpedersen, 2015-04-29:
Oh, crap - you're right - there are edits in the same bit of code which git can't merge automagically... @ajgdls  could you have a go at merging issue-101 branch into file_locks, please? It's just a single conflict so it should be quick to resolve when you remember what that bit of code is for...

ajgdls, 2015-04-30:
OK I've merged the issue-101 branch into the file_locks branch, so I think all is OK.  Let me know if there are any problems.  Thanks.

ulrikpedersen, 2015-05-05:
Merged into master through PR #100

closed 2015-05-05 by ulrikpedersen