areaDetector/ADCore #99: Ensure file writing plugins take/release the lock appropriately in openFile() and closeFile()

ulrikpedersen, 2015-04-27:
In commit 67168620f70f6eb6514c08a00e4624c5a4aed37c @MarkRivers removed the locking before calling derived file writer plugins openFile() and closeFile(). The reason is that it can potentially block drivers from putting NDArrays on the input queue of file writer plugins.

The change is described on #98 

However, some file writer plugins (at least HDF5 - possibly more) expect the old behavior where the lock is taken prior to calling openFile() and closeFile() - and will be accessing the parameters and private class data the lock protects. This intention of this ticket is to ensure the file writing plugins get reviewed and modified to do a more fine-grained locking when necessary to avoid race-conditions.

MarkRivers, 2015-04-27:
Good point.  We should carefully review and define the expected behavior.  Before I made the modification the mutex was locked when calling openFile() and closeFile(), but unlocked when calling writeFile().  We should probably be consistent, so it is either always locked or unlocked for all calls to derived classes.  I'm fine with having it locked, but then all derived classes must unlock during slow operations.

ulrikpedersen, 2015-04-27:
I think the new change of having unlocked before calling derived plugins is a good and sensible default as it has the smallest impact on upstream drivers performance. IMO it is more obvious to plugin developers to remember to take care to **lock** at the right points rather than unlock...

:+1:  I'm happy with this new and more consistent behaviour - but of course we need to fix the issues with the HDF5 plugin (and others?) before next release.

MarkRivers, 2015-04-27:
I agree, let's have derived class methods called with asynPortDriver mutex unlocked.
I'll take a first pass at fixing the derived classes today.

MarkRivers, 2015-04-27:
I have compared file writing with the master branch and the new file_locks branch.  The results are here:
https://docs.google.com/spreadsheets/d/1ZapYxjNoCuVq_aWcRlDZYSZnR85FBQhaWxUIa9_Yvf4/edit#gid=0
The tests were run with simDetector generating 100 frames/s, 1024x1024 UInt8 data.  The file plugins were run in Single mode.  The test was running for 4000 images.

The conclusions are:
- All file plugins except JPEG were slowing down simDetector in R2-2
- No plugins slow down simDetector in the files_locks branch
- HDF5 plugin has very poor performance.  This is probably because the files should be ~1MB but are actually 23MB.

Here is a screen shot of my HDF5 plugin medm screen.  The arrays are 1024x1024 8-bit, so the files should be ~1 MB plus the attributes.  However, the files are actually 23 MB.  I believe these are all the default settings for the HDF5 plugin.   Can someone explain why the files are so large?

[image]

ulrikpedersen, 2015-04-27:
HDF5 will probably always perform a bit rubbish in single mode if it creates NDAttribute datasets - and right now of course it has a problem where it writes too large files in single mode. Try comparing with Multiple (=4000) mode...

ulrikpedersen, 2015-07-23:
Closing this issue. Main fixes are on PR #100. 
We have now been testing this fairly well through the last few months.

closed 2015-07-23 by ulrikpedersen