areaDetector/ADCore #102: HDF5 Single Writer Multiple Reader (SWMR) support

ulrikpedersen, 2015-04-29:
This is just a ticket to inform that Alan @ajgdls is working on adding support for the new [HDF5 SWMR feature](http://www.hdfgroup.org/HDF5/docNewFeatures/NewFeaturesSwmrDocs.html) in the HDF5 file plugin.

Work started on @ajgdls fork of ADCore, but has now been moved to the this repo on the "swmr_support" feature branch.

stuwilkins, 2015-11-03:
HI @ajgdls and @ulrikpedersen, How is progress on this branch? I am very, very keen to see this added to the HDF5 file plugin and can offer to help and test here at NSLS-II. The branch looks a little stale, what is the current status? Thanks!

ajgdls, 2015-11-03:
Hello @stuwilkins , we are currently rolling out an internal release ready for operational testing.  We have been running unit/system tests for a while.  Currently the branch is behind areaDetector/ADCore master and we plan to merge the latest master into the branch very soon.  Once this has been completed we can push the branch back.
@ulrikpedersen can correct me if I'm wrong but I think you can fork from the Diamond controls group github repository if you would like to try out the latest.  Just remember that until we merge master back in the branch will be behind the current ADCore release.

Thanks, Alan

ulrikpedersen, 2015-11-03:
Yup that's right. @stuwilkins you can get our latest code from our DLS fork: https://github.com/dls-controls/ADCore  - see the swmr_support branch there.

As a general rule we have switched to do development only on our own fork - so that all our "upstream" contributions are done through Pull Requests. That is why the latest SWMR code is only on our own fork for now. You're welcome to fetch and test it!

stuwilkins, 2015-11-03:
Great thanks @ajgdls and @ulrikpedersen, I use the same model myself, having a beamline org fork etc. I will fetch the swmr_support branch and test! Thanks.

ulrikpedersen, 2015-11-04:
> Which version of the HDF5 library do you use?

The SWMR feature has not yet been released by the HDF Group. The main development is complete but they are still merging features together and the HDF5 library 1.10 is not scheduled to be released until spring 2016. However, they have provided development snapshots for testing - and that is what we're using: ftp://ftp.hdfgroup.uiuc.edu/pub/outgoing/SWMR/src

**BIG WARNING:** The HDF Group recommends **not** to write SWMR files in a production environment until HDF5 1.10 has been released. The file format may in principle still be subject to change - and scientists would be very upset if their files become unreadable because they were written with a prototype-library

stuwilkins, 2015-11-04:
Thanks @ulrikpedersen . There PDF has some link to an SVN source which scared me and I also found not remember how to use SVN ....

Yes my users may not be too happy if we write unreadable data ... Thanks for the warning. At this point I really would just like to be 100% ready when SWMR goes live by the HDF group.

Do you know if the API has stabilized? I.e. presumably the HDF5Plugin code is now fairly stable - modulo any problems when the final library comes out.

ulrikpedersen, 2015-11-05:
> There PDF has some link to an SVN source

If you really want to try this out (i.e. be an early tester) then use the tarball on the ftp site instead of the svn source. The svn branch is a moving target - and indeed they are currently merging it into 'trunk' so it may not be up-to-date anymore. 

> I really would just like to be 100% ready when SWMR goes live by the HDF group.

Yup, so do we. Our goal is to have the SWMR support thoroughly tested and merged into ADCore/master **prior** to the scheduled release of the HDF5 1.10 (spring 2016). 

Also note that we have added the SWMR feature in a backwards compatible fashion so that you can still build and use the writer (without SWMR) if you continue to use an older version of HDF5.

> Do you know if the API has stabilized? 

Yes, has been for over a year now. There may still be additional functionality added (i.e. new functions) - but they're not likely to change the existing ones.

> presumably the HDF5Plugin code is now fairly stable - modulo any problems when the final library comes out.

Yes, that is fairly stable as well now. We're just testing it with a lot of combinations of usage now and making the occasional tweak/fix.

Another *WARNING*: we just noticed yesterday that we've done a wrong merge into our "swmr_support" branch on dls-controls - we accidentally merged a bunch of DLS-specific additions and configurations into the branch - but unrelated to the HDF5 plugin code. We will be fixing that today - most likely by creating a new branch and abandoning this one... 

I know python is popular at NSLS-II... If you're a python developer then you can also use [h5py with SWMR](http://docs.h5py.org/en/latest/swmr.html). I added that functionality a while ago and it was accepted through a Pull Request.

danielballan, 2015-11-05:
@ulrikpedersen Thanks for SWMR in h5py! I was happily surprised to discover it -- I'm glad to know who to thank.

ulrikpedersen, 2015-11-05:
You're welcome @danielballan :smile: Let me know how you get on with it!

MarkRivers, 2016-10-07:
Closing this issue.  SWMR support will be included in R2-5 to be released very soon.

closed 2016-10-07 by MarkRivers