areaDetector/ADCore #103: Proposal to add devIocStats to commonDriverMakefile and commonDriverSupport.dbd

MarkRivers, 2015-05-04:
I think it would be very useful to add devIocStats to the iocs built in areaDetector.
Advantages:
  - CA clients can see CPU load (SYS_CPU_LOAD, IOC_CPU_LOAD) and memory usage (MEM_FREE, MEM_USED), which can be significant for AD IOCs.
  - The UPTIME and HEARTBEAT PVs can be used to see if the IOC is running.  This is useful when there are many AD IOC in use at a beamline, etc. and a single OPI screen can show the status of all detector IOCs at a glance.
- The SysReset PV allows the IOC to be restarted via Channel Access.
- devIocStats runs on Linux, Windows, RTEMS, vxWorks, Darwin, etc.

Disadvantages:
- Another module that needs to be built if AD iocs are being built.

ulrikpedersen, 2015-05-05:
In the short/mid term, I don't mind so much as it is a fairly common (and stable) support module. We also use devIocStats here although we don't have a Windows version so I'll have to figure that out. Does a native Windows (MSVC, not cygwin/mingw) version of devIocStats exist?

However, in the longer term I would like if we could improve the distinction between "hard" dependencies required for ADCore to build (asyn, base, etc) - and "softer" dependencies which are only really required by the IOCs (autosave and devIocStats). As a site with a fairly rigid internal release mechanism we often end up in dependency hell when support modules require each other. In that respect areaDetector/ADCore is "just" another support module with a relatively complicated set of dependencies that we try to maintain in a sensible fashion.
 
Off the top of my head I don't have a specific proposal on how to solve this for areaDetector. I only mention it as background for why I'm reluctant to add dependencies to ADCore :smiley:

MarkRivers, 2015-05-05:
> Does a native Windows (MSVC, not cygwin/mingw) version of devIocStats exist?

Yes, the standard devIocStats now builds fine on MSVC archs.

> However, in the longer term I would like if we could improve the distinction between "hard"

> dependencies required for ADCore to build (asyn, base, etc)  and "softer" dependencies which are

> only really required by the IOCs (autosave and devIocStats).

I thought the dependency types were already pretty clear.  "Hard" dependencies are defined in areaDetector/configure/EXAMPLE_RELEASE_LIBS.local.  Those are only base and asyn.  They are required to compile the libraries.  "Soft" dependencies are defined in EXAMPLE_RELEASE_PRODS.local.  They are only required to run an example IOC.  The "soft" dependencies could be further divided into "firm" dependencies that define records that are used in any of the ADCore template files and those that do not.  Currently those "firm" dependencies are busy for the busy record and calc for the sseq record.

ulrikpedersen, 2015-05-05:
> "Hard" dependencies are defined in areaDetector/configure/EXAMPLE_RELEASE_LIBS.local.  Those are only base and asyn.  They are required to compile the libraries.  "Soft" dependencies are defined in EXAMPLE_RELEASE_PRODS.local.  They are only required to run an example IOC. 

Yes, you're right of course. I'd forgotten about these (because I don't use the top-level areaDetector module) and put everything in my RELEASE.local. The reason that I put everything in my RELEASE was simply that the dependencies were required in ADApp/commonDriverMakefile - but this one is no longer used anywhere internally in ADCore... I have just tried to build my clone with only asyn and base and that works fine. Thanks.

MarkRivers, 2015-05-05:
> The reason that I put everything in my RELEASE was simply that the dependencies were
> required in ADApp/commonDriverMakefile - but this one is no longer used anywhere internally in ADCore

commonDriverMakefile is used in ADCore/iocs/simDetectorIOC/simDetectorApp/src/Makefile.  It is only used for building IOCs.  commonLibraryMakefile is used for building driver libraries.

ulrikpedersen, 2015-05-05:
Yes, exactly. I had missed out on that little detail before so wrongly assumed I needed all dependencies to build ADCore...

MarkRivers, 2015-07-23:
devIocStats is now optionally included in IOC applications.

closed 2015-07-23 by MarkRivers