areaDetector/ADCore #112: Split IOC source and configuration from ADCore into a separate repository

ulrikpedersen, 2015-07-23:
As per the discussion started on PR #109: The idea is to split the example IOCs out of the ADCore repo, keeping ADCore as a fairly clean and lean library build with minimal external dependencies and options.

The IOCs and related configuration and example files should go into a new repository. @MarkRivers  suggests "ADSimDetector". It can include (optional?) dependencies for a whole range of useful external modules (devIocStats & friends).

ulrikpedersen, 2015-07-23:
So I'm looking at how to practically do this work...

HOW
===
I think the following steps should be sufficient without risking loss of important history or worse:

1. Create a copy of the ADCore repository for the IOCs. Lets call it "ADSimIOCs"
2. Remove all "ADCore" dirs and sources - deleting their history as well. Leaving only IOCs and their history.
3. Manually massage the "ADSimIOCs" dir with Makefiles, dbds and so on to remove ADCore/ADApp references and get it to build
4. Back in the original ADCore repo (on a branch or fork): git rm all IOC related dirs and files (but not deleting their past history!)
5. Issue Pull Request on ADCore to apply this removal of IOCs.
6. Everyone test and comment on PR
7. Merge PR. Done.

WHAT
====
What exactly do we move around then?

Directories
------------

Here is a list of directories in ADCore, tagged with my proposal for whether they should move out to an "IOCs" repo or stay in "ADCore"...

    ├── ADApp                  (both*)
    │   ├── ADSrc              (ADCore)
    │   ├── Db                 (ADCore)
    │   ├── netCDFSrc          (ADCore)
    │   ├── nexusSrc           (ADCore)
    │   ├── op                 (ADCore)
    │   ├── pluginSrc          (ADCore)
    │   ├── simDetectorSrc     (iocs)
    ├── configure
    ├── documentation          (ADCore)
    ├── iocBoot                (iocs)
    ├── iocs                   (iocs)
    │   ├── simDetectorIOC        -
    │   └── simDetectorNoIOC      -
    └── Viewers                (*)
        ├── IDL
        └── ImageJ

Special IOC related files
--------------------------

The directories are the easy bit to move around. However, it gets a bit more complicated when it comes to individual files. The following files reside inside "ADCore" dirs, but really belong to IOCs as part of build/configuration:

    ADApp/commonDriverMakefile            => ADSimIOCsApp/commonDriverMakefile
    ADApp/ADSrc/commonDriverSupport.dbd   => ADSimIOCsApp/ADSimIOCsSrc/commonDriverSupport.dbd
    documentation/simDetector*            => documentation/
        
Viewers
---------

Where do we put the viewers? I personally never build any of those - we have our own viewers as separate applications at Diamond. 

Viewers are typically not EPICS IOCs (and only one is a CA client), and often built using other languages and build-systems than areaDetector (IDL, Java, Python). For that reason I suggest that we also split these off ADCore and create new repos for these... One big one or one for each viewer?

NOTE
====

Please note that the above is a proposal and I am not about to embark on making any changes until we have reached some form of consensus. Also note that my proposed method of implementing these changes do not pose any risk of destroying the ADCore repo or it's history!

Please comment! :smile:

kmpeters, 2015-07-23:
simDetectorSrc directory should stay in ADCore.  If we separate the IOCs from the other driver repos, the drivers will not move with the IOC dirs.

I'm not sure if it makes more sense to make the ADCore/iocs directory root of the new ADSimDetector repo, or make each of the directories in the ADCore/iocs dir its own repo.  In either case, top-level Makefiles and RELEASE files will need to be modified to remove the assumption that the ADCore directory has a predicable relative location to the IOC directory.

The databases from optional modules and autosave configuration from iocBoot/commonPlugins.cmd be moved to a cmd file in the new IOC repo.  iocBoot/commonPlugins.cmd should remain in ADCore, but contain only support that comes from ADCore.  This will make it easier to keep commonPlugins.cmd up-to-date when plugins get added/removed/modified.

I don't know where the xml files should reside.

ulrikpedersen, 2015-07-23:
> simDetectorSrc directory should stay in ADCore.

I disagree (Mark also suggested that simDetector move out of ADCore). My intention would be to create an ADSimIOCs repo - and as the name suggest keep the simDetector together with a range of example IOCs.

> If we separate the IOCs from the other driver repos, the drivers will not move with the IOC dirs.

Well, my proposal does not extend to move all other example IOCs from other driver repositories out into the new IOCs repo... My concern is just to move extra IOC related stuff *out* of ADCore. I think I would still prefer to keep other drivers/iocs together as we do now.

> I'm not sure if it makes more sense to make the ADCore/iocs directory the root of the new ADSimDetector repo, or make each of the directories in the ADCore/iocs dir its own repo.

I vote for making a single new ADSimIOCs repo - with multiple IOCs of various configurations inside. We have to strike a balance between separation and maintenance of a large number of repos. Only if or when the new IOCs repo turn out to grow large and unmaintainable (as ADCore is about to), would I split it into separate repos.

> In either case, top-level Makefiles and RELEASE files will need to be modified to remove the assumption that the ADCore directory has a predicable relative location to the IOC directory.

I agree.

> The databases from optional modules and autosave configuration from iocBoot/commonPlugins.cmd should be moved to a cmd file in the new IOC repo. iocBoot/commonPlugins.cmd should remain in ADCore, but contain only support that comes from ADCore. This will make it easier to keep commonPlugins.cmd up-to-date when plugins get added/removed/modified.

OK, I think I agree with these as well.

> I don't know where the xml files should reside.

New ADSimIOCs repo: etc/ dir?

MarkRivers, 2015-07-23:
I also disagree with Kevin.  simDetectorSrc and ADCore/iocs should be moved to the new repository.  It then looks like any other detector, except it will contain some files that other detectors will use (commonDriverMakefile, commonPluginCommands.cmd).  The simDetector template and op/ files also move to the new directory.  Other detector repositories do not change, they keep their iocs/ directory.  if you don't want to build those then just set the BUILD_IOCS flag to NO.

kmpeters, 2015-07-23:
>  It then looks like any other detector, except it will contain some files that other detectors will use (commonDriverMakefile, commonPluginCommands.cmd).

If commonDriverMakefile moves to the new ADSimIOCs repo, then the ADSimIOCs repo will be a required module for most (all?) areaDetector drivers when the BUILD_IOCS flag is set to YES.

ulrikpedersen, 2015-07-23:
> If commonDriverMakefile moves to the new ADSimIOCs repo, then the ADSimIOCs repo will be a required module for most (all?) areaDetector drivers when the BUILD_IOCS flag is set to YES.

Yes. The point is that ADCore should not contain any references to external dependencies which are only required for IOC related business. That means IOCs will have to pull in this information from another repository - or craft their own Makefile (using commonDriverMakefile is just a helping utility but not a requirement).

We could add a new helper makefile in ADCore which only contains references to the libs that ADCore provide and depend on.

MarkRivers, 2015-08-21:
I am looking to do this relatively soon.  I am going through Ulriks list and I have some questions.
> Remove all "ADCore" dirs and sources - deleting their history as well. Leaving only IOCs and their history.

How do I do that?  What command do I use to delete their history?
> Back in the original ADCore repo (on a branch or fork): git rm all IOC related dirs and files (but not deleting their past history!)

Why don't I want to delete their past history if that history is already in the new repository created in the previous steps?

ulrikpedersen, 2015-08-21:
That's would be great. I got distracted from this one (again)...

> How do I do that? What command do I use to delete their history?

I use a script like this:

    #!/bin/bash
    set -o errexit
    # Permanently delete files and or folders from your git repository.
    # To use it, cd to your repository's root and then run the script with a list of paths
    # you want to delete, e.g., git-delete-history path1 path2
    
    if [ $# -eq 0 ]; then
        exit 0
    fi
     
    # make sure we're at the root of git repo
    if [ ! -d .git ]; then
        echo "Error: must run this script from the root of a git repository"
        exit 1
    fi
     
    # remove all paths passed as arguments from the history of the repo
    files=$@
    git filter-branch --prune-empty --index-filter "git rm -rf --cached --ignore-unmatch $files" HEAD
     
    # remove the temporary history git-filter-branch otherwise leaves behind for a long time
    rm -rf .git/refs/original/ && git reflog expire --expire=now --all && git gc --prune=now  &&  git gc --aggressive --prune=now

    # after that git push origin branch_name --force

> Why don't I want to delete their past history if that history is already in the new repository created in the previous steps?

You don't want to delete/modify the history of our common ADCore repository because:
 
 1. its not necessary - it's just honest history and the fact that we're *now* splitting parts off to another repo doesn't really change it.
 2. modifying the history of a central, shared repository screws up every local clone (and fork) out there. It can be fixed, but will definitely cause a LOT of headache for all developers as they would be required to exercise guru-level git-fu.

BTW; I'm off on vacation tomorrow for a week and a bit. I might not see/respond to messages in a timely fashion...

MarkRivers, 2015-08-21:
I tried to create an ADExample repository using the instructions on this Web page:
https://confluence.atlassian.com/bitbucket/split-a-repository-in-two-313464964.html

I created this simple 3-line script that would start the process by just removing the Viewers directory from the new ADExample repository:
```
git clone git@github.com:areaDetector/ADCore ADExample
cd ADExample
git remote rm origin
git filter-branch --index-filter 'git rm --cached -r Viewers' -- --all
```

However, that failed with the following error:
```
corvette:~/scratch/ADExample>git filter-branch --index-filter 'git rm --cached -r Viewers' -- --all
Rewrite ce6c760d8b66aca30de92e3aface339fe67ff0f0 (1/3522)fatal: pathspec 'Viewers' did not match any files
index filter failed: git rm --cached -r Viewers
```
Viewers is clearly a directory at the top-level in ADExample, so why did the git command fail?

MarkRivers, 2015-08-21:
This issue is now addressed by the following:
- New repository areaDetector/ADExample contains the simDetector source and IOCs.  It also contains commonDriverMakefile and commonPlugins.cmd.
- Pull request #122 removes the simDetector code and iocs directory from ADCore

MarkRivers, 2015-08-21:
The pull request is for a new noIOCs branch that can be used for testing.  Note that all detector repositories will need minor tweaks for the new locations of commonDriverMakefile and commonPlugins.cmd.

ulrikpedersen, 2015-09-08:
I have merged PR #122 with the "noIOCs" branch into master.

I'm still working my way through ADExample - it will require a bit of tweaking to build on Travis etc...

MarkRivers, 2015-09-08:
This has been done in pull request #122.

closed 2015-09-08 by MarkRivers

MarkRivers, 2015-09-15:
I have been getting strong objections to moving commonDriverMakefile, commonPlugins.cmd, and commonPlugin_settings.cmd from ADCore to ADExample.  It introduces an additional dependency (ADExample) for all detectors.  Having those files in ADCore does not create any dependencies of ADCore on synApps modules or optional plugins, though the files do reference them.  I was not sure what the right approach was when I originally moved them, and I am now thinking of restoring them back to ADCore from ADExample.  In addition to the additional dependency they create, the argument is that these are really needed to build all of the detector IOCs in areaDetector, so they belong in ADCore. 
Thoughts?

ulrikpedersen, 2015-09-15:
My main concern (and reason for arguing for the split) was to reduce the number of dependencies - and (related) reduce the size of matrix of possible build configurations. We have achieved a leaner configuration of ADCore mainly by splitting the IOCs off. The particular files ```commonDriverMakefile```, ```commonPlugins.cmd```, and ```commonPlugin_settings.cmd``` do not effect that as they are not used inside ADCore anymore.

So I think it is a reasonable practical compromise to simply host these files in ADCore - if it does not explicitly use them internally. Can we make a special folder for them where the build system does not enter?

mdmoo1978, 2015-09-15:
commonDriveMakefile, commonPlugins.cmd, and commonPlugin_settings.req aren't used in ADCore at all, they are used to build IOCs. With the move of the simDetector, there aren't any IOCs in ADCore that would use commonDriveMakefile. So I would think that having them (even in the same location) wouldn't have any impact on the build system matrix or the dependencies of ADCore.

I would also say that the ADCore/iocBoot directory move was done in error and now creates a dependance on ADExample for ADCore. The reason is the iocBoot directory has all of the schema information for the xml files. These are required by the plugins created in ADCore to validate the xml files you are using for data collection. They are part of the ADCore, because changes to hdf5 plugin could impact the hdf5_xml_layout_schema.xsd and both should be updated together. If they are not in the same module, you have created a dependency for ADCore on ADExample (even if it is not code dependency).

I believe that your objective was reached by just moving the SimDetector code and IOCs directory out of ADCore. Is anything else in ADExample dependent on any other EPICS module?

ulrikpedersen, 2015-09-15:
> I would also say that the ADCore/iocBoot directory move was done in error and now creates a dependance on ADExample for ADCore.

I disagree. This dir is intended for IOC Boot files - it's in the name... IOCs belong elsewhere as do the iocBoot dir. It is not the removal of the iocBoot dir which creates the dependency on ADExample - it is (some of) the files which lived in it.

> The reason is the iocBoot directory has all of the schema information for the xml files. These are required by the plugins created in ADCore to validate the xml files you are using for data [...]

Agreed - the XSD files naturally belongs in ADCore. I suggest that they could be put in some new dir in ADCore. As EPICS does not dictate a specific location for such files we can make one up, right? I always disliked the location in iocBoot as they're not used to boot IOCs.

mdmoo1978, 2015-09-15:
> I would also say that the ADCore/iocBoot directory move was done in error
> and now creates a dependance on ADExample for ADCore.
>
> I disagree. This dir is intended for IOC Boot files - it's in the [...]
I would contend it is the move of all the files in that directory that
creates the dependency, since all of the files are either needed for ADCore
or examples of how to use the plugins created by ADCore.

> The reason is the iocBoot directory has all of the schema information for
> the xml files. These are required by the plugins created in ADCore to
> validate the xml files you are using for data [...]
You can disagree with the name all you want, but the he fact is the
commonPlugins.cmd is a boot file and it belongs in ADCore. I'm not sure how
you get around the fact that it should be in a directory call iocBoot.

As for the XSD files, I don't know of any other cases using XSD files. So
unless someone else knows of an EPICS standard, you can put them where ever
you want in ADCore.

ulrikpedersen, 2015-09-15:
OK, so we could make a new dir with all the XSD and common makefiles - and then also make an iocBoot dir with commonPlugins.cmd in it. Just seems a bit silly with a dir to host a single file, but life is too short... :-)

MarkRivers, 2015-09-15:
Here’s my plan:

-          Restore commonPlugins.cmd and commonPlugin_settings.req in iocBoot

-          Move the XML schema files to a new top-level directory XML_schema

-          Restore commonDriverMakefile to ADApp.  The reasons are

o   This requires no changes to existing driver Makefiles

o   It is then in the same directory as commonLibraryMakefile.  commonDriverMakefile and commonDriverMakefile often need the same edits done to them as I did recently when I changed how HDF5_LIB and SZIP_LIB are handled.

mdmoo1978, 2015-09-15:
In the future the commonPlugins.cmd file could be split up multiple files to show the examples of how to load a plugin and commonPlugin can pull those in. It could be a new requirement of plugins to produce such a file, so that user understand how to load the new plugin.

MarkRivers, 2015-09-15:
I have done the following:
* Placed commonPlugins.cmd and commonPlugin_settings.req in ADCore/iocBoot.  commonPlugin_settings.req was previously in ADApp/Db.
* Changed commonPlugins.cmd so that ADCore/iocBoot is in the autosave requestfile_path.
* Moved the XML schema files to a new top-level directory ADCore/XML_schema
* Restored commonDriverMakefile to ADCore/ADApp.
* Removed all of these files from ADExample.
* Reverted some detector repositories that had already been converted to use ADExample.
* Committed everything to the master branch in each repository.

ulrikpedersen, 2015-09-16:
Looks good to me @MarkRivers  :+1:

MarkRivers, 2015-09-16:
I have made one more change.  The problem is that ADCore/iocBoot/commonPlugins.cmd and commonPlugin_settings.req are intended to be locally edited depending on what optional modules and plugins are to be used (sscan, alive, ADPluginEdge, etc.).  Those files should thus not be in git.  I moved those files to EXAMPLE_commonPlugins.cmd and EXAMPLE_commonPlugin_settings.req, and added a note in areaDetector/INSTALL that those file must be copied to commonPlugins.cmd and commonPlugin_settings.req.  I added commonPlugins.cmd and commonPlugin_settings.req to ADCore/iocBoot/.gitignore.