areaDetector/ADCore #115: Expand CI services to cover more configurations

ulrikpedersen, 2015-07-30:
Currently we have Travis CI building ADCore on every Push and Pull Request event. Travis is limited to do Linux builds. https://travis-ci.org/areaDetector/ADCore/builds 

In addition we also have Jenkins build for the top-level areaDetector module, pulling in ADCore and all drivers. https://jenkins.aps.anl.gov/view/AD2/ 

I would like to expand on the existing systems to perform a more complete (automatic) Continuous Integration which supports at least:

* building the extended platform matrix:
 * linux-x86 (i.e. 32 bit linux) static/dynamic
 * linux-x86_64 (64 bit linux) static/dynamic
 * win32-x86 (32 bit Windows) MSVC debug/release/static/dynamic
 * windows-x64 (64 bit Windows) MSVC debug/release/static/dynamic
* Trigger builds on *every* Push and Pull Request event.
* Feed the build status directly back to the repository for a convenient view for maintainers
 * tagging commits and commenting on PRs as travis-ci does.
 * For a low-level DIY CI service, see the API here: https://developer.github.com/v3/repos/statuses/

In addition to the above mentioned combinations of builds, we should also consider expanding the build matrix with the available optional configurations enabled/disabled. For example EPICS_LIBCOM_ONLY, BOOST, and EPICS_V4_PLUGIN. Naturally with this many combinations the build matrix is rather large. 

Are there any combinations in the build matrix which we can definitely discard as useless because we have no detectors/users which require it?

Creating this ticket after some discussion on PR #104

ericonr, 2025-05-08:
@ulrikpedersen do you think we still need 32bit coverage? Current CI has 64bit linux and Windows.

ulrikpedersen, 2025-05-12:
Hi @ericonr 
That's a good question... I am no longer working at Diamond which is where I was when raising this ticket back in 2017. Back then we had one or two Windows-based detector systems that only worked on 32bit. However, I imagine they would be phasing those out and in my opinion areaDetector could move on and drop 32 bit support now. I have no objection to closing this ticket.

Of course, I have no longer any skin in this particular game (I still work with EPICS and areaDetector occasionally but not at a large facility). If you would like an opinion from Diamond then perhaps ask someone like @GDYendell who might have more up-to-date information.

ericonr, 2025-05-29:
Awesome! Will close this then, thanks all!