areaDetector/ADCore #127: pluginTests/Makefile missing USR_INCLUDES line

MarkRivers, 2015-09-08:
Shouldn't this Makefile contain USR_INCLUDES += $(BOOST_INCLUDE)?

MarkRivers, 2015-09-08:
I added BOOST_INCLUDE to Makefile.  I also had to add the hdf5, sz, and xml2 libraries.  I don't understand how the plugin-test application could have built without these?  I also fixed an error in AsynPortClientContainer.cpp.  It was using the incorrect data type and was a fatal error on 32-bit Linux.

ulrikpedersen, 2015-09-09:
Hmmm... Strange. It works for us at DLS (RHEL6, linux-x86_64) and externally on Travis (Ubuntu 12.04 linux-x86_64) and I'm trying to figure out how:

> I also fixed an error in AsynPortClientContainer.cpp. It was using the incorrect data type and was a fatal error on 32-bit Linux.

Whoops, we don't have a (working) 32bit system any more so that must have slipped through.

> I added BOOST_INCLUDE to Makefile.

That has probably worked for us because we have tested with the system libraries. In other boost applications I have found that I don't need to specify the header include dir... The fix is correct though as boost could be installed anywhere. In fact I will change my local clone now to use a non-standard installation of boost.

> I also had to add the hdf5, sz, and xml2 libraries. I don't understand how the plugin-test application could have built without these?

This is because NDPlugin has been linked against these libraries and then you don't need to specify that when linking the application (at least on Linux, building shared lib). That works on both my RHEL6 machine and Travis Ubuntu.

Here is the linking on my RHEL6 machine (before your changes):

```
/usr/bin/g++ -o plugin-test  -L/home/up45/github/ADCore/lib/linux-x86_64
-L/dls_sw/prod/R3.14.12.3/support/asyn/4-26/lib/linux-x86_64 
-L/usr/lib64 -Wl,-rpath,/home/up45/github/ADCore/lib/linux-x86_64 
-Wl,-rpath,/dls_sw/prod/R3.14.12.3/support/asyn/4-26/lib/linux-x86_64 
-Wl,-rpath,/usr/lib64     --coverage   -m64               plugin-test.o 
test_NDPluginCircularBuff.o test_NDFileHDF5.o    -lboost_unit_test_framework 
-lADTestUtility -lNDPlugin -lADBase -lasyn   
```

and ldd reports that all the dependencies have been found, including xml2, hdf5 and sz and EPICS base libs:

```
[up45@pc0009 pluginTests]$ ldd O.linux-x86_64/plugin-test
	linux-vdso.so.1 =>  (0x00007ffde31f1000)
	libboost_unit_test_framework.so.5 => /usr/lib64/libboost_unit_test_framework.so.5 (0x00007ff5f610e000)
	libADTestUtility.so => /home/up45/github/ADCore/lib/linux-x86_64/libADTestUtility.so (0x00007ff5f5eb7000)
	libNDPlugin.so => /home/up45/github/ADCore/lib/linux-x86_64/libNDPlugin.so (0x00007ff5f5b7e000)
	libADBase.so => /home/up45/github/ADCore/lib/linux-x86_64/libADBase.so (0x00007ff5f5927000)
	libasyn.so => /dls_sw/prod/R3.14.12.3/support/asyn/4-26/lib/linux-x86_64/libasyn.so (0x00007ff5f56ab000)
	libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x0000003621800000)
	libm.so.6 => /lib64/libm.so.6 (0x000000361b800000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x0000003621400000)
	libc.so.6 => /lib64/libc.so.6 (0x000000361ac00000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x000000361b000000)
	libreadline.so.6 => /lib64/libreadline.so.6 (0x000000361c400000)
	libncurses.so.5 => /lib64/libncurses.so.5 (0x000000362ba00000)
	librt.so.1 => /lib64/librt.so.1 (0x000000361c000000)
	libdl.so.2 => /lib64/libdl.so.2 (0x000000361b400000)
	libnetCDF.so => /home/up45/github/ADCore/lib/linux-x86_64/libnetCDF.so (0x00007ff5f539e000)
	libNeXus.so => /home/up45/github/ADCore/lib/linux-x86_64/libNeXus.so (0x00007ff5f5189000)
	libCom.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libCom.so.3.14 (0x00007ff5f4f30000)
	libhdf5.so.9 => /dls_sw/prod/tools/RHEL6-x86_64/hdf5/1-8-14/prefix/lib/libhdf5.so.9 (0x00007ff5f4a4f000)
	libsz.so.2 => /dls_sw/prod/tools/RHEL6-x86_64/szip/2-1/prefix/lib/libsz.so.2 (0x00007ff5f483b000)
	libtiff.so.3 => /usr/lib64/libtiff.so.3 (0x000000362cc00000)
	libjpeg.so.62 => /usr/lib64/libjpeg.so.62 (0x000000362dc00000)
	libxml2.so.2 => /usr/lib64/libxml2.so.2 (0x0000003624c00000)
	libz.so.1 => /lib64/libz.so.1 (0x000000361bc00000)
	librecIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/librecIoc.so.3.14 (0x00007ff5f4611000)
	libsoftDevIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libsoftDevIoc.so.3.14 (0x00007ff5f440a000)
	libmiscIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libmiscIoc.so.3.14 (0x00007ff5f4205000)
	librsrvIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/librsrvIoc.so.3.14 (0x00007ff5f3ff4000)
	libdbtoolsIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libdbtoolsIoc.so.3.14 (0x00007ff5f3def000)
	libasIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libasIoc.so.3.14 (0x00007ff5f3be0000)
	libdbIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libdbIoc.so.3.14 (0x00007ff5f399e000)
	libregistryIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libregistryIoc.so.3.14 (0x00007ff5f379c000)
	libdbStaticIoc.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libdbStaticIoc.so.3.14 (0x00007ff5f357f000)
	libca.so.3.14 => /dls_sw/epics/R3.14.12.3/base/lib/linux-x86_64/libca.so.3.14 (0x00007ff5f331a000)
	/lib64/ld-linux-x86-64.so.2 (0x000000361a800000)
	libtinfo.so.5 => /lib64/libtinfo.so.5 (0x000000362ac00000)
```

Did this not work on your system? I don't mind adding these libs to the linking as you've done in your fix - but I would like to understand why they're required. As a basic principle I prefer the build configuration to include only what is required.

ulrikpedersen, 2015-09-09:
Triggered by this discussion, but not necessarily directly related, so mostly for information:

I have just done a few tweaks and fixes, mainly to get the travis build working after Marks changes.

* Download asyn from github rather than APS as the APS website is currently down. Marks latest commit actually failed to build due to this issue. I reckon it's better to download from github anyway because if github is down the build will not even start!
* pluginTests/Makefile: make linking of the "sz" library optional dependent on the user having configured the SZIP macro SHA: 44cee5e
* Various tweaks and fixes to improve the Travis build configuration. Mainly to bail out on any setup/configuration error so that the build does not try to continue - and then later fail with an obscure error...

MarkRivers, 2015-09-09:
I am building statically, so the libraries are required when building the product.

I found a serious deficiency in the EPICS build system (well actually with the Linux ld loader) which I discussed with Andrew Johnson yesterday.  The problem showed up in my case where I have an old version of hdf5 in the system libraries, and a newer version which is needed by areaDetector in a non-standard location.  My boost installation is in the system libraries.  I defined both HDF5_LIB and BOOST_LIB in CONFIG_SITE.local.  This leads to a serious problem: it happened to put the BOOST_LIB (/usr/lib64) first in the ld command (-L /usr/lib64), before HDF5_LIB (/usr/local64/lib) which then meant it found the old incorrect version of hdf5.  The only solution I could find was to change boost to be a PROD_SYS_LIB, rather then a PRO_LIB.  That way it finds boost in the system library location and hdf5 in the non-standard location.  Andrew thought this was a good solution.  In general there is no solution if there are 2 libraries A and B, where there is a location with a good version of A and a bad version of B, and a second location with a bad version of A and a good version of B.  The way ld works it will always find the bad version of one of them.

ulrikpedersen, 2015-09-09:
> I am building statically, so the libraries are required when building the product.

Ah, yes of course...

> This leads to a serious problem: it happened to put the BOOST_LIB (/usr/lib64) first in the ld command (-L /usr/lib64), before HDF5_LIB (/usr/local64/lib) which then meant it found the old incorrect version of hdf5. 

Argggg - this is a horrible issue. I think I may have come across it before and stumbled through a work around, similar to yours. 

> The only solution I could find was to change boost to be a PROD_SYS_LIB, rather then a PRO_LIB.

Normally at DLS we install most dependencies in a non-standard location (our "tools" area) as we like to have more up-to-date libraries than RHEL6 provides. Sometimes, we accidentally mix them up with the system libraries. For a while and for unknown reasons I've been using boost from the system, but hdf5 from a non-standard location. That works with your solution.

However, if I now use my newer non-standard install of boost, I run into the same problem, even with your tweak: the application appears to build OK, but ldd reports using boost from the system - so not what I intended. I can fix this by removing the _SYS part of ```plugin-test_SYS_LIBS += boost_unit_test_framework``` (I have not committed this change of course!).

I have two options: 
 1. I use the system version of boost - which is probably OK as we only use the boost test framework and nothing else.
 2. remove the _SYS part of ```plugin-test_SYS_LIBS += boost_unit_test_framework``` on our local installation at DLS.

For now, I will go with the first option as I have no real reason to use a non-standard boost right now. This might change in the future if we start using boost more.

Thus, the current state of the build (incl. unittest) works for me and for Travis. Are you happy with the current state, despite this work-around, @MarkRivers? If so, then I think we can close the ticket.

MarkRivers, 2015-09-09:
Yes, I'm happy with the current version.

closed 2015-09-09 by ulrikpedersen