areaDetector/ADCore #129: V4 plugin branch crashed when using attributes.

ajgdls, 2015-09-10:
I'm able to produce an exception within the V4 plugin.  The exception is generated by loading a set of Attributes into a detector and starting an acquisition with the V4 plugin enabled.  The error is repeatable, and I have reproduced it on Linux (64bit) and Windows (32bit).

Further investigation has revealed that the createAttributes method of ntndArrayConverter.cpp is generating the exception (throwing runtime_error) due to the check for getDataType() on the attribute.  This occurs when the attribute in question is an EPICS PV that does not exist.  This is a valid case where the attribute data type is NDAttrUndefined and so I don't think this method should be raising the exception.

Perhaps in the case of an undefined attribute the method should simply not add the attribute to its svector object, or else add it as an undefined attribute?  Does anyone have any comments regarding this?

ulrikpedersen, 2015-09-10:
Tagging @brunoseivam as he is the original author and may or may not get notifications on tickets directly on ADCore...

brunoseivam, 2015-09-11:
Sorry guys, I am on vacation, I will take a look at it on Monday

brunoseivam, 2015-09-14:
So, we could either not add the attribute to the list or reset the value pointer. I think the latter is more appropriate. Any thoughts?

ajgdls, 2015-09-15:
The second option keeps the attribute rather than losing it across the plugin/driver boundary, and so I think that is a better choice.

ulrikpedersen, 2015-09-15:
> So, we could either not add the attribute to the list or reset the value pointer.

What do you mean by "reset the value pointer"? - and how would that allow you to keep the attribute? Isn't the issue that the datatype is unknown - and for that reason you cannot serialize a value (as you don't know how many bits it takes up)?

I agree with Alan's comment that it is of course best not to lose the the attribute if at all possible... I just don't understand how :-)

brunoseivam, 2015-09-15:
Code-wise, it's just:

    value->get().reset();

The union that holds the value of the attribute would be empty, but the other fields (like attribute name, source, etc) would be kept. I don't know exactly how the serialization works, but it seems to just skip the value. It seems to work under my tests. 

On a side note, I just realized that on the client side (driver receiving an NTNDArray) the source type and source string would be lost when converting back to NDArray. I will fix that.

ajgdls, 2015-09-15:
Hi, you may need a fix made to NDAttribute recently as I noticed the soureType was not initialised correctly.  I found that the values passed between the V4 plugin and detector were nonsense.  Mark has applied the fix.  Reference #130

ulrikpedersen, 2015-09-15:
> The union that holds the value of the attribute would be empty, but the other fields (like attribute name, source, etc) would be kept. I don't know exactly how the serialization works, but it seems to just skip the value. It seems to work under my tests.

OK, clever things in V4 then. Thanks :+1:

ulrikpedersen, 2016-04-06:
@ajgdls and @brunoseivam: Can we close this ticket? I think this has been fixed...

ajgdls, 2016-04-06:
I certainly haven't noticed any problems in this area whilst using the V4 plugin and detector.  The plugin and detector have now been in use for several months.

closed 2016-04-06 by ulrikpedersen