areaDetector/ADCore #134: New version of ImageJ plugin that uses V4

MarkRivers, 2015-09-16:
Now that we have a plugin that exports NTNDArrays over PVAccess, it would be really nice to have a new version of the ImageJ plugin that directly receives NTNDArrays, rather than V3 waveform records.  The problem with waveform records is that they have a static maximum size at startup, and more importantly a static datatype.  This means that ImageJ cannot change between receiving 8-bit and 16-bit data when the detector changes its data type.  This should not be that hard since there is a native Java implementation of the V4 PVAccess client end, right?  Volunteers?

argonnexraydetector, 2016-11-02:
I am taking a look at this. Considering all the Java code from EPICS V4, this seems doable.

MarkRivers, 2017-01-28:
Tim Madden from the APS is working on this.

MarkRivers, 2017-04-01:
Closed via #262

closed 2017-04-01 by MarkRivers