areaDetector/ADCore #135: V4 plugin support preparation for merge with master

ulrikpedersen, 2015-09-17:
As the development of the V4 plugin/driver feature is now mostly complete (some debugging still ongoing) it is time to get it ready for merging into master. In preparation for merging to master we need to create all the example configurations, docs, etc. which makes the V4 plugin support usable for others. Should be done no the "v4-plugin" branch.

In summary [from @MarkRivers on #133](https://github.com/areaDetector/ADCore/pull/133#issuecomment-140726100):

* Merge master back into the v4-plugin branch, make sure it still works.
* Make a v4-plugin branch in areaDetector/areaDetector that adds what is need to .gitmodules, EXAMPLE_RELEASE_PATHS.local, EXAMPLE_CONFIG_SITE.local, Makefile, INSTALL.md, documentation/Doxyfile, etc.
* Make a v4-plugin branch in ADExample that allows testing the plugin and driver with simDetector.

brunoseivam, 2015-09-17:
Should pvaDriverSrc stay in ADCore/ADApp? Or move to ADExample? Or move into another repo?

ulrikpedersen, 2015-09-18:
> Should pvaDriverSrc stay in ADCore/ADApp? Or move to ADExample? Or move into another repo?

This is a tricky question... My opinion (which is probably about to be shot down in flames :-) ) is that the V4 plugin and driver inherently belongs together - in ADCore. I realise that @MarkRivers has just spent some time extracting simDetector into ADExample and there are no more drivers left in ADCore. However, the V4 plugin/driver works together in a fashion which is the first step to allow real distribution across network of an areaDetector plugin chain. That I think is a good argument for keeping the two together.

The example IOCs, demonstrating the V4 plugin/driver I would suggest belong in ADExample of course.

> I think the third option would be the cleanest

When we agreed to adopt your V4 plugin/driver we already discussed (and discarded) the idea of maintaining this in a separate repo and decided to adopt it directly in ADCore. I still think we should stick to that decision.

brunoseivam, 2015-09-18:
I agree with you, I should have elaborated:

I think keeping them together would be the best outcome, I suggested the other two in light of the recent move of simDetector out of ADCore, as you mentioned. If pvaDriver is to be moved out too, however, a new repo would be better than ADExample (hence the 'cleanest solution' :) ).

MarkRivers, 2015-09-21:
I’m OK with having them in ADCore provided:

-          It is easy to disable building them

-          When building is enabled it works without problems on all platforms

-          There is documentation like all other plugins and drivers

-          There are operator displays for all display managers to control them

-          They can be tested in the example simDetector IOC in ADExample

Mark

ulrikpedersen, 2015-09-24:
OK, so I think that is a green light to go ahead and keep the V4 plugin and driver in ADCore, @brunoseivam.

BTW, I've send you an invite to a team on the areaDetector github organisation - if you accept this then I can properly assign the ticket to you.

brunoseivam, 2015-09-28:
@ulrikpedersen I accepted your invitation.

@MarkRivers Regarding your points, I created OPI screens and @ajgdls had created MEDM ones. So currently missing are EDM and UI screens. I see that a lot of them are autoconverted. How is this done? Other than that, it is ready to be merged ([link](https://github.com/brunoseivam/ADCore/commits/v4-plugin)).

I also prepared ADExample ([link](https://github.com/brunoseivam/ADExample/commits/v4-driver-plugin)) with a pvaDriverIOC and added two commented out lines to simDetectorIOC's st.cmd that instantiate a NDPva plugin. 

In areaDetector ([link](https://github.com/brunoseivam/areaDetector/commits/v4-driver-plugin)), I added a conditional compilation var (`EPICS_V4`) to EXAMPLE_CONFIG_SITE.local and paths to the v4 modules to EXAMPLE_RELEASE_LIBS.local

ajgdls, 2015-09-29:
Hi @brunoseivam, just to let you know that I've created EDM screens, and not MEDM screens.

ulrikpedersen, 2015-09-29:
Thanks @brunoseivam. I have just made PR #138 to get this reviewed and merged into ADCore master. The PR is straight from brunoseivam/ADCore/v4-plugin - but the areaDetector/ADCore/v4-plugin branch has been synchronised as well.

MarkRivers, 2016-10-07:
I am closing this issue, NDPluginPva and pvaDriver are on master branch and tested.  They are now built only if the Makefile macro WITH_EPICS_V4=YES.

closed 2016-10-07 by MarkRivers