areaDetector/ADCore #144: Write to structure array in the V4-plugin is unsafe

dhickin, 2016-01-20:
<p>The V4 plugin (on the v4-plugin branch) seg faults whilst serialising the structure array of attributes.</p>

</p>The cause is an unsafe write to the structures pointed to by the elements of the structure array:</p>

<p>NTNDArrayConverter::fromAttributes() calls PVStructureArray::reuse(). When more than one reference is held to the shared_vector this will cause a copy to be made. The elements of both copies will point to the same PVStructures and so a write to the the latter is unsafe.</p>

<p>This can be fixed by checking unique() on the PVStructures and creating a new PVStructure if necessary. [Compare with the code for PVStructureArray::deserialize().]</p>

dhickin, 2016-01-20:
I've raised a pull request (<a href="https://github.com/brunoseivam/ADCore/issues/2">brunoseivam/ADCore/#2</a>)  to merge the fix into Bruno's v4-plugin branch. When the latter is merged into master through #138 this will fix the issue in areaDetector/ADCore.

ulrikpedersen, 2016-01-21:
Thanks @dhickin for looking into this and feeding back your fix. :+1:

brunoseivam, 2016-01-21:
Thanks @dhickin , I merged your PR

ulrikpedersen, 2016-04-08:
@dhickin: @brunoseivam has merged your fix. Can we close this issue?

commit 3cecfd96 (2016-04-19): Fix unsafe write to structure array

Fixes #144
closed 2016-07-17 by MarkRivers