areaDetector/ADCore #145: option to not write NDAttr* attributes in HDF5 dataset

prjemian, 2016-01-29:
Can we add an option to indicate that the default attributes (NDAttrName, NDAttrSource, NDAttrType) NOT be written to any specific dataset in an HDF5 file?

MarkRivers, 2016-01-29:
Can you explain the motivation?

prjemian, 2016-01-29:
Working with Francesco, he wants to have datasets that do not have any 
(attribute) content he did not describe in the layout file.

Particularly, he wants to apply this option to EPICS PVs that are 
defined in an attributes file.

The alternative is to not write those datasets from the HDF plugin at 
all.  Rather, later, for a different tool to acquire the EPICS PV values 
and then add that into the HDF5 file.

decarlof, 2016-01-29:
Pete summary is correct basically only for PV that are not generated by areadetector but defined
In the xml file it would be nice to have the option to turn off the default attributes. This will reduce duplication and conflicts in attribute definitions. Also it will make easy to validate and to read  the schema.

MarkRivers, 2016-01-31:
This seems like a reasonable request providing it is done in a backwards-compatible way and does not make the code significantly more complex.  Is this something Arthur could do, since he was one of the authors of the new HDF5 plugin?

ulrikpedersen, 2016-02-01:
Can we have a proposal (by example perhaps - or even better as a schema) for how should look in the HDF5 XML layout? 

/cc @ajgdls

prjemian, 2016-02-01:
will show this in branch [issue-145-NDattributes-optional](https://github.com/areaDetector/ADCore/tree/issue-145-NDattributes-optional) as backwards-compatible revision to XML Schema for layout files

prjemian, 2016-02-01:
example usage using [hdf5_layout_demo.xml](https://github.com/areaDetector/ADExample/blob/master/iocs/simDetectorIOC/iocBoot/iocSimDetector/hdf5_layout_demo.xml), line 15:

before:
```
 <dataset name="data2" source="detector">
```

after:
```
 <dataset name="data2" source="detector" write_ndattributes="false">
```

In this example, the default attributes would be written for all datasets EXCEPT "data2".

MarkRivers, 2016-02-01:
On which tags is the write_ndattributes attribute name to be supported?  Your schema shows it only for <dataset>.  But what about  <attribute> or <group>?  If I understand what Francesco wants to do it is not really to disable writing ndattributes, it is writing the default attributes of the ndattributes, i.e. the (NDAttrName, NDAttrSource, NDAttrType) .  It seems like the attribute value "write_ndattributes" is confusing.  You don't want to disable writing the ndattributes, you want to disable writing the default attributes of the ndattributes.

prjemian, 2016-02-01:
only on the dataset tags  (could not read part of your message regarding "what about" cases)

You are correct that it is only the default attributes which should not be written.  If the name of the proposed attribute were to be changed, would that be acceptable?  Perhaps this:
``write_default_attributes="false"``
when (NDAttrName, NDAttrSource, NDAttrType) should not be written?

MarkRivers, 2016-02-01:
Sorry, I should have previewed my previous message because it contained XML syntax that got clobbered.  It should have read:
 Your schema shows it only for "dataset". But what about "attribute" or "group"? One could argue that it should be possible to suppress these default_attributes individually for a specific ndattribute, or globally for a group?

ulrikpedersen, 2016-02-02:
> If the name of the proposed attribute were to be changed, would that be acceptable? Perhaps this:
> ``write_default_attributes="false"``

Definitely ``write_default_attributes="false"`` is clearer than ``write_ndattributes="false"`` :+1: 

> Your schema shows it only for "dataset". But what about "attribute" or "group"? One could argue that it should be possible to suppress these default_attributes individually for a specific ndattribute, or globally for a group?

For completeness this flag should be supported both dataset and group elements. Now for a group element you then have to decide whether:
 * it is a recursive flag which applies to all elements below the group? Or
 * does it apply only to datasets within this group? Or
 * does it apply only to attributes which can be attached to this group?

Finally, I can't help but think that is this really necessary for just not writing a few HDF5 attributes? They are really quite easily ignored...

Thinking about the big picture, I'm concerned about adding yet another niche-case; and adding extra code to an already quite big (bloated?) plugin.

aglowacki, 2016-02-03:
I am a little confused. Do you want to disable the writing of the default datasets ( ex  /defaults/timestamp ) or the writing of the attributes for datasets ( ex NDAttrDescription , NDAttrName, NDAttrSource, and NDAttrSourceType) ?

ulrikpedersen, 2016-02-03:
> I am a little confused. Do you want to disable the writing of the default datasets ( ex  /defaults/timestamp ) or the writing of the attributes for datasets ( ex NDAttrDescription , NDAttrName, NDAttrSource, and NDAttrSourceType) ?

I think it's the latter... But it is a bit confusing - that's why I'd like to see an example at the back of an envelope :-)

prjemian, 2016-02-03:
Hardly confusing, as written in the original statement.  Ulrik is correct,
it is the latter.

An example will be forthcoming...

decarlof, 2016-02-04:
Sorry for the delay getting back to you. 

Let’s say, for example, I want to have the hdf writer plugin save the monochromator energy value using the data exchange schema. Using the enclosed xml.zip (simDetectorAttribute.xml and the txm.xml), this value will be saved in the hdf5 file under measurement/instrument/monochromator/energy with attribute “units” keV.

I currently get the enclosed “hdf5_now”. What I propose is to allow for a write_default_attributes="false" that generates “hdf5_proposed”, i.e. with all the NDAttr of “energy” removed.

More generally, having this feature for all dataset derived from type=”EPICS_PV” will allow for custom hdf files (data exchange, neXus etc.) to be cleaner (NDAttrName = energy is redundant) and to match their schema definition (= without extra attributes that are not part of the schema definition). 

Attachments:
[xml.zip](https://github.com/areaDetector/ADCore/files/116640/xml.zip)

[image]

[image]

ulrikpedersen, 2016-02-04:
Thanks for the example @decarlof - turns out I did understand what you were on about - but it's much clearer now :+1: 

I'm sorry, but I'm still not convinced it is a good idea though:

> More generally, having this feature for all dataset derived from type=”EPICS_PV” will allow for custom hdf files (data exchange, neXus etc.) to be cleaner

To me, the perceived benefit of a "cleaner" hdf5 file is not worth adding extra code/complexity to actively throw away bits of data which records the origin. In fact I think it's almost non-scientific to actively remove data provenance. From a support point of view this data is also incredibly useful when users ask what a certain dataset is for and why it's there... 

> NDAttrName = energy is redundant

In the general case: no it is not redundant, because it is possible to define a different name for the dataset than the NDAttribute.

> match their schema definition (= without extra attributes that are not part of the schema definition).

The XML layout definition allow outlining the layout of the file - but it is not restrictive in the sense that it is possible to add additional bits of data (much like the NeXus standard allows). 

If you really don't want these bits of data in your resulting file then you could quite easily remove it with a simple post-processing step. I recommend python/h5py for this kind of thing :-)

ulrikpedersen, 2016-02-04:
Here is a bit of brute-force python which will remove **all** attributes that starts with NDAttr:

```
#!/bin/env python2.7
import sys
import h5py

fname = sys.argv[1]

def remove_ndattr_attr( name, obj ):
    for attr_name, attr_value in obj.attrs.iteritems():
        if attr_name.startswith(u'NDAttr'):
            del obj.attrs[attr_name]

f = h5py.File(fname, 'r+')
f.visititems( remove_ndattr_attr )
f.close()
```

MarkRivers, 2016-02-04:
I agree with Ulrik's comments.  Going further, 

> More generally, having this feature for all dataset derived from type=”EPICS_PV” will allow for custom hdf files (data exchange, neXus etc.) to be cleaner (NDAttrName = energy is redundant) and to match their schema definition (= without extra attributes that are not part of the schema definition).

I don't see why this change would be restricted to EPICS_PV attributes, logically it should also apply to PARAM and FUNCT attributes.  By removing the attribute type you would not know if that value came from an EPICS PV which might have some latency, or directly from a detector parameter, which is guaranteed to be current.  This is the type of source information that is bad to lose.

tacaswell, 2016-02-04:
At a minimum it should be possible to rename the attributes and set where they go at file creation time.

decarlof, 2016-02-04:
Ulrik at the moment I am already doing a post-processing step not to remove as you suggested but to add the data exchange layout/values/attributes. There I also keep track of the PV name that generated a value  the very same way the hdf plugin does it i.e. I store it as attribute, I just use a different naming convention for the attribute (no "NDAttr" prefix etc.)

Not all I save comes from PVs yet ;-), for example user/experiment information are pulled from data bases etc. but we are discussing into making also user/experiment information available as PV. 
When this happens I can eliminate the post processing step.

The reason for this is that I have too many different areadetector running and different way to control the scan. This  makes  the post processing step less reliable than having area detector doing the task.

So your and Mark conclusion seems that if I want the area detector plug to do this task I must accept its naming convention for attributes?

decarlof, 2016-02-04:
Yes this will work for me

Sent by T-mobile

MarkRivers, 2016-02-04:
Suppressing those default attributes in the code is not difficult.  This block of code occurs 2 times on NDFileHDF5::createAttributeDataset.  This code should just not be executed if a flag is set.
```
      // Write some description of the NDAttribute as a HDF attribute to the dataset
      for (i=0; attrNames[i] != NULL; i++)
      {
        size = strlen(attrStrings[i]);
        if (size <= 0) continue;
        this->writeStringAttribute(hdfAttrNode->hdfdataset, attrNames[i], attrStrings[i]);
      }
```
The trick is adding the code to the XML parser to store the flag that says you want to do the suppression.  If someone wants to add this feature and make a pull request I am willing to consider merging it.  That decision will depend on the amount of complexity it adds to the driver.

Note that  commit 2ef740f76b8187f7b62786db5493ef872873ee0d is where I added 2 of these default attributes and changed the namespace so it was unlikely to conflict with any user attribute names.  This was the commit message:
 
>  Changed the default attributes when writing datasets based in NDAttributes; previously they were 'description' and 'source'; now they are 'NDAttrDescription', 'NDAttrSource', 'NDAttrName', and 'NDAttrSourceType', i.e. added 2 new ones and added namespace to prevent name conflicts.

prjemian, 2016-02-04:
To proceed, we should agree on exactly what we want to do.  Which of the following options to be implemented?  (The changeset to the XML Schema posted previously, 51743eb, implements only option 1)

1. specify for any dataset not to include the default attributes ('NDAttrDescription', 'NDAttrSource', 'NDAttrName', and 'NDAttrSourceType'
1. specify specific default attributes that should not be written
1. specify at group level that no dataset in the group have the default attributes
1. restrict group-level instruction to only apply to datasets derived from EPICS PVs

The XML Schema can be modified for any of these (and possibly others).

ulrikpedersen, 2016-02-04:
> So your and Mark conclusion seems that if I want the area detector plug to do this task I must accept its naming convention for attributes?

@decarlof: That's my most favourite solution, but the debate appears to continue ;-)

> To proceed, we should agree on exactly what we want to do. Which of the following options to be implemented?

@prjemian: Agreed, if we **must**... But I think you're missing a 5th option, namely one (loosely) based on the suggestion from @tacaswell:

> At a minimum it should be possible to rename the attributes and set where they go at file creation time.

5) In the layout XML you can specify for a dataset (with an NDAttribute origin) the (hdf5) attributes names in which to record the original NDAttribute name, source, description, and sourcetype. If not specified the defaults will remain: 'NDAttrDescription', 'NDAttrSource', 'NDAttrName', and 'NDAttrSourceType'.

Option 5 would be my preferred solution: you can rename the NDAttr* attributes if you wish, but don't loose the information. By keeping the defaults we maintain backwards compatibility.

decarlof, 2016-02-05:
@ulrikpedersen do you mean I can change 'NDAttrDescription' into 'MyDescription' ?

ulrikpedersen, 2016-02-05:
@decarlof: yes

decarlof, 2016-02-05:
@ulrikpedersen that could be all I need then. Can you point me to some instruction on how to do this?

ulrikpedersen, 2016-02-05:
@decarlof: what I mean is that is what my proposed "option 5" would do for you. Needs implementing first... But it's good to have a :+1: from you on this option :smile:

decarlof, 2016-02-05:
@ulrikpedersen Great! I will talk with Arthur and see if he can implement a clean solution

prjemian, 2025-05-08:
A 9 years old issue, not seeing additional requests for this.  Ok to close for me.  Open new if request recurs.

closed 2025-05-29 by ericonr