areaDetector/ADCore #157: Race condition in NDPluginDriver constructor

ulrikpedersen, 2016-03-31:
Once in a while our unittests will crash with a segfault. 

A couple of examples: [build no. 229](https://travis-ci.org/areaDetector/ADCore/builds/119205876) and [build no. 218](https://travis-ci.org/areaDetector/ADCore/builds/118113158). Re-running the same travis build job will usually succeed.

The issue is not easily reproducible - at least I cannot reproduce it on RHEL6 (EPICS R3.14.12.3), but I have been able to reliably reproduce it on Ubuntu 14.04 with EPICS R3.14.12.5 and get the following stack-trace from GDB:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff2788b80 (LWP 63130)]
0x00000000006888c0 in ?? ()
(gdb) bt
#0  0x00000000006888c0 in ?? ()
#1  0x00007ffff759c070 in processTask (drvPvt=0x6868d0) at ../NDPluginDriver.cpp:158
#2  0x00007ffff5dfca6f in start_routine (arg=0x68af60) at ../../../src/libCom/osi/os/posix/osdThread.c:389
#3  0x00007ffff651b182 in start_thread (arg=0x7ffff2788b80) at pthread_create.c:312
#4  0x00007ffff682b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

(gdb) up
#1  0x00007ffff759c070 in processTask (drvPvt=0x6868d0) at ../NDPluginDriver.cpp:158
158	    pPvt->processTask();
(gdb) print pPvt
$1 = (NDPluginDriver *) 0x6868d0
(gdb) print pPvt->processTask
$2 = {void (NDPluginDriver * const)} 0x7ffff759c084 <NDPluginDriver::processTask()>
(gdb) print pPvt->msgQId
$3 = (epicsMessageQueueId) 0x0
```

So in [NDPluginDriver.cpp:158](https://github.com/areaDetector/ADCore/blob/master/ADApp/pluginSrc/NDPluginDriver.cpp#L158) it segfaults when dereferencing the plugin's `processTask()` function.

I expect this is because the thread is created in the [NDPluginDriver constructor](https://github.com/areaDetector/ADCore/blob/master/ADApp/pluginSrc/NDPluginDriver.cpp#L492). Occasionally the thread runs faster and starts before the construction of the final plugin object is complete: the virtual `processTask()` has not yet been initialised and can't be dereferenced (although GDB seems to understand that it's there :confused: ). 

Even if the thread had succeed in running `processTask()` it would soon fail as `NDPluginDriver::msgQId` has not yet been initialized (GDB reports its value as 0x0).

If I insert (as a test, not a solution) an `epicsThreadSleep(0.1)` just before `pPvt->processTask();` then the problem does not reproduce.

ulrikpedersen, 2016-03-31:
How to ensure the plugin object is fully constructed before starting up it's thread? 

We could create the thread with another epicsEvent to wait for - but the event need to be triggered somewhere after the plugin constructor has completed... This seems a bit cumbersome to me.

Or we could inherit the `epicsThreadRunable` class from EPICS base (I like this better). Again the thread has to be started from somewhere with `start()` after the object has been constructed. The documentation kind of address this construction/start problem (from AppDevGuide.pdf):

> The C++ interface is just a wrapper around the C interface. Two differences are the method start and the class epicsThreadRunable.
>
> The method start must be called only after the epicsThead object is constructed. It in turn calls the run method of the epicsThreadRunable object.

So where to call the `start()` method? In `iocInit()` using one of the available hooks? I think that would be quite reasonable that plugins start their threads up in iocInit, but it probably has some side-effects... :question:

ajgdls, 2016-03-31:
If you don't want to call start using an iocInit hook then you could use a timer (I think there is even an epicsTimer), although I would suggest using the hook is cleaner.

MarkRivers, 2016-03-31:
I will discuss this with Andrew Johnson today and see what he recommends.

MarkRivers, 2016-03-31:
> So where to call the start() method? In iocInit() using one of the available hooks? I think that would be quite reasonable that plugins start their threads up in iocInit, but it probably has some side-effects...

I don't like that, since we have specifically designed and documented areaDetector drivers and plugins to be usable outside the context of an IOC.  In fact don't the unit tests take advantage of that?

ulrikpedersen, 2016-03-31:
> In fact don't the unit tests take advantage of that?

yes it does.

MarkRivers, 2016-03-31:
I suspect this problem is not restricted to plugins.  Many drivers also create task threads in their constructors, and these are probably susceptible to the same race condition.

ulrikpedersen, 2016-03-31:
> I suspect this problem is not restricted to plugins. Many drivers also create task threads in their constructors, and these are probably susceptible to the same race condition.

Yes, I think you're right.

thomascobb, 2016-03-31:
This might be useful

http://stackoverflow.com/questions/16718663/c-11-start-thread-with-member-function-and-this-as-parameter#16719159

It looks like you can start a thread in the constructor as long as all the member functions you need access to are declared above the constructor...

ulrikpedersen, 2016-03-31:
> This might be useful
>
> http://stackoverflow.com/questions/16718663/c-11-start-thread-with-member-function-and-this-as-parameter#16719159
> 
> It looks like you can start a thread in the constructor as long as all the member functions you need access to are declared above the constructor...

The ordering of method declarations in a C++ class shouldn't matter at runtime in this case. Ordering of member declarations in a class just affect location in memory - and possibly padding - but I don't think that's relevant to our issue.

Also this stackoverflow message seems to relate to a (C++11) compilation error and not a race condition.

ulrikpedersen, 2016-03-31:
BTW I found a neat set of GDB commands to automatically re-run an application until it crashes :smile:. Makes it a lot easier debug an infrequent crash:

```
(gdb) set pagination off
(gdb) break exit
Function "exit" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 1 (exit) pending.
(gdb) commands
Type commands for breakpoint(s) 1, one per line.
End with a line saying just "end".
>run
>end

# This starts the loop going:
(gdb) r
```

Just start it running and go do something else for a while...

ulrikpedersen, 2016-03-31:
Ah Tom edited the stackoverflow link. The new one seems to be a more relevant topic - but the proposed solution doesn't work in this case.

MarkRivers, 2016-03-31:
I talked to Andrew Johnson and I believe I understand the problem.  It is because we are creating the thread in the base class, not the derived class.  The problem is that the base class object has been constructed but the derived class object has not.  So the problem typically won't occur for drivers, because they create the thread in the derived class, not the base class.

I believe I have a solution.  I will create a branch with a proposed fix for one plugin. 

Ulrik: What plugin were you testing with?

MarkRivers, 2016-03-31:
Interestingly I put a call to epicsThreadSleep(0.5) immediately after the call to epicsThreadCreate().  I expected that would cause a reproducible crash, because the thread would be running for 0.5 seconds before the constructor was done.  But it did not cause a crash.

MarkRivers, 2016-03-31:
I have created a new plugin_fix branch.  It contains a fix for the code in NDPluginDriver.h and NDPluginDriver.cpp.  NDPluginStdArrays.cpp uses the new createTaskThread() function to see if it fixes the problem.  Currently no other plugins will work when callbacksBlock=No, until they have the same fix as NDPluginStdArrays.

MarkRivers, 2016-03-31:
That should say other plugins won't work when callbacksBlock=1, not No.

MarkRivers, 2016-04-01:
I have deleted the plugin_fix branch.  I now believe that the problem is actually a simple, but serious error in NDPluginDriver.cpp.  The clue is in the original message on this issue:
`(gdb) print pPvt->msgQId`
`$3 = (epicsMessageQueueId) 0x0`
pPvt->msgQId should not be 0, because it was assigned in the constructor before the processTask thread was created.  I believe the problem is just that the NDPluginDriver class member data was not being protected with a mutex from access by the processTask thread.  I have now added the appropriate calls to lock and unlock() in the constructor and the processTask thread.  Branch plugin_fix2 and pull request #159 have this fix.  Ulrik needs to see if these fix the problem.

ulrikpedersen, 2016-04-01:
Phew - it's nice to come into work in the morning and see so many comments and suggestions for fixes (on this thread and various emails as well!) :smile: 

I'm already busy trying out the various suggestions (some of which overlap a little). It will take me a few hours to get through them all and then I'll report back on here... For the moment I can report on a few comments:

> Ulrik: What plugin were you testing with?

I was just running the full test suite and the crash happened in various plugins (never the same twice). The full test suite include the HDF5 file writer, Time Series, Circular Buffer plugins. If I limited the test to only run for one plugin it would just take a longer before the crash happened.

My testing environment: initially discovered the issue on travis (Ubuntu 12.04 on a VM). This was too slow for testing/debugging so I installed Ubuntu 14.04 on a VM on my laptop, running on 2 CPU cores.

Tests are run like this: `gdb --args ./bin/linux-x86_64/plugin-test --log_level=message`

ulrikpedersen, 2016-04-01:
> I have deleted the plugin_fix branch. I now believe that the problem is actually a simple, but serious error in NDPluginDriver.cpp. The clue is in the original message on this issue:
> (gdb) print [...]

OK, I've merged in the plugin_fix2 branch and also added the suggested `epicsThreadSleep(0.5)` to really force the issue:

> Interestingly I put a call to epicsThreadSleep(0.5) immediately after the call to epicsThreadCreate(). I expected that would cause a reproducible crash, because the thread would be running for 0.5 seconds before the constructor was done. But it did not cause a crash.

Now I get a crash right away (I'm running only the TimeSeries tests this time to be brief) - so the plugin_fix2 branch does not fix the issue. However, this time the crash is located inside the `NDPluginDriver::processTask()` method when using the `msgQId` (which is again 0x0). 

The original crash happened a level up when dereferencing `pPvt->processTask()` - so it never even got to the `msgQId == 0x0` issue...

```
ulrik@ubuntu:~/github/ADCore$ gdb --args ./bin/linux-x86_64/plugin-test --log_level=message --run_test=TimeSeries*
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/linux-x86_64/plugin-test...done.
(gdb) r
Starting program: /home/ulrik/github/ADCore/bin/linux-x86_64/plugin-test --log_level=message --run_test=TimeSeries\*
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff328d700 (LWP 7507)]
[Thread 0x7ffff328d700 (LWP 7507) exited]
Running 7 test cases...
port: simTS_0
port: TS_1
[New Thread 0x7ffff7fcb700 (LWP 7508)]
[New Thread 0x7ffff2a8c700 (LWP 7509)]
[New Thread 0x7ffff298b700 (LWP 7510)]
[New Thread 0x7ffff288a700 (LWP 7511)]
[New Thread 0x7ffff2788b80 (LWP 7512)]
Validating the input NDArrays - if any of these fail the test-cases are also likely to fail
port: simTS_2
port: TS_3
[New Thread 0x7ffff25a0700 (LWP 7513)]
[New Thread 0x7ffff249f700 (LWP 7514)]
[New Thread 0x7ffff239db80 (LWP 7515)]
Expecting stdout message "error, number of array dimensions..."
2016/04/01 10:45:26.773 NDPluginTimeSeries::processCallbacks: error, number of array dimensions must be 1 or 2
port: simTS_4
port: TS_5

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff239db80 (LWP 7515)]
epicsMessageQueueReceive (pmsg=0x0, message=0x7ffff239d268, size=8) at ../../../src/libCom/osi/os/default/osdMessageQueue.cpp:310
310	    return myReceive(pmsg, message, size, true, false, 0.0);
(gdb) 
(gdb) bt
#0  epicsMessageQueueReceive (pmsg=0x0, message=0x7ffff239d268, size=8) at ../../../src/libCom/osi/os/default/osdMessageQueue.cpp:310
#1  0x00007ffff759c153 in NDPluginDriver::processTask (this=0x69d570) at ../NDPluginDriver.cpp:179
#2  0x00007ffff759c096 in processTask (drvPvt=0x69d570) at ../NDPluginDriver.cpp:158
#3  0x00007ffff5dfca6f in start_routine (arg=0x69a0d0) at ../../../src/libCom/osi/os/posix/osdThread.c:389
#4  0x00007ffff651b182 in start_thread (arg=0x7ffff239db80) at pthread_create.c:312
#5  0x00007ffff682b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) up
[... 18 lines cut ...]
#3  0x00007ffff651b182 in start_thread (arg=0x7ffff17dcb80) at pthread_create.c:312
#4  0x00007ffff682b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) up
#1  0x00007ffff759c046 in processTask (drvPvt=0x6f98d0) at ../NDPluginDriver.cpp:158
158	    pPvt->processTask();
(gdb) print pPvt
$1 = (NDPluginDriver *) 0x6f98d0
```

Summary
----------

The plugin_fix2 branch and PR #159 does not fix the issue. 

There are 2 different types of crashes:

1. When putting a delay in the constructor after creating the thread we can (easily on Ubuntu) force a crash when using an uninitialized `this->msgQId`
2. The original crash: segfault when dereferencing `pPvt->processTask()` from the thread C function `void processTask(void *drvPvt)`. Note that we at this crash point have not yet entered the `NDPluginDriver::processTask()` method (so the msgQId issue has not yet had a chance to happen)

My conclusions
-----------------

I think @anjohnson's is probably right with his explanation of the order of construction of C++ objects at various levels of inheritance (I'm wondering about what order vtables are constructed:  `NDPluginDriver::processTask()` is declared virtual). I don't think we can lock/unlock our way around that by using member objects as they don't seem to have been initialised properly by the time we get there.

Which leads me back to my original proposal (also suggested by @anjohnson): use the epicsThreadRunable class and defer starting the thread until it is definitely safe: i.e. after the constructor have finished and the object is completely constructed. I guess there is a reason why the AppDevGuide specifically mention only calling the `start()` method outside of the constructor... (BTW, who wrote that part? Someone with knowledge of this?)

MarkRivers, 2016-04-01:
I am concerned that we might be looking at something else, even a compiler bug.  Yesterday I put a 0.5 second sleep after the call to epicsThreadCreate and it worked perfectly (this was on the original master branch).  I am instantiating all the standard plugins on a Linux machine with 16 cores, so if we have a multi-thread race condition why do you see it on a 2-core machine and I don't see it on a 16 core machine?  I now have a lock in the processTask which is only released when the base class constructor is exiting, and processTask only accesses class member from the base class, not the derived class. From what I have read on C++ constructors that should be legal.

MarkRivers, 2016-04-01:
I just checked out the plugin_fix2 branch and added epicsThreadSleep(0.5) right after the call to epicsThreadCreate().  It works fine.  This is gcc 4.6.3 on a 16-core system, quite fast.  So I don't think we can just be looking a subtle race condition, we are seeing some different in compilers?  Can you reproduce the problem on a different OS or OS version?

MarkRivers, 2016-04-01:
I really don't understand how this->msgQId is 0.  It has definitely been assigned in the NDPluginDriver constructor before the thread is created.  My understanding is that derived class instantiation involves adding additional information to the "this" structure, but not copying it to a different pointer or resetting the base class fields in the object.  Can someone please explain what could be going on?

MarkRivers, 2016-04-01:
Ulrik, here is a suggestion to try.  Remove the lock() and unlock() in the NDPluginDriver base class.  Add lock() to one of the derived plugin classes before calling the base class contructor, and then call unlock() at the end.  Leave the 0.5 second delay in the base class constructor.  Does it crash?

ulrikpedersen, 2016-04-01:
> I am concerned that we might be looking at something else, even a compiler bug. Yesterday I put a 0.5 second sleep after the call to epicsThreadCreate and it worked perfectly (this was on the [...]

I agree that it is highly suspicious that it only seems to be reproducible for me (and travis)... It could be an issue related to running on a Virtual Machine(?). However, the fact that it crashes like that *anywhere* is also worrying.

My available systems are:

Computer                             | OS                 | GCC                                         | Crash
-------------------------------------|--------------------|---------------------------------------------|--------------------------------
Travis VM                            | Ubuntu 12.04.5 LTS | gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3    | Happens regularly on builds triggered by git push - no GDB trace easily available
Desktop 8 CPU cores                  | RHEL 6.7           | gcc (GCC) 4.4.7 20120313 (Red Hat 4.4.7-16) | Cannot reproduce
VMware Player on laptop. 2 CPU cores | Ubuntu 14.04.3 LTS | gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4     | Easily reproducible with/without gdb

> Ulrik, here is a suggestion to try. Remove the lock() and unlock() in the NDPluginDriver base class. Add lock() to one of the derived plugin classes before calling the base class contructor, and then call unlock() at the end. Leave the 0.5 second delay in the base class constructor. Does it crash?

I'll try that now.

ulrikpedersen, 2016-04-01:
> Ulrik, here is a suggestion to try. Remove the lock() and unlock() in the NDPluginDriver base class. Add lock() to one of the derived plugin classes before calling the base class contructor, and then call unlock() at the end. Leave the 0.5 second delay in the base class constructor. Does it crash?

Actually I don't think I can do that. The NDPluginDriver base class constructor has to be called with arguments which has to be done in the derived class initialization list - but you can't call lock/unlock in that list. I've tried moving the call to the base class constructor from the init list and down in the body of the constructor (between lock/unlock) - but it doesn't build.

ref: http://stackoverflow.com/questions/120876/c-superclass-constructor-calling-rules

I'm afraid I've run out of time today - I'm in meetings for the remainder of the day now :grimacing: 

I'd recommend trying out VMware Player (it's free) and install a virtual Ubuntu machine - it's really quite useful to have a VM around for testing a different OS...

anjohnson, 2016-04-01:
The call to lock() just before the thread creation does not implement a write memory barrier, it's only a read barrier. Can I suggest calling lock() at the beginning of the constructor, unlock() immediately before epicsThreadCreate() and lock() immediately afterwards, and leave the unlock() at the end of the constructor. The new lock() and unlock() calls inside NDPluginDriver::processTask() should also remain.

MarkRivers, 2016-04-01:
I implemented Andrew's suggestions in branch plugin_fix3.  They did not appear to fix the problems, the Travis build still crashed with an access violation.

I then made a new branch plugin_fix4.  In NDPluginDriver::processTask I loop at the beginning waiting for msgQId to become non-zero.  This did not help, it still crashed.  I then added debugging printf statements in processTask and NDPluginDriver::processTask. processTask (C function) is being called, but there is no printout in the member function NDPluginDriver::processTask, so it looks like its address is invalid?  There is no way to print the address of a member function in C++, so I can't print the address of NDPluginDriver::processTask in processTask.

MarkRivers, 2016-04-02:
I created branch plugin_fix5.  I removed the loop waiting for msgQId to become non-zero, since it is too late in NDPluginDriver::processTask.  The crash happens because NDPluginDriver::processTask is not a valid pointer at that time.  Instead I added an epicsThreadSleep(0.1) in processTask (C function) before calling NDPluginDriver::processTask.  This fixes the problem on Travis.  Interestingly Travis is using gcc 4.6.3-1 while on my Fedora system I am using 4.6.3-2, very similar.  Mine does not crash when Travis does.  Why?

Here are some potential fixes:
- Use the sleep in processTask.  This is a hack, but it will probably work and requires no changes to other plugin code.
- Make msgQId be public and poll it in processTask to see when it becomes non-zero.  That is probably a good indication that the constructors are done.  This is also a hack, but not so subject to timing issues.
- Create the thread in a new run() method, and make all plugins call that after instantiating the object.  Cleanest solution, but the most work.

MarkRivers, 2016-04-02:
Tested polling msgQId.  Does not work, I still get a crash in Travis build.  I really think we need the clean solution of only creating the thread when we are sure the derived class constructors are complete.  Not that hard to do, I will do it now on branch plugin_fix_clean.

MarkRivers, 2016-04-02:
I have created a new public NDPluginDriver::run() method.  I have changed all derived class in ADCore to call run() after the NDPluginXXX object is instantiated.  This passes the Travis CI build, and seems to work fine with ADExample/simDetector.  This is in the plugin_fix_clean branch.

ulrikpedersen, 2016-04-04:
> I have created a new public NDPluginDriver::run() method. I have changed all derived class in ADCore to call run() after the NDPluginXXX object is instantiated. This passes the Travis CI build, and seems to work fine with ADExample/simDetector. This is in the plugin_fix_clean branch.

Thanks Mark - this is what I believe is the proper fix. 

And it sort of works in the unittest case - if I don't start the thread with the call to `run()`. However, if I do call `run() (see patch below) it still crashes in the same way. I can't believe it!!!

```
diff --git a/ADApp/pluginTests/test_NDFileHDF5.cpp b/ADApp/pluginTests/test_NDFileHDF5.cpp
index 6fb6c2c..5fe7f36 100644
--- a/ADApp/pluginTests/test_NDFileHDF5.cpp
+++ b/ADApp/pluginTests/test_NDFileHDF5.cpp
@@ -54,6 +54,7 @@ struct NDFileHDF5TestFixture
 
 
     // Enable the plugin
+    hdf5->run(); // start the plugin thread although not required for this unittesting
     hdf5->write(NDPluginDriverEnableCallbacksString, 1);
     hdf5->write(NDPluginDriverBlockingCallbacksString, 1);
 
diff --git a/ADApp/pluginTests/test_NDPluginCircularBuff.cpp b/ADApp/pluginTests/test_NDPluginCircularBuff.cpp
index f468d1e..39e18b0 100644
--- a/ADApp/pluginTests/test_NDPluginCircularBuff.cpp
+++ b/ADApp/pluginTests/test_NDPluginCircularBuff.cpp
@@ -53,6 +53,7 @@ struct PluginFixture
 
         // This is the plugin under test
         cb = new NDPluginCircularBuff(testport.c_str(), 50, 0, dummy_port.c_str(), 0, 1000, -1, 0, 2000000);
+        cb->run(); // start the plugin thread although not required for this unittesting
 
         // This is the mock downstream plugin
         ds = new TestingPlugin(testport.c_str(), 0);
diff --git a/ADApp/pluginTests/test_NDPluginTimeSeries.cpp b/ADApp/pluginTests/test_NDPluginTimeSeries.cpp
index 1fe9164..e0096f8 100644
--- a/ADApp/pluginTests/test_NDPluginTimeSeries.cpp
+++ b/ADApp/pluginTests/test_NDPluginTimeSeries.cpp
@@ -87,6 +87,7 @@ struct TimeSeriesPluginTestFixture
     downstream_plugin = new TestingPlugin(testport.c_str(), 0);
 
     // Enable the plugin
+    ts->run(); // start the plugin thread although not required for this unittesting
     ts->write(NDPluginDriverEnableCallbacksString, 1);
     ts->write(NDPluginDriverBlockingCallbacksString, 1);
 
```

The unittest framework does not really require the threads to run because we always run in blocking mode anyway. The tests pass without the thread running. I just added the call to `run()` because it's closer to a real use-case in an IOC. Is there something fundamentally wrong in how we create/start the plugin in the unittests?

MarkRivers, 2016-04-04:
I was wondering if we needed to add the run() call for the unit tests, and I know we will for the simDetectorNoIOC.  Now we really have a problem we don't understand, since the constructor has definitely finished by the time you are calling run there.  We need to get to the bottom of this.  I can only reproduce the problem by pushing to github and seeing if Travis works, but then I can only debug with printf.  Can you crash it on your computer, and show us the stack trace?

ulrikpedersen, 2016-04-04:
Yes, this is a more worrying problem. I can accept that there are all sorts of weird and wonderful stuff going on in a constructor with inheritance and virtual functions. However, I don't understand why this keeps happening even when starting the thread *after* the constructor has completed!

As I mentioned before, I can only re-create the issue on my Ubuntu 14.04 Virtual Machine. Here is the stack trace from GDB:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff2183700 (LWP 659)]
0x00000000006906e0 in ?? ()
(gdb) 
(gdb) 
(gdb) bt
#0  0x00000000006906e0 in ?? ()
#1  0x00007ffff759c3f1 in NDPluginDriver::processTask (this=0x6906f0) at ../NDPluginDriver.cpp:178
#2  0x00007ffff759c37e in processTask (drvPvt=0x6906f0) at ../NDPluginDriver.cpp:158
#3  0x00007ffff5dfca6f in start_routine (arg=0x69ad80) at ../../../src/libCom/osi/os/posix/osdThread.c:389
#4  0x00007ffff651b182 in start_thread (arg=0x7ffff2183700) at pthread_create.c:312
#5  0x00007ffff682b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

ulrikpedersen, 2016-04-04:
I might have a clue. I think the unittest fixture *destructor* gets called before the thread has started up! Remember the thread is not actually used in the unittest which operates in blocking mode - but it also don't have any reason to wait for the thread to start up!!!

I'm investigating that idea a bit further now...

MarkRivers, 2016-04-04:
That would make sense.  If so, then in some ways we have been chasing a non-problem.  But it seems like waiting to start the thread until after the constructor is complete is a cleaner solution.

ulrikpedersen, 2016-04-04:
> I think the unittest fixture destructor gets called before the thread has started up! 

Yes, this seems to be the case. A small 0.1s delay in the destructor of the plugin also avoids the segfault. This may very well have been the problem all along - except we've been focused on the constructor.

> If so, then in some ways we have been chasing a non-problem.

In some ways yes and in some ways no... IOCs don't really need proper destructor support as plugins are typically never destroyed. However, it does irk me to have an unclean shutdown... I know we can't completely delete the underlying asyn ports (as discussed in our meeting here at DLS) - but I do think we should have as good as possible destructor support in our plugin base class (shutting down the thread before deleting the object).

> But it seems like waiting to start the thread until after the constructor is complete is a cleaner solution.

I agree. I think we should stick to this. 

I will work on a solution to the destructor problem now.

ulrikpedersen, 2016-04-04:
> I will work on a solution to the destructor problem now.

I'll commit my additions to be part of #161

ulrikpedersen, 2016-04-05:
> in some ways we have been chasing a non-problem

Yes, as the title of this issue indicate; we were chasing a constructor race condition - when in reality it was a destructor race condition :smile: 

I have now made a fix and run it successfully over 10000 times on my Ubuntu VM. It includes the constructor/thread locking as this is still relevant (to avoid a race condition in the constructor). I will raise another PR and reference this ticket.

PR #163 (2016-04-05): Destructor race condition issue
This fixes #157 which was originally an issue with a race condition between the NDPluginDriver constructor and a thread being started in the constructor. Eventually we also found that there was a similar issue with the destructor. The two are somewhat related and this PR include fixes for both.

- Adding a locking mechanism to the NDPluginDriver constructor and thread so that a thread does not start running until the object is fully initialized.
- Switching NDPluginDriver from using the epics thread C interface to the epicsThreadRunable C++ class. This class has [... cut]
closed 2016-04-05 by MarkRivers