areaDetector/ADCore #165: segmentation fault in NDPluginTimeSeries

ulrikpedersen, 2016-04-06:
As I was testing the imagej 1D change in PR #162 I stumbled upon a segmentation fault.

Setup as follows:

 - TimeSeries plugin connected to ADC simdetector. No other plugins enabled.
 - Start ADC simdetector running continuously.
 - Start TimeSeries acquisition into circular buffer with number of time points set to 20.
 - Increase number of timepoints from 20 to 1024. Crash.

I'm on my RHEL6 desktop. The ADCore code version is the current tip of the "imagej_1d! branch at rev. e29eba5. As this has just been updated with the master branch it should have all fixes for the TimeSeries plugin. 

This crash feels a little similar to the segfault I discovered in reviewing #148 - but that was fixed...

Here is a GDB stacktrace:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff690df700 (LWP 20178)]
0x00007ffff7b4c82e in NDPluginTimeSeries::doTimeSeriesCallbacksT<float> (this=0x741000) at ../NDPluginTimeSeries.cpp:282
282	        signalData_[timeOut] = timeCircular[signal*numTimePoints_ + timeIn];
Missing separate debuginfos, use: debuginfo-install glibc-2.12-1.166.el6_7.7.x86_64 libgcc-4.4.7-16.el6.x86_64 libjpeg-turbo-1.2.1-3.el6_5.x86_64 libstdc++-4.4.7-16.el6.x86_64 libtiff-3.9.4-10.el6_5.x86_64 libxml2-2.7.6-20.el6_7.1.x86_64 ncurses-libs-5.7-4.20090207.el6.x86_64 nss-softokn-freebl-3.14.3-23.el6_7.x86_64 readline-6.0-4.el6.x86_64 zlib-1.2.3-29.el6.x86_64
(gdb) bt
#0  0x00007ffff7b4c82e in NDPluginTimeSeries::doTimeSeriesCallbacksT<float> (this=0x741000) at ../NDPluginTimeSeries.cpp:282
#1  0x00007ffff7b47f7e in NDPluginTimeSeries::doTimeSeriesCallbacks (this=0x741000) at ../NDPluginTimeSeries.cpp:324
#2  0x00007ffff7b48c08 in NDPluginTimeSeries::writeInt32 (this=0x741000, pasynUser=0x133ca28, value=1)
    at ../NDPluginTimeSeries.cpp:465
#3  0x00007ffff6886471 in writeInt32 (drvPvt=0x741000, pasynUser=0x133ca28, value=1)
    at ../../asyn/asynPortDriver/asynPortDriver.cpp:1390
#4  0x00007ffff68907a3 in processCallbackOutput (pasynUser=0x133ca28) at ../../asyn/devEpics/devAsynInt32.c:486
#5  0x00007ffff68690a8 in queueRequest (pasynUser=0x133ca28, priority=asynQueuePriorityLow, timeout=0)
    at ../../asyn/asynDriver/asynManager.c:1483
#6  0x00007ffff688ec77 in processBo (pr=0x781b70) at ../../asyn/devEpics/devAsynInt32.c:1045
#7  0x00007ffff66354c8 in writeValue (prec=0x781b70) at ../boRecord.c:394
#8  process (prec=0x781b70) at ../boRecord.c:242
#9  0x00007ffff59cb2d5 in dbProcess (precord=0x781b70) at ../dbAccess.c:637
#10 0x00007ffff59d672d in scanList (psl=0x18756d0) at ../dbScan.c:688
#11 0x00007ffff59d7968 in periodicTask (arg=0x18756d0) at ../dbScan.c:561
#12 0x00007ffff511329d in start_routine (arg=0x1a7d280) at ../../../src/libCom/osi/os/posix/osdThread.c:397
#13 0x0000003107a07aa1 in start_thread () from /lib64/libpthread.so.0
#14 0x00000031076e893d in clone () from /lib64/libc.so.6
(gdb) print signalData_
$1 = (double *) 0x7ffef4023fc0
(gdb) print timeOut
$2 = 12
(gdb) print numTimePoints_
$4 = 1024
(gdb) print timeIn
$5 = 12
(gdb) print signal
$6 = 3
(gdb) print maxSignals_
$7 = 8
```

ulrikpedersen, 2016-04-08:
After applying the fix in #166 I am no longer able to reproduce the crash. I'm guessing the change in [NDPluginTimeSeries.cpp:280](https://github.com/areaDetector/ADCore/pull/166/files#diff-1ec0cb1bbd9e05aa5fdd791811f9526cL280) is the fix...

closed 2016-04-08 by ulrikpedersen

MarkRivers, 2016-04-08:
I'm a little suspicious that I fixed it, since it should have only fixed a problem when the number of signals (first array dimension) was different from that in the constructor.  So it may arise again, but is difficult to reproduce.

ulrikpedersen, 2016-04-08:
> I'm a little suspicious that I fixed it, since it should have only fixed a problem when the number of signals (first array dimension) was different from that in the constructor. So it may arise again, but is difficult to reproduce.

Yes, I know - I don't quite understand it either. However, it was relatively easy to reproduce the crash yesterday - but it completely went away when merging your fix.

MarkRivers, 2016-04-08:
OK, hopefully it's fixed.

commit 1b897b94 (2016-04-08): Merge pull request #166 from areaDetector/time_series_fix

Time series fix. Fixes #165