areaDetector/ADCore #169: Create unittest for NDPluginDriver

ulrikpedersen, 2016-04-13:
I've started work on a unittest for the NDPluginDriver class. Initially because I wanted to reproduce the leak mentioned in PR #168. However, I think a more extensive test setup for one of the most fundamental classes would be useful. We should be doing terrible things to the class in our unittest! :grinning: 

The starting point (which is not quite working yet) is here: https://github.com/dls-controls/ADCore/tree/unittest-ndplugindriver

ajgdls, 2016-04-13:
OK sure I'll try hard to break it.

MarkRivers, 2017-01-30:
@ulrikpedersen and @ajgdls is there any progress on this?  It is the last item on the R2-6 milestone.  If you think you can get it done this week then we can wait on R2-6, otherwise we can push it to a future release.

MarkRivers, 2017-01-30:
I've moved this to the new Release 2-7 milestone.

ulrikpedersen, 2025-09-20:
Goodness this is a blast from the past... I don't think this is relevant today, 8 years later. Let's close this in the interest of tidying up old tickets.