areaDetector/ADCore #181: Plugin for fitting curves

hinxx, 2016-05-17:
Here at ESS we developed a AD plugin that can fit a curve - e.g. Gaussian with n-peaks - and are willing to share if community finds it useful.

Raw data comes from ROI plugin, 1xN. User can set amplitude, mu, sigma for each peak and baseline.
User can setup st.cmd to have 1 .. n peaks to fit.
Plugin utilizes library http://apps.jcns.fz-juelich.de/doku/sc/lmfit.

It comes with demo CSS OPI:
[image]

mp49, 2019-06-19:
Hi,

I only just noticed this issue. I'm interested in this. Something like this may be useful for fitting neutron time-of-flight or d-space peaks. 

How widely used is it at ESS?

hinxx, 2019-06-19:
Currently it is in production on one-off system, on accelerator. It is expected to be used on upcoming non-invasion profile monitors as well. Main reason for low usega count is due to systems still being designed and developed, with only handful of them deployed.

rettigl, 2021-02-02:
Hi,
I've also found this issue just now. Is this plugin available somewhere, or could you send me the code? I would be very interested. This sounds like a good solution for monitoring beam positions with cameras, or is there something better for that?

MarkRivers, 2021-02-03:
> This sounds like a good solution for monitoring beam positions with cameras, or is there something better for that?

If you have a single beam then the centroid calculation in the NDPluginStats may work well for you. You can set a background pedestal to ignore.

hinxx, 2021-02-16:
> Is this plugin available somewhere, or could you send me the code?

I have not published the code anywhere. You can find a ZIP file [here](https://drive.google.com/file/d/1462qiyhtul3dvPo1yx6Xn2_GSyn70v4w/view?usp=sharing).

closed 2021-02-16 by hinxx