areaDetector/ADCore #183: Data source plugin

hinxx, 2016-05-17:
Here at ESS we developed an AD plugin that accepts external waveform PV as data source and transforms it into NDArray. This way one can utilize all the nifty AD plugins with data not originating from AD supported IOC.
We used it to retrieve data from a USB based power meter that was supported using StreamDevice.
If community fins this useful we are wiling to share.

Screenshot source plugin:
[image]

MarkRivers, 2016-09-29:
I am interested to generalize this into a new ADDriverStdArrays driver.  It will do the inverse operation of the NDPluginStdArrays, i.e. converting from waveform records to NDArrays.  I plan to add support the following properties:

- All asynXXXArray interfaces are supported (i.e. asynInt[8,16,32]Array, asynFloat[32,64]Array
- All values of DataType are supported, so it converts from asynXXXArray type to DataType
- It is fully N-dimensional, i.e. NDimensions PV is supported, as well as arrayDims[] and arrayDims0, arrayDims1, etc.
- It supports a WriteMode parameter = Overwrite or Append.  In Overwrite mode it replaces the current NDArray.  In Append mode it appends to the end of the current NDArray.  This will allow support for constructing 2-D arrays from a 1-D scan record, etc.

This will provide a general purpose driver that allows any Channel Access client to send arrays to an areaDetector plugin chain.

If you can send me the current version I will make the enhancements.

hinxx, 2016-09-29:
Sounds great!
Here is the code: https://bitbucket.org/europeanspallationsource/m-epics-adpluginsource

MarkRivers, 2017-02-26:
Closed via new NDDriverStdArrays driver (https://github.com/areaDetector/NDDriverStdArrays)

closed 2017-02-26 by MarkRivers