areaDetector/ADCore #187: tiffSupport fails to build for static Windows architectures using Visual Studio 2015 Community Edition

kmpeters, 2016-08-25:
I'm using the following:
```
Windows 7 (64-bit)
Visual Studio 2015 Community Edition
Make 4.1 (from the EPICS web site)
Strawberry perl 5.20.3001
```
The following error occurs when building ADCore revision 280b97d8c7277cb1db1ca455bbb07e680f818d40 against both base 3.14.12.5 and base 3.15.4 for static Windows architectures:
```
cl                -nologo -D__STDC__=0 -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE   -Ox -GL -Oy-   -W3 -w44355        -MT    -I. -I../O.Common -I. -
I. -I../os/WIN32 -I.. -I../../../../include/compiler/msvc -I../../../../include/os/WIN32 -I../../../../include       -ID:/synApps-testing/VS15/3.15/base-3.15.4/
include/compiler/msvc -ID:/synApps-testing/VS15/3.15/base-3.15.4/include/os/WIN32 -ID:/synApps-testing/VS15/3.15/base-3.15.4/include      -ID:/synApps-testing/V
S15/3.15/synApps/support/asyn/include  -ID:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADBinaries/include/os/WIN32 -ID:/synApps-testing/VS15/3.15/sy
nApps/support/areaDetector/ADBinaries/include  -ID:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore/include/os/WIN32 -ID:/synApps-testing/VS15/3.1
5/synApps/support/areaDetector/ADCore/include                       -c ../tif_aux.c
tif_aux.c
C:\Program Files (x86)\Windows Kits\10\include\10.0.10586.0\ucrt\stdio.h(1925): warning C4005: 'snprintf': macro redefinition
d:\synapps-testing\vs15\3.15\synapps\support\areadetector\adcore\adapp\tiffsupport\tiffsrc\tif_config.h(50): note: see previous definition of 'snprintf'
C:\Program Files (x86)\Windows Kits\10\include\10.0.10586.0\ucrt\stdio.h(1927): fatal error C1189: #error:  Macro definition of snprintf conflicts with Standard
 Library function declaration
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_BUILD:226: recipe for target 'tif_aux.obj' failed
make[6]: *** [tif_aux.obj] Error 2
make[6]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/tiffSrc/O.win32-x86-static'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_ARCHS:61: recipe for target 'install.win32-x86-static' failed
make[5]: *** [install.win32-x86-static] Error 2
make[5]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/tiffSrc'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'tiffSrc.install' failed
make[4]: *** [tiffSrc.install] Error 2
make[4]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'tiffSupport.install' failed
make[3]: *** [tiffSupport.install] Error 2
make[3]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore/ADApp'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'ADApp.install' failed
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore.install' f
ailed
make[1]: *** [D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADCore.install] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector'
D:/synApps-testing/VS15/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'D:/synApps-testing/VS15/3.15/synApps/support/areaDetector.install' failed
make: *** [D:/synApps-testing/VS15/3.15/synApps/support/areaDetector.install] Error 2
```

kmpeters, 2016-08-30:
The stdio.h file in the above error message comes from the Windows 10 SDK.  If the Windows 10 SDK is not installed, it is likely this error would not occur.

The problem can be fixed by removing the definition of snprintf from tif_config.h:
```
#define snprintf _snprintf
```

This isn't likely to cause any problems; None of the code in tiffSupport calls snprintf.

MarkRivers, 2016-09-01:
According to MSDN (https://msdn.microsoft.com/en-us/library/2ts7cx93(v=vs.140).aspx) snprintf is not present in Visual Studio prior to VS 2015, only _snprintf is defined.  I have changed tif_config.h so it only defines snprintf for versions of Visual Studio prior to 2015.  I have committed to github.  @kmpeters please test with VS 2010 and VS 2015 when you can.

kmpeters, 2016-09-01:
Commit 729dd626d53f82848e176f9218ce31befd9f80e3 fixes this issue.  I'm now able to build tiffSupport dynamically and statically with both VS2010 and VS2015.