areaDetector/ADCore #188: nexusSrc fails to build for dynamic Windows architectures

kmpeters, 2016-08-25:
I'm using the following:
```
Windows 7 (64-bit)
Visual Studio 2010 Professional & Visual Studio 2015 Community Edition
Make 4.1 (from the EPICS web site)
Strawberry perl 5.20.3001
base-3.14.12.5 & base-3.15.4
EPICS_HOST_ARCH windows-x64 & win32-x86
```
The following error occurs when building ADCore revision 280b97d8c7277cb1db1ca455bbb07e680f818d40 using various combinations of Visual Studio and EPICS base.  The errors for windows-x64 and win32-x86 are almost identical; the only difference is that unresolved external symbols are prefixed with two underscores for windows-x64 and three underscores for win32-x86.

**Visual Studio 2010 & base-3.15.4**
```
link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release  /MACHINE:X64      -out:NeXus.dll -implib:NeXus.lib         napi.obj napi5.obj napi
u.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj      D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libhdf5.
lib  D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libszip.lib  D:/synApps-testing/VS10/3.15/synApps/support/areaDetector
/ADBinaries/lib/windows-x64/libzlib.lib
LIBCMT.lib(timeset.obj) : error LNK2005: __daylight already defined in MSVCRT.lib(MSVCR100.dll)
LIBCMT.lib(timeset.obj) : error LNK2005: __timezone already defined in MSVCRT.lib(MSVCR100.dll)
LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; use /NODEFAULTLIB:library
NeXus.dll : fatal error LNK1169: one or more multiply defined symbols found
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_BUILD:281: recipe for target 'NeXus.dll' failed
make[1]: *** [NeXus.dll] Error 1169
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/nexusSrc/O.windows-x64'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_ARCHS:61: recipe for target 'install.windows-x64' failed
make: *** [install.windows-x64] Error 2
```

**Visual Studio 2010 & base-3.14.12.5**
```
link /nologo /subsystem:windows /dll /LTCG /incremental:no /opt:ref /release   /MACHINE:X64      /implib:NeXus.lib /out:NeXus.dll          napi.obj napi5.obj na
piu.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj      D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADBinaries\\lib\\windows-x
64\\libhdf5.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADBinaries\\lib\\windows-x64\\libszip.lib  D:\\synApps-testing\\VS10\\3.14\\sy
nApps\\support\\areaDetector\\ADBinaries\\lib\\windows-x64\\libzlib.lib
LIBCMT.lib(timeset.obj) : error LNK2005: __daylight already defined in MSVCRT.lib(MSVCR100.dll)
LIBCMT.lib(timeset.obj) : error LNK2005: __timezone already defined in MSVCRT.lib(MSVCR100.dll)
LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; use /NODEFAULTLIB:library
NeXus.dll : fatal error LNK1169: one or more multiply defined symbols found
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_BUILD:300: recipe for target 'NeXus.dll' failed
make[5]: *** [NeXus.dll] Error 1169
make[5]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp/nexusSrc/O.windows-x64'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_ARCHS:64: recipe for target 'install.windows-x64' failed
make[4]: *** [install.windows-x64] Error 2
make[4]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp/nexusSrc'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'nexusSrc.install' failed
make[3]: *** [nexusSrc.install] Error 2
make[3]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'ADApp.install' failed
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore.install' failed
make[1]: *** [D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore.install] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector.install' failed
make: *** [D:/synApps-testing/VS10/3.14/synApps/support/areaDetector.install] Error 2
```

**Visual Studio 2015 & base-3.15.4**
```
link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release  /MACHINE:X64      -out:NeXus.dll -implib:NeXus.lib         napi.obj napi5.obj napi
u.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj      D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libhdf5.
lib  D:/synApps-testing/VS15/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libszip.lib  D:/synApps-testing/VS15/3.15/synApps/support/areaDetector
/ADBinaries/lib/windows-x64/libzlib.lib
LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; use /NODEFAULTLIB:library
libhdf5.lib(H5Gbtree2.obj) : error LNK2001: unresolved external symbol __iob_func
libhdf5.lib(H5HFdblock.obj) : error LNK2001: unresolved external symbol __iob_func
libhdf5.lib(H5.obj) : error LNK2001: unresolved external symbol __iob_func
libhdf5.lib(H5Eint.obj) : error LNK2001: unresolved external symbol __iob_func
libhdf5.lib(H5C.obj) : error LNK2001: unresolved external symbol __iob_func
libhdf5.lib(H5HF.obj) : error LNK2001: unresolved external symbol __iob_func
[... 27 lines cut ...]
libhdf5.lib(H5system.obj) : error LNK2001: unresolved external symbol timezone
OLDNAMES.lib(timezone.obj) : error LNK2001: unresolved external symbol timezone
libhdf5.lib(H5system.obj) : error LNK2001: unresolved external symbol _vscprintf
libhdf5.lib(H5system.obj) : error LNK2001: unresolved external symbol _vsnprintf_s
libhdf5.lib(H5Ztrans.obj) : error LNK2001: unresolved external symbol sscanf
OLDNAMES.lib(timezone.obj) : error LNK2001: unresolved external symbol _timezone
NeXus.dll : fatal error LNK1120: 6 unresolved externals
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_BUILD:300: recipe for target 'NeXus.dll' failed
make[5]: *** [NeXus.dll] Error 1120
make[5]: Leaving directory 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore/ADApp/nexusSrc/O.windows-x64'
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_ARCHS:64: recipe for target 'install.windows-x64' failed
make[4]: *** [install.windows-x64] Error 2
make[4]: Leaving directory 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore/ADApp/nexusSrc'
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'nexusSrc.install' failed
make[3]: *** [nexusSrc.install] Error 2
make[3]: Leaving directory 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore/ADApp'
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'ADApp.install' failed
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore'
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore.install' failed
make[1]: *** [D:/synApps-testing/VS15/3.14/synApps/support/areaDetector/ADCore.install] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector'
D:/synApps-testing/VS15/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS15/3.14/synApps/support/areaDetector.install' failed
make: *** [D:/synApps-testing/VS15/3.14/synApps/support/areaDetector.install] Error 2
```

MarkRivers, 2016-09-06:
I cannot reproduce your problems with Visual Studio 2010.  It builds fine for me.  I have tested the following:
- win32-x86, win32-x86-static on 3.14.12.5 and 3.15.4
- windows-x64, windows-x64-static on 3.14.12.5 and 3.15.4

I have not tested Visual Studio 2015.

@kmpeters:
Are you building with debug?
Are you building with GraphicsMagick?

kmpeters, 2016-09-06:
I don't believe I am building with debug.  I'm building for the same architectures you are and I have not modified any of the base-*/configure/os/* files to enable debugging for non-debugging architectures.  Here is a log of my build of the nexusSrc dir using VS2010 & base-3.15.4:
[makelog.txt](https://github.com/areaDetector/ADCore/files/457486/makelog.txt)

I assume I'm not building with GraphicsMagick.  I'm currently building withtout a CONFIG_SITE.local, so I should be doing whatever the default is.   I'll copy EXAMPLE_CONFIG_SITE.local to CONFIG_SITE.local to make it explicit that I'm not building with GraphicsMagick and see if that resolves any of the build problems.

MarkRivers, 2016-09-06:
You are having failures building with VS2010 and 3.14.12.5.  I don't think there is anything that has changed from ADCore R2-4 to the current master branch that would cause the failure you are seeing.  Have you tried to build R2-4 with everything else the same ( VS 2010, 3.14.12.5) and see if that works?

MarkRivers, 2016-09-06:
I see your problem.  This is the link command that is failing for you:

    link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release  /MACHINE:X64      -out:NeXus.dll -implib:NeXus.lib        napi.obj napi5.obj napiu.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj       
    D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libhdf5.lib  
    D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libszip.lib  
    D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADBinaries/lib/windows-x64/libzlib.lib 
    LIBCMT.lib(timeset.obj) : error LNK2005: __daylight already defined in MSVCRT.lib(MSVCR100.dll)
    LIBCMT.lib(timeset.obj) : error LNK2005: __timezone already defined in MSVCRT.lib(MSVCR100.dll)

This is the link command that is succeeding for me:

    link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release  /MACHINE:X64      -out:NeXus.dll -implib:NeXus.lib        napi.obj napi5.obj napiu.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj       
    J:/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/windows-x64/hdf5.lib  
    J:/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/windows-x64/szip.lib  
    J:/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/windows-x64/zlib.lib
    Creating library NeXus.lib and object NeXus.exp
    Generating code
     Finished generating code
    "Installing shared library ../../../bin/windows-x64/NeXus.dll"
    "Installing library ../../../lib/windows-x64/NeXus.lib"

Note that you are linking against libhdf5.lib, libszip.lib, and libzip.lib, while I am linking against hdf5.lib, szip.lib, and zlib.lib.  

Which libraries to link with is determined by these lines in commonLibraryMakefile

    ifeq ($(HDF5_STATIC_BUILD), NO)
      USR_CXXFLAGS_WIN32    += -DH5_BUILT_AS_DYNAMIC_LIB
      USR_CFLAGS_WIN32      += -DH5_BUILT_AS_DYNAMIC_LIB
      LIB_LIBS_WIN32        += hdf5 szip zlib
    else
      USR_CXXFLAGS_WIN32    += -DH5_BUILT_AS_STATIC_LIB
      USR_CFLAGS_WIN32      += -DH5_BUILT_AS_STATIC_LIB
      LIB_LIBS_WIN32        += libhdf5 libszip libzlib
   endif

So HD5_STATIC_BUILD must not be set to NO for you when building windows-x64 (dynamic), but it should be.  

That is because you are not using a CONFIG_SITE.local, which must exist and contain these lines:

    ifeq ($(OS_CLASS), WIN32)
        HDF5_STATIC_BUILD=$(STATIC_BUILD)
    endif

I believe that your problems will go away if you create areaDetector/configure/CONFIG_SITE.local using the supplied EXAMPLE_CONFIG_SITE.local.

kmpeters, 2016-09-06:
> I believe that your problems will go away if you create areaDetector/configure/CONFIG_SITE.local using the supplied EXAMPLE_CONFIG_SITE.local.

I just tested this.  Copying EXAMPLE_CONFIG_SITE.local to CONFIG_SITE.local eliminates the nexusSrc build problems.

Are you able to build the ADExample IOCs using VS2010 and base-3.15.4?  I'm running into a problem, but only when building for ﻿windows-x64-static.

kmpeters, 2016-09-06:
I'll create another issue for the ADCSimDetectorIOC build issue so this one can be closed.

MarkRivers, 2016-09-07:
Closing this issue, problem was that CONFIG_SITE.local was not being used, so HDF5_STATIC_BUILD was not correctly defined.

closed 2016-09-07 by MarkRivers