areaDetector/ADCore #189: netCDF fails to build against base-3.15.4 for dynamic Windows architectures

kmpeters, 2016-08-25:
I'm using the following:
```
Windows 7 (64-bit)
Visual Studio 2010 Professional & Visual Studio 2015 Community Edition
Make 4.1 (from the EPICS web site)
Strawberry perl 5.20.3001
EPICS base-3.15.4
EPICS_HOST_ARCH windows-x64 & win32-x86
```
The following error occurs when building ADCore revision 280b97d8c7277cb1db1ca455bbb07e680f818d40 using both Visual Studio 2010 & 2015.  The error messages are identical for both windows-x64 and win32-x86.
```
link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release  /MACHINE:X64      -out:netCDF.dll -implib:netCDF.lib         attr.obj dim.obj look
up3.obj nc.obj nc3dispatch.obj nclistmgr.obj ncx.obj posixio.obj putget.obj string.obj utf8proc.obj v1hpg.obj var.obj att.obj dim1.obj dispatch.obj error.obj fi
le.obj nc_uri.obj var1.obj stub.obj
Generating code
Finished generating code
"Installing shared library ../../../bin/windows-x64/netCDF.dll"
"Installing library ../../../lib/windows-x64/netCDF.lib"
installEpics.pl: No such file 'netCDF.lib' at D:/synApps-testing/VS10/3.15/base-3.15.4/bin/windows-x64/installEpics.pl line 53.
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_BUILD:391: recipe for target '../../../lib/windows-x64/netCDF.lib' failed
make[5]: *** [../../../lib/windows-x64/netCDF.lib] Error 2
make[5]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/netCDFSrc/O.windows-x64'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_ARCHS:61: recipe for target 'install.windows-x64' failed
make[4]: *** [install.windows-x64] Error 2
make[4]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/netCDFSrc'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'netCDFSrc.install' failed
make[3]: *** [netCDFSrc.install] Error 2
make[3]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'ADApp.install' failed
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore.install' f
ailed
make[1]: *** [D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore.install] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector.install' failed
make: *** [D:/synApps-testing/VS10/3.15/synApps/support/areaDetector.install] Error 2
```

MarkRivers, 2016-08-28:
This should be fixed in commit https://github.com/areaDetector/ADCore/commit/681d3b732ec777342ccbe7a1a34287d4fab13ce9

The fix works for mingw, please test on VS2010 and VS2015.

Mark

kmpeters, 2016-08-29:
I updated to the head (cbeabd93df3cb858ad957500d32c3d100359e0e3) and confirmed that the fix also works for VS2010 and VS2015.

MarkRivers, 2016-08-30:
Closing, it is fixed.

closed 2016-08-30 by MarkRivers