areaDetector/ADCore #190: Prior to releasing ADCore R2-5 it should build with 3.15.4 and mingw

MarkRivers, 2016-08-30:
Prior to releasing R2-5 ADCore should be fixed

- To build with EPICS base 3.15.4 
- To build with mingw from Windows (e.g. using Strawberry Perl which comes with mingw) and cross-compiled from Linux.

MarkRivers, 2016-09-11:
I am becoming convinced that we should get rid of ADBinaries.  Here are the reasons:

- The number of libraries we need to supply will become very large.  
  - We are finding that a library built with Visual Studio 2010 cannot be used with Visual Studio 2015, even when it contains only C code, not C++.  This means we would need to supply separate libraries for VS 2010, 2013, and 2015 at present to satisfy most users current build tools.
  - We need separate libraries for 32 and 64-bit
  - We need separate libraries for static and dynamic builds
  - We need separate libraries for debug and release builds
  - For Visual Studio we thus need 3*2*2*2 = 24 versions of the library.
  - We need libraries for win32-x86-mingw.  This also needs 32/64-bit, and static/dynamic (and debug/release?).  That is 8 more versions of the library, for a total of 32.
- Someone needs to maintain these, and this cannot be done with the standard EPICS build system.  hdf5, szip, zlib need cmake, GraphicsMagick and xml2 are Visual Studio projects.  These are a pain because the project format is specific to a specific version of Visual Studio.

I propose that we do for xml2, szip, zlib, hdf5, and GraphicsMagick the same thing that is already being done for tiff, jpeg, netCDF, and NeXus, i.e. use the EPICS build system to build the libraries from source.  It is a bit of work to set it up initially, but it is fairly easy to maintain when we decide we want to move to a new release of one of these libraries.

I have begun this process in a new branch of ADCore called no_ADBinaries.  I created a new source directory, ADApp/xml2src.  This uses libxml2-2.9.4.  I have converted to the EPICS build system, and I have successfully built for 6 architectures with VS 2010 and 32-bit mingw: win32-x86, win32-x86-static, windows-x64, windows-x64-static, win32-x86-ming2, and win32-x86-mingw-static.

I hope to finish the other libraries in the next couple of weeks.

MarkRivers, 2016-09-11:
Another advantage of building using the EPICS build system is that it raises the possibility of supporting hdf5 and GraphicsMagick on vxWorks and RTEMS.  As ADCore is now being used for detectors like quadEM and ADCs which often run on real-time OS, so this becomes something to consider.

ulrikpedersen, 2016-09-12:
> I am becoming convinced that we should get rid of ADBinaries.  Here are the reasons:
> - The number of libraries we need to supply will become very large.
>   - We are finding that a library built [...]

I agree in that this is a pain and the build-matrix unmanageable in the long run. In fact it has already proven difficult to maintain a consistent build of the current 4 builds of HDF5 - having to maintain 32 builds is just not possible.

> I propose that we do for xml2, szip, zlib, hdf5, and GraphicsMagick the same thing that is already being done for tiff, jpeg, netCDF, and NeXus, i.e. use the EPICS build system to build the [...]

In my view there is an alternative: document the required 3rd party dependencies for users to install. Most if not all of these libraries come with proper windows installers - and build instructions for those who wish to do further developments.

> I have begun this process in a new branch of ADCore called no_ADBinaries. 

It's a good idea (but a *big* job) to provide an EPICS based build of these dependencies as an alternative to native installs (on windows at least). But why in ADCore? Can we not re-use ADBinaries? OK, the name "ADBinaries" is no longer appropriate if it contain sources... I am worried about putting *all* these dependencies back into ADCore again and create this big, complex build.

Not all of these libraries are strictly required: they can either be separately installed in the system or some (like hdf5) relate only to a single plugin which could be separated from ADCore. I would like to consider which of the dependencies are truly required by core functionality and whether we should start splitting some plugins out into separate repos...

MarkRivers, 2016-09-16:
We may want to move these into a separate repository in the future.  

However, for now I have put szipSrc, zlibSrc, xml2Src, and hdf5Src in ADCore/ADApp.  Getting them to build under the EPICS build system was actually quite easy.  I just copied the .c and .h files to the appropriate directories and created a Makefile to install the .h files and compile the .c files.  I had to edit a config.h or equivalent to put in some architecture specific changes, but they were minor.  I committed the vendor version of the library first so all of my changes are in the git history.

I am working in the no_ADBinaries branch in ADCore.  There is also a new no_ADBinaries branch in areaDetector (top-level), which removes ADBinaries from EXAMPLE_RELEASE_LIBS.local.  This file needs to be copied to RELEASE_LIBS.local so that it uses the files from ADCore, not ADBinaries.

szipSrc, zlibSrc and xml2Src build fine on Visual Studio 2010 (32/64, static/dynamic), mingw, and vxWorks. 

hdf5 builds on Visual Studio (32/64, static/dynamic).  More work is needed for mingw and vxWorks.

The ADExample applications link fine with hdf5 on Visual Studio.  However, when the IOC startup script creates the plugin I get the following messages:

    NDFileHDF5Configure("FileHDF1", 20, 0, "SIM1", 0)
    HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 0:
    #000: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #001: ..\H5Tconv.c line 7749 in H5T__conv_long_float(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #002: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #003: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #004: ..\H5Tconv.c line 7724 in H5T__conv_uint_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #005: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #006: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #007: ..\H5Tconv.c line 7649 in H5T__conv_int_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #008: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #009: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #010: ..\H5Tconv.c line 7574 in H5T__conv_ushort_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #011: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #012: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #013: ..\H5Tconv.c line 7499 in H5T__conv_short_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #014: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #015: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #016: ..\H5Tconv.c line 7424 in H5T__conv_uchar_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #017: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #018: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
[... 23 lines cut ...]
    major: Datatype
    minor: Unable to initialize object
    #027: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #028: ..\H5Tconv.c line 7219 in H5T__conv_double_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
    #029: ..\H5T.c line 2297 in H5T_register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
    #030: ..\H5T.c line 4520 in H5T_path_find(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
    #031: ..\H5Tconv.c line 7160 in H5T__conv_float_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object

When the IOC exits I get the following error:

    epics> exit
    HDF5: infinite loop closing library
      P,P,FD,E,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I,I

It appears that something is not initializing correctly when the library starts up.  There don't appear to be any serious compiler warnings when building.  I could use some help in figuring out what is wrong.  I think the benefit of building everything with the EPICS build system makes it worthwhile to fix.  This is the only way I know to be able to support all of the architectures we want to support and to minimize the work for people who want to build on Windows with various compiler versions.

MarkRivers, 2016-09-17:
I figured out the problem.  H5Tinit.c is normally created as part of the build process when using CMake on Windows or the ./configure generated Makefile on Linux.  I was using a version of H5Tinit.c built on Linux, which is not correct for the Windows compiler. 

I changed the Makefile to build H5detect executable, and then ran that manuall to create H5Tinit.c on 64-bit Windows.  That worked fine, for both static and dynamic.  I am now able to save HDF5 files using the library built in ADCore.  I have tested szip and zlib compression and Single and Stream modes.  I need to create several versions of H5Tinit.c for 32/64 bit Visual Studio, etc.

MarkRivers, 2016-09-18:
I have created H5Tinit_win32.c and H5Tinit_windows-x64.c, and changed the Makefile to compile the appropriate one when building the library. It now builds correctly and appears to create HDF5 files OK on 32/64 static/dynamic with VS 2010.  

The only problem is that the library files are much larger than the ones that are included in ADBinaries.  I believe this is causing the link step when building applications to be significantly slower.

These are the ADBinaries files:

    corvette:~/devel/areaDetector/ADBinaries>ls -lh lib/windows-x64*/*hdf5* bin/windows-x64*/*hdf5*
    -rwxrwxr-x 1 epics epics 3.6M Jul 19 11:00 bin/windows-x64/hdf5.dll
    -rwxrwxr-x 1 epics epics  17M Jul 19 11:00 lib/windows-x64-static/libhdf5.lib
    -rwxrwxr-x 1 epics epics 551K Jul 19 11:00 lib/windows-x64/hdf5.lib

These are the files built in ADCore:

    corvette:devel/areaDetector-2-5/ADCore>ls -lh lib/windows-x64*/*hdf5* bin/windows-x64*/*hdf5*
    -rwxr-xr-x 1 epics epics  12M Sep 18 09:20 bin/windows-x64/hdf5.dll
    -rwxr-xr-x 1 epics epics  44M Sep 18 08:29 lib/windows-x64-static/hdf5.lib
    -rwxr-xr-x 1 epics epics 550K Sep 18 09:20 lib/windows-x64/hdf5.lib

The dynamic .lib file is the same size (550 K).  However, the dynamic  .dll file is more than 3 times larger, and the static .lib file is more than 2.5 times larger.  I think this must be either because I am including additional source files that are not needed and were not included with the CMake build, or the compile or link flags are different.  Since I have the log files that were produced with CMake I will try to figure this out and see I can get similar size files to ADBinaries.

Even in the current version it is ready for testing.  It should now work with any version of Visual Studio, with any supported EPICS architecture (win32-x86, win32-x86-static, win32-x65-debug, etc.).

MarkRivers, 2016-09-19:
I figured out the issue.  The files being compiled into the library are the same.  The issue is compiler flags.  The important difference is that CMake is using 

    /O2 /Ob1

    /O2 Generate fast code.  Equivalent to:
      /Og Global optimization
      /Oi Generate intrinsic functions
      /Ot Favor fast code
      /Oy Frame pointer omission
      /Ob2 Inline expand not only explicit inline; compiler can inline other functions 
           This is changed to /Ob1 by the next flag on the compiler line
      /Gs Buffers security check
      /GF Enables string pooling
      /Gy Enables function-level linking
    /Ob1 Inline expand only explicit inline functions

EPICS uses different flags, determined by this line in base/configure/CONFIG.win32-x86.win32-x86
OPT_CFLAGS_YES = /Ox /GL

    /Ox This is equivalent to
      /Og Global optimization
      /Oi Generate intrinsic functions
      /Ot Favor fast code
      /Oy Frame pointer omission
      /Ob2 Inline expand not only explicit inline; compiler can inline other functions
    /GL  Enable whole program optimization

So CMake first enables /Ob2 (via /O2) but then disables it via /Ob1.  /Ob2 enables inline function expansion even for functions not declared inline.

EPICS enables /Ob2 (via /Ox) and never disables it with /Ob1.

For testing I changed EPICS base CONFIG_SITE.local.windows-x64.windows-x64 to have the lines:
OPT_CFLAGS_YES = /O2 /Ob1
OPT_CXXFLAGS_YES = /O2 /Ob1

It is thus then using similar flags to CMake.  When I do this the size of O.windows-x64-static/hdf5.lib is reduced from 44 MB to 14 MB.  This is actually less than the CMake version which is 17 MB.  So it is clearly these flags that are making the file large.  I am not changing the hdf5Src/Makefile to reduce the file size, but users can change CONFIG_SITE.local.windows-x64.windows-x64 if desired to reduce file size and reduce the time to link.

ulrikpedersen, 2016-09-19:
Wow! I'm well impressed that you got this working! :+1:  

Please, could you document the details of the configuration options that you've built the the various libraries with? I'm thinking a README.md in each Src/ dir would be appropriate. What I'm after is to be able to re-create the same build of these libraries externally by using the package `configure` script (or cmake)... 

For example we need to know whether the HDF5 library thread-safe option was enabled at build time.

> We may want to move these into a separate repository in the future.

Yes, I would really like to see that.

MarkRivers, 2016-09-20:
I have created a new repository, areaDetector/ADSupport.  I have moved the following directories out of ADCore/ADApp into ADSupport/supportApp

hdf5Src
netCDFSrc
nexusSrc
szipSrc
tiffSupport
xml2Src
zlibSrc

ADSupport effectively replaces ADBinaries, but it is used not only for Windows, but also for Linux and other archs (for netCDFSrc and nexusSrc).

I have preserved the revision history when moving from ADCore to ADSupport.

I have modified areaDetector/Makefile and areaDetector/configure to build ADSupport.

I have tested linux-x86 and windows-x64-static builds from the top-level of areaDetector.  I encourage others to test this as soon as possible.

I like this solution, since all of the support needed for plugins is now in this separate repository.  Most of these packages are available as package installs on Linux, but we build them from source on Windows and vxWorks.

I will follow Ulrik's suggested and add documentation in ADSupport on the configuration options used to build each module.

xiaoqiangwang, 2016-09-20:
Although HDF5 is available for most Linux system, but it normally not up to date. Would it be possible to build HDF5 as well on Linux?

MarkRivers, 2016-09-20:
I forgot to mention that to test this you need to use the master branch of ADSupport and the no_ADBinaries branch of both ADCore and top-level areaDetector.

MarkRivers, 2016-09-20:
>Although HDF5 is available for most Linux system, but it normally not up to date. Would it be possible to build HDF5 as well on Linux?

On Linux it is quite easy to build the package in the "normal" manner, e.g.

    ./configure --with-szlib=[path_to_szip]
    make

This does not require root permission if it is done in a user directory.  It generates an H5config.h file that reflects the host architecture and the host software environment.  It also autogenerates H5Tinit.c as part of the build process which knows about the host hardware (x-x64 vs arm, etc.).  So I would prefer not to build on Linux in ADSupport.  The only problem is that it requires some effort to build both the 32-bit and 64-bit versions on the same host.  I do that as part of my testing of areaDetector, but most people don't need to do that.

Windows does not come with support for ./configure and requires CMake to build.  That is something I don't want users to have to do.  Building with EPICS also lets me easily test 32/64 bit, Visual Studio 2010/2015, debug/release just by using different EPICS_HOST_ARCH values in that same build tree.  That is hard to do with CMake.

./configure does not support cross-compiling like vxWorks and mingw running on Linux.  So we really need EPICS builds for those.

MarkRivers, 2016-09-21:
>I will follow Ulrik's suggested and add documentation in ADSupport on the configuration options used to build each module.

I have now added a README.epics file to ADSupport/supportApp/[hdf5Src, netCDFSrc, nexusSrc, szipSrc, tiffSupport/jpegSrc, tiffSupport/tiffSrc, xml2Src, zlibSrc]

These files document what I changed to get these packages to build on EPICS.  For hdf5 README.epics includes a detailed analysis of the CMake log files to see what flags were being passed to the compiler.  I found that I had indeed missed passing the H5_HAVE_THREADSAFE flag for dynamic builds) and the NDEBUG flag for all builds.  I have fixed that.

I found that the no_ADBinaries branch of ADCore had not been correctly pushed to github.  It is now OK.

I just tested the ADExample simDetector IOC with HDF file writing on the following architectures with Visual Studio 2010 and EPICS base 3.14.12.5

win32-x86
win32-x86-static
windows-x64
windows-x64-static

On all architectures I tested HDF5 compression = none, szip, and zlib.

Everything seems to work correctly.  I encourage others to test it, particularly to make sure it works with base 3.15.4 and Visual Studio 2015.

MarkRivers, 2016-09-23:
I tested 2 things today:

I was able to run the simDetector IOC in ADExample on the win32-x86-mingw architecture.  I built this with the cross-compiler running on Linux.  This means that we can now build Windows executables without a Windows build system.  This could be very useful.  I tested JPEG, TIFF, and netCDF file writing.  HDF5 still needs to be tweaked to build with mingw.  There seems to be something flaky with exception handling (?).  The IOC will sometimes fail to start with this error, but trying again it will work.

    terminate called after throwing an instance of 'ParamValNotDefined'
      what():  paramVal::getInteger value not defined

    This application has requested the Runtime to terminate it in an unusual way.
    Please contact the application's support team for more information.

I booted the simDetector IOC under vxWorks.  I tested JPEG, TIFF, and netCDF file writing.  JPEG and TIFF were not previously supported under vxWorks.  All of them work fine.  HDF5 still needs to be tweaked to build on vxWorks.

hinxx, 2016-09-24:
One benefit to have hdf5 compiling for Linux would be not to worry about the location of libs on different Linux flavors. I prefer using Ubuntu over CentOS but I need to work with both, and location of hdf5 libs on those is totaly different.

MarkRivers, 2016-09-25:
I have now succeeded in building HDF5 and Nexus for win32-x86-mingw, win32-x86-mingw-static, and vxWorks-ppc32 in ADSupport.  This means that HDF5 and Nexus are now supported on all platforms that areaDetector supports.

Here are the results of very quick testing in ADExample simDetector.

- win32-x86-mingw-static.  The application will often crash on startup, reporting exception problems as reported previously.  I don't know if this is a bug in asynPortDriver, something flaky with mingw, or some other problem.  However, on repeatedly starting it does eventually succeed.  I tested compression=none, szip, and zlib and all seem OK.

- win32-x86-mingw.  The application fails to start with a Windows dialog error. This happens before the startup script is read, so it seems like perhaps a DLL path problem.  The error is very uninformative, code 0xc00000cb, nothing about a missing DLL or anything.  I will work on this.

- vxWorks-ppc32.  I tested compression=none, szip, and zlib.  Using h5dump on Linux on the resulting files showed none and zlib to be OK, but szip gave an error dumping the data (but not the header).  vxWorks required dummy implementations of flock() and ftruncate(), so there may well be errors in some use cases.

Everything is committed to the master branch in ADSupport and ADExample, and the no_ADBinaries branch in ADCore.

MarkRivers, 2016-09-27:
I have now built win32-x86-mingw and win32-x86-mingw-static on a Windows machine, rather than cross-compiling on Linux.  I used the latest version of MinGW for this build, which uses gcc 5.3.0.  The problems described above with flaky startup on static build and complete failure to startup on dynamic do not occur, both work fine when built on Windows.  So the problems appear to be confined to the cross-compile and I am not going to worry about them very much.  The cross-compiler I was using was gcc 4.5.3 on Fedora 15.

I also found that the HDF5 files produced with szip compression not being readable also happened on mingw and Visual Studio as well, not only vxWorks.  The problem turned out to be the version of h5dump I was running.  I was using the system version in /usr/bin, which is apparently built without the szip option.  Once I switched to the version in /usr/local/bin which I built with szip enabled, all of the HDF5 files produced with Visual Studio, mingw, and vxWorks with szip compression can be read fine.  So I think the versions zlib, szip, and hdf5 being built in ADSupport are working properly on Visual Studio, mingw, and vxWorks.

There is a set of 4 HDF5 "TIME" datatypes that will not compile on vxWorks, and so are temporarily disabled for that architecture.  The code is deeply nested macros that is very hard to understand, so I plan to ask the hdfgroup what could be causing that problem.

MarkRivers, 2016-09-30:
@xiaoqiangwang wrote:

> Although HDF5 is available for most Linux system, but it normally not up to date. Would it be possible to build HDF5 as well on Linux?

And @hinxx wrote:

> One benefit to have hdf5 compiling for Linux would be not to worry about the location of libs on different Linux flavors. I prefer using Ubuntu over CentOS but I need to work with both, and location of hdf5 libs on those is totaly different.

On further thought it is not difficult to support building HDF5 in ADSupport on Linux, so I am in the process of doing that.  I have changed areaDetector/configure/CONFIG_SITE.local to include the following:

    # Configure which 3rd party libraries to use and where to find them.
    # For each library XXX the following definitions are used:
    # WITH_XXX      Build the plugins and drivers that require this library.
    #               Build the source code for this library in ADSupport if XXX_EXTERNAL=NO.
    # XXX_EXTERNAL  If NO then build the source code for this library in ADSupport.
    # XXX_INCLUDE   If XXX_EXTERNAL=YES then this is the path to the include files for XXX.
    #               However, if XXX is a system library whose include files are in a 
    #               standard include search path then do not define XXX_INCLUDE.
    # XXX_LIB       If XXX_EXTERNAL=YES then this is the path to the library files for XXX.
    #               However, if XXX is a system library whose library files in a 
    #               standard library search path then do not define XXX_LIB.

    # The default defined here is appropriate for most Linux systems.

    # HDF5 is required for the NDFileHDF5 and NDFileNexus plugins
    WITH_HDF5     = YES
    HDF5_EXTERNAL = YES
    #HDF5_INCLUDE =
    #HDF5_LIB     =

Similar definitions are used for SZIP, XML2, NETCDF, etc.

These settings can be overridden in CONFIG_SITE.local.$(OS_CLASS), CONFIG_SITE.local.$(EPICS_HOST_ARCH), and CONFIG_SITE.local.$(EPICS_HOST_ARCH).$(T_A)

ADSupport, commonLibraryMakefile, and commonDriverMakefile will use these flags.  If you set HDF5_EXTERNAL=NO then it will build HDF5 in ADSupport and use that version of the library when building your applications.  

If you set WITH_JPEG=NO for example then no code that uses JPEG will be built.

hinxx, 2016-09-30:
I am very interested in trying this out once available.

MarkRivers, 2016-10-05:
@hinxx and @xiaoqiangwang local building of HDF5 on Linux is now ready for testing on the no_ADBinaries branch of ADCore and areaDetector (top-level) and the master branch of ADSupport and ADExample.  You need to change areaDetector/configure/CONFIG_SITE.local to set SZIP_EXTERNAL=NO, ZLIB_EXTERNAL=NO, HDF_EXTERNAL=NO and then build ADSupport.  Let me know if you have problems.

xiaoqiangwang, 2016-10-05:
I tested the builds on 64bit Scientific Linux 6, with SZIP and HDF5. 

When I look at ``os/Linux/H5pubconf_64.h``, does it mean the thread support is disabled?
```
/* #undef H5_HAVE_LIBPTHREAD */
/* #undef H5_HAVE_PTHREAD_H */
/* #undef H5_HAVE_THREADSAFE */
/* #undef H5_SYSTEM_SCOPE_THREADS */
```

MarkRivers, 2016-10-05:
Thanks for spotting that.

I ran configure again in the hdf5-1.10.0-patch1 directory, but this time adding the --enable-threadsafe and --disable-hl flags.  I have updated the following files in ADSupport/supportApp/hdf5Src/.

    README.epics
    os/Linux/H5pubconf_32.h. 
    os/Linux/H5pubconf_64.h. 

I built ADSupport, ADCore and ADExample with the new version OK.

I have not tested 2 HDF5 writers in one IOC so please let me know if that works.

xiaoqiangwang, 2016-10-05:
Thanks. Now it works with 2 HDF5 writers in one IOC. Will this also be configured for Windows?
```
$ grep THREADSAFE WIN32/H5pubconf.h
/* #undef H5_HAVE_THREADSAFE */
```

MarkRivers, 2016-10-05:
> Thanks. Now it works with 2 HDF5 writers in one IOC. Will this also be configured for Windows?

That should already be handled OK in the Makefile.  This is the relevant section.  Note that it sets -D H5_HAVE_THREADSAFE if this is a dynamic build using Visual Studio.  I was not able to get it to work with dynamic builds with mingw.  The HDF5 documentation says that H5_HAVE_THREADSAFE does not work with static builds on Windows, though I am not sure why.

    USR_CFLAGS += -D NDEBUG
    ifeq ($(SHARED_LIBRARIES), YES)
      USR_CFLAGS_WIN32 += -D H5_BUILT_AS_DYNAMIC_LIB 
      USR_CFLAGS_WIN32 += -D hdf5_shared_EXPORTS
      ifeq (mingw, $(findstring mingw, $(T_A)))
        USR_CFLAGS_WIN32 += -D__MSVCRT_VERSION__=0x800
      else
        # The thread code on mingw needs work, get compiler errors
        USR_CFLAGS_WIN32 += -D H5_HAVE_THREADSAFE
      endif
    endif

MarkRivers, 2016-10-06:
I believe this issue is now all taken care of.

closed 2016-10-06 by MarkRivers