areaDetector/ADCore #191: tiffSupport fails to build against base-3.15.4 for win32-x86 & windows-x64

kmpeters, 2016-08-30:
I'm using the following:
```
Windows 7 (64-bit)
Visual Studio 2015 Community Edition & Visual Studio 2010 Professional Edition
Make 4.1 (from the EPICS web site)
Strawberry perl 5.20.3001
base-3.15.4
```

The following error occurs when building ADCore revision cbeabd93df3cb858ad957500d32c3d100359e0e3 for both win32-x86 and windows-x64:
```
link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release       -out:jpeg.dll -implib:jpeg.lib         jcapimin.obj jcapistd.obj jccoefct.obj
 jccolor.obj jcdctmgr.obj jchuff.obj jcinit.obj jcmainct.obj jcmarker.obj jcmaster.obj jcomapi.obj jcparam.obj jcphuff.obj jcprepct.obj jcsample.obj jctrans.obj
 jdapimin.obj jdapistd.obj jdatadst.obj jdatasrc.obj jdcoefct.obj jdcolor.obj jddctmgr.obj jdhuff.obj jdinput.obj jdmainct.obj jdmarker.obj jdmaster.obj jdmerge
.obj jdphuff.obj jdpostct.obj jdsample.obj jdtrans.obj jerror.obj jfdctflt.obj jfdctfst.obj jfdctint.obj jidctflt.obj jidctfst.obj jidctint.obj jidctred.obj jqu
ant1.obj jquant2.obj jutils.obj jmemmgr.obj jmemnobs.obj
Generating code
Finished generating code
"Installing shared library ../../../../bin/win32-x86/jpeg.dll"
"Installing library ../../../../lib/win32-x86/jpeg.lib"
installEpics.pl: No such file 'jpeg.lib' at D:/synApps-testing/VS10/3.15/base-3.15.4/bin/win32-x86/installEpics.pl line 53.
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_BUILD:391: recipe for target '../../../../lib/win32-x86/jpeg.lib' failed
make[2]: *** [../../../../lib/win32-x86/jpeg.lib] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/jpegSrc/O.win32-x86'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_ARCHS:61: recipe for target 'install.win32-x86' failed
make[1]: *** [install.win32-x86] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/jpegSrc'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'jpegSrc.install' failed
make: *** [jpegSrc.install] Error 2
```

This problem is similar to https://github.com/areaDetector/ADCore/issues/189

The error can be eliminated by changing these two lines in **tiffSupport/jpegSrc/os/WIN32/jmorecfg.h** from this:
```
#define GLOBAL(type)		type
#define EXTERN(type)		extern type
```
to this:
```
#define GLOBAL(type)		__declspec(dllexport) type
#define EXTERN(type)		extern __declspec(dllexport) type
```
A better solution would probably involve adding a symbol that is only defined when building shared libraries on Windows.

The build then progresses until a similar error occurs in tiffSrc:
```
link -nologo /subsystem:windows /dll -LTCG -incremental:no -opt:ref -release       -out:tiff.dll -implib:tiff.lib         tif_aux.obj tif_close.obj tif_codec.ob
j tif_color.obj tif_compress.obj tif_dir.obj tif_dirinfo.obj tif_dirread.obj tif_dirwrite.obj tif_dumpmode.obj tif_error.obj tif_extension.obj tif_fax3.obj tif_
fax3sm.obj tif_getimage.obj tif_jbig.obj tif_jpeg.obj tif_jpeg_12.obj tif_ojpeg.obj tif_flush.obj tif_luv.obj tif_lzw.obj tif_next.obj tif_open.obj tif_packbits
.obj tif_pixarlog.obj tif_predict.obj tif_print.obj tif_read.obj tif_stream.obj tif_swab.obj tif_strip.obj tif_thunder.obj tif_tile.obj tif_version.obj tif_warn
ing.obj tif_write.obj tif_zip.obj tif_win32.obj      User32.lib
Generating code
Finished generating code
"Installing shared library ../../../../bin/win32-x86/tiff.dll"
"Installing library ../../../../lib/win32-x86/tiff.lib"
installEpics.pl: No such file 'tiff.lib' at D:/synApps-testing/VS10/3.15/base-3.15.4/bin/win32-x86/installEpics.pl line 53.
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_BUILD:391: recipe for target '../../../../lib/win32-x86/tiff.lib' failed
make[2]: *** [../../../../lib/win32-x86/tiff.lib] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/tiffSrc/O.win32-x86'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_ARCHS:61: recipe for target 'install.win32-x86' failed
make[1]: *** [install.win32-x86] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.15/synApps/support/areaDetector/ADCore/ADApp/tiffSupport/tiffSrc'
D:/synApps-testing/VS10/3.15/base-3.15.4/configure/RULES_DIRS:88: recipe for target 'tiffSrc.install' failed
make: *** [tiffSrc.install] Error 2
```

There doesn't appear to be an easy to way to make a similar change for the tiff code.

MarkRivers, 2016-09-01:
I have fixed ADApp/nexusSrc, ADApp/tiffSupport/jpegSrc, and ADApp/tiffSupport/tiffSrc using the same mechanism used in ADApp/netCDFSrc.  

I am now able to build ADCore and ADExample both statically and dynamically with Visual Studio 2010 on both 3.14.12.5 and 3.15.4.

On mingw I can build ADCore statically.  However, when I build ADExample it has problems with the HDF5 library that was built with Visual Studio.  It has lots of errors like these:

/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__create_chunk_mem_map_hyper]+0x108): undefined reference to `@__security_check_cookie@4'
/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__create_chunk_mem_map_hyper]+0x18b): undefined reference to `@__security_check_cookie@4'
/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__create_chunk_mem_map_hyper]+0x27d): undefined reference to `__allmul'
/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__create_chunk_mem_map_hyper]+0x319): undefined reference to `@__security_check_cookie@4'
/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__create_chunk_mem_map_hyper]+0x41f): undefined reference to `@__security_check_cookie@4'
/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw-static/libhdf5.lib(hdf5-static.dir/RelWithDebInfo/H5Dchunk.obj):(.text[_H5D__chunk_file_cb]+0xa): undefined reference to `___security_cookie'

When building mingw dynamically it fails in nexusSrc, again because of problems with the HDF5 library. It has errors like these:

make[3]: Entering directory `/home/epics/devel-3.15/areaDetector-2-5/ADCore/ADApp/nexusSrc/O.win32-x86-mingw'
/usr/bin/i686-pc-mingw32-g++ -o NeXus.dll -shared -Wl,--out-implib,libNeXus.dll.a -L/home/epics/devel-3.15/areaDetector-2-5/ADCore/lib/win32-x86-mingw -L/home/epics/devel-3.15/areaDetector-2-5/ADBinaries/lib/win32-x86-mingw        -m32               napi.obj napi5.obj napiu.obj nxdataset.obj nxio.obj nxstack.obj nxxml.obj stptok.obj    -lhdf5 -lszip -lzlib      -lws2_32 -lwinmm 
Creating library file: libNeXus.dll.a
napi5.obj:napi5.c:(.text+0x69): undefined reference to `_H5T_NATIVE_DOUBLE_g'
napi5.obj:napi5.c:(.text+0x6f): undefined reference to `_H5T_NATIVE_DOUBLE_g'
napi5.obj:napi5.c:(.text+0x7b): undefined reference to `_H5T_C_S1_g'
napi5.obj:napi5.c:(.text+0x81): undefined reference to `_H5T_C_S1_g'
napi5.obj:napi5.c:(.text+0x8d): undefined reference to `_H5T_NATIVE_SCHAR_g'
napi5.obj:napi5.c:(.text+0x93): undefined reference to `_H5T_NATIVE_SCHAR_g'
napi5.obj:napi5.c:(.text+0xa6): undefined reference to `_H5T_NATIVE_UCHAR_g'
napi5.obj:napi5.c:(.text+0xac): undefined reference to `_H5T_NATIVE_UCHAR_g'
napi5.obj:napi5.c:(.text+0xb8): undefined reference to `_H5T_NATIVE_SHORT_g'
napi5.obj:napi5.c:(.text+0xbe): undefined reference to `_H5T_NATIVE_SHORT_g'
napi5.obj:napi5.c:(.text+0xca): undefined reference to `_H5T_NATIVE_USHORT_g'
napi5.obj:napi5.c:(.text+0xd0): undefined reference to `_H5T_NATIVE_USHORT_g'
napi5.obj:napi5.c:(.text+0xdc): undefined reference to `_H5T_NATIVE_INT_g'
napi5.obj:napi5.c:(.text+0xe2): undefined reference to `_H5T_NATIVE_INT_g'
napi5.obj:napi5.c:(.text+0xee): undefined reference to `_H5T_NATIVE_UINT_g'

I have updated the master branch.

MarkRivers, 2016-09-01:
I also fixed ntndArrayConverterSrc.  It needed fixes to correctly build its library for dynamic builds on Windows.

kmpeters, 2016-09-01:
I updated to commit 62b9d82d39494224301f2f9625958ab13e9f9048.  The errors that I documented in this issue have been resolved.  I can now build tiffSupport dynamically with Visual Studio 2010 and 2015 on both base-3.14.12.5 and base-3.15.4.  I can also build tiffSupport statically if I delete the snprintf definition (https://github.com/areaDetector/ADCore/issues/187).

I'm still not able to build nexusSrc after upgrading to 62b9d82d39494224301f2f9625958ab13e9f9048. These problems remain: https://github.com/areaDetector/ADCore/issues/188

MarkRivers, 2016-09-06:
The problem with snprintf was fixed in commit https://github.com/areaDetector/ADCore/commit/729dd626d53f82848e176f9218ce31befd9f80e3 and @kmpeters verified that it fixed his problem with Visual Studio 2015.

closed 2016-09-06 by MarkRivers