areaDetector/ADCore #193: areaDetector licensing

MarkRivers, 2016-09-21:
It has been pointed out that most areaDetector repositories do not include any license file.  We need to fix this.  I have prepared a draft license file and placed it in ADCore/LICENSE (on the no_ADBinaries branch for now).  This is derived from the EPICS base LICENSE file, but removing things that I do not think apply.

Please take a look at this file and share your comments.

Once we agree on this I will put it in the top-level of each repository in areaDetector that does not already contain any license file.

prjemian, 2016-09-21:
As noted to the previous commit (04923e0c260b2156ff33b0662adfeed427260a79):

1. in sections 3 and 4, there are two instances (each) where the text *must this* should be changed to *must include this*
1. It is customary to include and require a copyright notice in the source code or documentation of copies of a software project.  Don't you want that requirement in the LICENSE, as well?

MarkRivers, 2016-09-22:
I talked to Michael Davidsaver, and he suggested using the EPICS V4 license as the model.  I took their LICENSE file and modified it to reflect the main institutions contributing to areaDetector.  Please look at the new version of LICENSE in the no_ADBinaries branch and send comments.

ulrikpedersen, 2016-09-22:
I think this is a good idea, in particular in combination with moving the 3rd party dependencies (with all sorts of different licenses) out of ADCore...

The EPICS V4 licence seems appropriate to me.

MarkRivers, 2016-10-06:
Are there any more comments on the proposed license file?  
@mp49 should we include ORNL in the copyright list?  If so can you send me the boilerplate for that?   
@ajgdls should we include Observatory Sciences in the copyright list?  If so can you send me the boilerplate text for that?

prjemian, 2016-10-06:
ok

mp49, 2016-10-06:
Hi,

Yes, the ORNL copyright should be:

Copyright (c) 2008-2016, UT-Battelle LLC

(or 2013-2016 would be more accurate).

closed 2016-10-06 by MarkRivers