areaDetector/ADCore #195: [no_ADBinaries branch] Windows link errors in pluginSrc

kmpeters, 2016-09-30:
I get the following link errors for win32-x86 builds of the no_ADBinaries branch of ADCore:
```
make[5]: Entering directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp/pluginSrc/O.win32-x86'
link /nologo /subsystem:windows /dll /LTCG /incremental:no /opt:ref /release        /implib:NDPlugin.lib /out:NDPlugin.dll         fft.obj NDArrayRing.obj NDFil
eTIFF.obj NDFileJPEG.obj NDFileNull.obj NDPosPlugin.obj NDPosPluginFileReader.obj NDFileNexus.obj NDFileHDF5.obj NDFileHDF5Dataset.obj NDFileHDF5AttributeDatase
t.obj NDFileHDF5LayoutXML.obj NDFileHDF5Layout.obj NDFileNetCDF.obj NDPluginAttribute.obj NDPluginCircularBuff.obj NDPluginColorConvert.obj NDPluginDriver.obj N
DPluginFFT.obj NDPluginFile.obj NDPluginOverlay.obj NDPluginOverlayTextFont.obj NDPluginProcess.obj NDPluginROI.obj NDPluginROIStat.obj NDPluginStats.obj NDPlug
inStdArrays.obj NDPluginTimeSeries.obj NDPluginTransform.obj NDFileMagickStub.obj       ..\\..\\..\\lib\\win32-x86\\ADBase.lib  D:\\synApps-testing\\VS10\\3.14\
\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x86\\netCDF.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x
86\\NeXus.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\asyn\\lib\\win32-x86\\asyn.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x8
6\\recIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\softDevIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\
\miscIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\rsrvIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\dbt
oolsIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\asIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\dbIoc.
lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\registryIoc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\dbStatic
Ioc.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\ca.lib  D:\\synApps-testing\\VS10\\3.14\\base-3.14.12.5\\lib\\win32-x86\\Com.lib  D:\\
synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x86\\hdf5.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetecto
r\\ADSupport\\lib\\win32-x86\\tiff.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x86\\jpeg.lib  D:\\synApps-testin
g\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x86\\szip.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\
lib\\win32-x86\\zlib.lib  D:\\synApps-testing\\VS10\\3.14\\synApps\\support\\areaDetector\\ADSupport\\lib\\win32-x86\\xml2.lib ws2_32.lib user32.lib gdi32.lib o
leaut32.lib
Com.lib(Com.dll) : error LNK2005: "public: class epicsThreadRunable & __thiscall epicsThreadRunable::operator=(class epicsThreadRunable const &)" (??4epicsThrea
dRunable@@QAEAAV0@ABV0@@Z) already defined in NDPosPlugin.obj
Com.lib(Com.dll) : error LNK2005: "public: __thiscall epicsThreadRunable::epicsThreadRunable(class epicsThreadRunable const &)" (??0epicsThreadRunable@@QAE@ABV0
@@Z) already defined in NDPosPlugin.obj
   Creating library NDPlugin.lib and object NDPlugin.exp
NDPosPlugin.obj : error LNK2001: unresolved external symbol _pasynTrace
NDPlugin.dll : fatal error LNK1120: 1 unresolved externals
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_BUILD:300: recipe for target 'NDPlugin.dll' failed
make[5]: *** [NDPlugin.dll] Error 1120
make[5]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp/pluginSrc/O.win32-x86'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_ARCHS:64: recipe for target 'install.win32-x86' failed
make[4]: *** [install.win32-x86] Error 2
make[4]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp/pluginSrc'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'pluginSrc.install' failed
make[3]: *** [pluginSrc.install] Error 2
make[3]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore/ADApp'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'ADApp.install' failed
make[2]: *** [ADApp.install] Error 2
make[2]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore.install
' failed
make[1]: *** [D:/synApps-testing/VS10/3.14/synApps/support/areaDetector/ADCore.install] Error 2
make[1]: Leaving directory 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector'
D:/synApps-testing/VS10/3.14/base-3.14.12.5/configure/RULES_DIRS:87: recipe for target 'D:/synApps-testing/VS10/3.14/synApps/support/areaDetector.install' faile
d
make: *** [D:/synApps-testing/VS10/3.14/synApps/support/areaDetector.install] Error 2
```

The error occurs with all combinations of VS2010, VS2015, base-3.14.12.5, and base 3.15.4

kmpeters, 2016-09-30:
I'm using the following versions:
areaDetector: 4d7098e2ebd97bb5ce4ecbf1e97e1ddc5cd7a480
ADCore: 9392054cacdcbc2e06105d7c58a51106c005de73
asyn: 1ce81fa680dbe928eec365efae03a653f3f4edec

MarkRivers, 2016-09-30:
I was able to reproduce that error when building ADCore dynamically.  I have fixed NDPosPlugin.cpp to handle dynamic builds on Windows.  @ulrikpedersen did you test building ADCore dynamically on Windows?  It should not have worked.

ulrikpedersen, 2016-10-03:
> I was able to reproduce that error when building ADCore dynamically. I have fixed NDPosPlugin.cpp to handle dynamic builds on Windows. @ulrikpedersen did you test building ADCore dynamically on Windows? It should not have worked.

No I did not. I thought I did at my very first go at it - but our EPICS base is static and thus so is ADCore. I only built DLLs of the 3rd party libs in ADSupport. 

ADCore and EPICS base are static builds in our environment. I do not have an alternative dynamic build of EPICS base around.

MarkRivers, 2016-10-05:
I had to make a few iterations of NDPosPlugin.cpp to get it to build dynamically on both Visual Studio and mingw, but it is working now.

kmpeters, 2016-10-05:
I'm able to build the no_ADBinaries branches of ADCore (1ba29d9869c9708ae2e109821530545c8e8c758c) and ADSupport (7f24922b7b88d44a0ca488970785f3b1c98495e8) using both VS2010 and VS2015 against both base-3.14.12.5 and base-3.15.4.

MarkRivers, 2016-10-06:
I am closing this issue since I believe it is fixed.

closed 2016-10-06 by MarkRivers