areaDetector/ADCore #197: Problem using NDPluginPva

MarkRivers, 2016-10-06:
I am trying to test the NDPluginPva plugin for the first time.  It is building and loading OK.  I have created an medm screen, which is correctly displaying the PvName_RBV field.  

However, when I enable the plugin and start the simDetector the ArrayRate is 0, and the DroppedArrays increases by 1 on each array callback.

[image]

pvinfo on a Linux machine finds the V4 PV:

    corvette:/usr/local/lib>pvinfo 13SIM1:Pva1:NTNDArray
    CHANNEL  : 13SIM1:Pva1:NTNDArray
    STATE    : CONNECTED
    ADDRESS  : 164.54.160.82:34841
    epics:nt/NTNDArray:1.0
    union value
        boolean[] booleanValue
        byte[] byteValue
        short[] shortValue
        int[] intValue
        long[] longValue
        ubyte[] ubyteValue
        ushort[] ushortValue
        uint[] uintValue
        ulong[] ulongValue
        float[] floatValue
        double[] doubleValue
    codec_t codec
        string name
        any parameters
    long compressedSize
    long uncompressedSize
    dimension_t[] dimension
        dimension_t
            int size
            int offset
            int fullSize
            int binning
            boolean reverse
    int uniqueId
    time_t dataTimeStamp
        long secondsPastEpoch
        int nanoseconds
        int userTag
    epics:nt/NTAttribute:1.0[] attribute
        epics:nt/NTAttribute:1.0
            string name
            any value
            string[] tags
            string descriptor
            int sourceType
            string source
    string descriptor
    alarm_t alarm
        int severity
        int status
        string message
    time_t timeStamp
        long secondsPastEpoch
        int nanoseconds
        int userTag
    display_t display
        double limitLow
        double limitHigh
        string description
        string format
        string units

However, pvget appears to show that there is no data or timestamp?

    corvette:/usr/local/lib>pvget -r "value,alarm,timeStamp" 13SIM1:Pva1:NTNDArray
    13SIM1:Pva1:NTNDArray
    structure 
        union value
            (none)
        alarm_t alarm NO_ALARM NO_STATUS <no message>
        time_t timeStamp <undefined> 0

@brunoseivam can you tell me what I am doing wrong?

MarkRivers, 2016-10-06:
I BlockingCallbacks from No to Yes and then it seems to work OK.  But when BlockingCallbacks is No it drops every arrray.

brunoseivam, 2016-10-06:
Hi @MarkRivers, I could reproduce it here. `NDPluginPva` never called the `start()` method after instantiation, so `processTask` never ran and the queue was never emptied. I will make a pull request that fixes this issue.

MarkRivers, 2016-10-06:
Thanks, I found and fixed the problem the same way you did before I saw your comment.  It is fixed on the no_ADBinaries branch which will be merged with master soon.

MarkRivers, 2016-10-06:
Closing this issue, fixed.

closed 2016-10-06 by MarkRivers