areaDetector/ADCore #204: Add new features to pvaDriver

MarkRivers, 2016-10-10:
I have added some new features to pvaDriver:

- Implements ADAcquire, ADImageMode, ADNumImages.  So acquisition can be started and stopped, and can be done in Single, Multiple, or Continuous mode

- New PvName PV allows changing the EPICS V4 PV name at runtime, so it can switch between receiving arrays from different servers

- Added medm screens for pvaDriver and NDPluginPVA

- Implements autosave

I should have done this in a branch with a pull request, but I did it on the master branch.  @brunoseivam can you take a look and see if it looks OK?  I have tested the new features and they seem to work OK.

brunoseivam, 2016-10-10:
Looks good! I tested locally and it seems to work fine here too

MarkRivers, 2016-10-10:
OK, thanks.

closed 2016-10-10 by MarkRivers