areaDetector/ADCore #206: Travis build fails missing epicsString.h

ulrikpedersen, 2016-10-10:
The build currently fails on travis-ci.org https://travis-ci.org/areaDetector/ADCore/builds/166407456

The error:
```
/usr/bin/g++ -c  -D_POSIX_C_SOURCE=199506L -D_POSIX_THREADS -D_XOPEN_SOURCE=500           -D_X86_64_  -DUNIX  -D_BSD_SOURCE -Dlinux  -D_REENTRANT   -g   -Wall     --coverage  -m64     -fPIC -MMD -I. -I../O.Common -I. -I.. -I../../../include/os/Linux -I../../../include        ../NDAttribute.cpp 
../NDAttribute.cpp:11:25: fatal error: epicsString.h: No such file or directory
```

Travis installs EPICS Base from NSLS-II debian repos (libepics3.14.12.3). Off the top of my head I don't know why epicsString.h is not found.

Asyn and ADSupport builds have completed before getting to this point.

MarkRivers, 2016-10-10:
asynManager.h includes <epicsString.h>, so it must have found it when building asyn.

ulrikpedersen, 2016-10-10:
Ah, I think it went wrong with commit ca2f2d8ceeb614519acfe0e2059d08b6e60291ff When building ADCore stand-alone (without the top-level areaDetector module) we still use RELEASE.local to define dependencies like EPICS base. In the above commit the usual include statements have been replaced with a single include from the top-level module. 

Can we retain the `--include $(TOP)/RELEASE.local` statement, please?

MarkRivers, 2016-10-10:
I put `--include $(TOP)/RELEASE.local` and `--include $(TOP)/configure/RELEASE.local` back in ADSupport, ADCore and ADExample configure/RELEASE files.  I them removed from top-level RELEASE_LIBS_INCLUDE and RELEASE_PRODS_INCLUDE.

MarkRivers, 2016-10-10:
Travis is now failing because it cannot find NDFileHDF5.h when building HDF5PluginWrapper.cpp.  This is probably because you have not defined WITH_HDF5=YES in ADCore/configure/RELEASE.local.

PR #207 (2016-10-11): Travis tweaks
Update the pluginTest/Makefile to use the new `WITH_*` macro to enable/disable 3rd party dependencies. 

Make the travis-ci configuration define WITH_HDF5=YES and WITH_BOOST=YES so that the HDF5 plugin and Boost-based unittest suite is built.

This fixes #206 (although the original issue was fixed already, this PR takes care of the secondary issue [mentioned in the thread](https://github.com/areaDetector/ADCore/issues/206#issuecomment-252762884) )
closed 2016-10-11 by MarkRivers