areaDetector/ADCore #212: New ImageJ plugin to define ROIs, Overlays and detector readout regions

MarkRivers, 2016-11-01:
A new ImageJ plugin should be created that will allow defining the following:

- Overlay shapes for NDPluginOverlay
- ROI shapes for NDPluginROI
- Readout regions for detectors that support ROI readout

This ImageJ plugin should be independent of the EPICS_AD_Viewer plugin.  It should work on any ImageJ window, which could be the EPICS_AD_Viewer window, but might also be a window created by reading a file, etc.

The plugin will allowing defining 3 different prefixes: one for ROI, one for overlay, one for detector.

ROI and detector plugins will only work with rectangular ImageJ ROIs

Overlays will work with rectangular, line (becomes cross), or ellipsoidal overlay shapes.

The use must press a button to send the definition to the appropriate destination PVs.

argonnexraydetector, 2016-11-01:
Mark

Would you like me to take a stab at the new IJ plugin?
I have decent Java chops.

Tim Madden

MarkRivers, 2016-11-01:
Tim,

I think I can handle this one, since Chris Roehrig has already written most of what I need.  His was combined with the EPICS_AD_Viewer plugin.  I just want to separate out the code that does the ROIs, detector readout region, and overlays (new) into a separate plugin.

However, since you are offering I would really like you to tackle #134.  This is a bigger project, and one where good Java skills will be needed.  We now have the NDPluginPva plugin that exports NDArrays as EPICS V4 NTNDArrays.  So it would be really nice to have a new ImageJ plugin that receives that structured data, rather than having to convert to waveform records, etc.  There is a complete Java implementation of EPICS V4, so in principle I think it should be straightforward.  We will then have access to all of the NDAttribute metadata in ImageJ.  

As a further project one could imaging having ImageJ do some processing and then convert the data back into an NTNDArray and push the processed array back into an IOC.  This would use the Java PVA server and the existing pvaDriver in the IOC.

argonnexraydetector, 2016-11-01:
OK I will take a look at 143.

T

argonnexraydetector, 2016-11-02:
Mark

I will give *134 a shot. 
T

MarkRivers, 2016-12-02:
I have completed this task.  There is a new ImageJ plugin in ADCore/Viewers/ImageJ/EPICS_areaDetector/EPICS_AD_Controller.java.  It is documented here:

[EPICS_AD_Controller](http://cars.uchicago.edu/software/epics/areaDetectorViewers.html#ImageJController)

It is also in the [RELEASE.md](https://github.com/areaDetector/ADCore/blob/master/RELEASE.md)

To use the Transform record you must run the master branch from Github, because I added a new feature.  It now sets the NDArraySizeX_RBV and NDArraySizeY_RBV PVs, like the ROI plugin does.  This is very convenient and saves a lot of bookkeeping in the ImageJ plugin, because those PVs account for the fact that the X and Y axes may have been swapped, and keeps X and Y correct when using RGB1, RGB2, or RGB3 arrays.

closed 2016-12-02 by MarkRivers