areaDetector/ADCore #214: PVAttribute.cpp can cause multiple monitor subscriptions on a single PV

mp49, 2016-11-02:
Hi,

I've come across an monitor issue with PVAttribute.cpp. When the PV is first connected we set up a monitor subscription with the server in the connectCallback() function. If the PV disconnects and reconnects while the areaDetector IOC is still up then connectCallback() is called again, and on a connection event sets up another monitor subscription. This results in two monitor subscriptions and the callback function being called twice for every monitor event from the server. The original subscriptions seems to persist in the client library and is reestablished on connection. 

In the CA reference manual I found this sentence in the section for ca_create_subscription():

Each call to this function consumes resources in the client library and potentially a CA server until one of ca_clear_channel() or ca_clear_subscription() is called.

I'm not sure if the best solution is to attempt to clear the subscription before calling ca_create_subscription() again, or to figure out if one is already setup and if it is don't call ca_clear_subscription().

Shall I pass this by tech-talk?

Cheers,
Matt

MarkRivers, 2016-12-05:
Hi Matt,
I suspect you need to create another monitor subscription, because on the reconnect the PV could be from another host?  But am not a CA expert so I think you should pass it  by tech-talk.
Thanks,
Mark

MarkRivers, 2017-01-27:
@mp49 have you had a chance to look at this?  It would be good to get it fixed for R2-6.

mp49, 2017-01-27:
Hi Mark, I forgot all about this. I'll send an email to tech talk about it, but I'm not sure I can do much in the next week.

mp49, 2017-01-27:
I've tested moving the PV that we monitor to a different host, and the CA client library automatically handles this. That works fine both with camonitor and PVAttribute.

I'm not sure about changing data types though (see Andrew Johnsons comment on tech talk):

> I was able to fix this in the same was as camonitor.c handles this,
> by using a flag to indicate that we’ve already connected (and set up
> a monitor) to a PV. Is this the most appropriate method to handle
> this? In the CA reference manual I see that there are functions like
> ca_clear_channel() and ca_clear_subscription() that we could possibly
> use on a disconnection event.

Given that reconnection happens automatically inside the CA library and
you usually want that to happen, it's generally a bad idea to clear your
subscriptions just because a channel disconnects. However the connection
handler may need to make sure that the channel's native data type hasn't
changed since the last time it connected. If the client's code assumes
that the subscription always uses the channel's native type then the
connection handler will have to clear and resubscribe if the type
changes, but some clients may always subscribe with a specific data
type, in which case they won't need to worry about that.

I think with the existing code this would have worked, but we would end up with a disconnected channel with the old data type, and a connected channel with the new data type. The patch in the branch might break this.

mp49, 2017-01-27:
I've tested switching data types. The connection and monitor seems to be automatically restored, but since there's no new connection it means the data type is wrong. So there's the possibility of losing data if the data type changes (eg. initially setting up a monitor on a longout record, then it changes to a ao record - the PVAttribute subscription will still be of type DBR_LONG).

mp49, 2017-01-27:
I suspect we don't want to support switching data types at runtime due to PV data types changing. It may affect other things downstream, like file writing plugins. 

In which case the patch on issue-214-PVAttribute-Bug branch should be sufficient.

MarkRivers, 2017-01-28:
Closed by #242

closed 2017-01-28 by MarkRivers