areaDetector/ADCore #219: Multiple instances of NDPluginPva call startPVAServer too many times

brunoseivam, 2016-11-09:
`startPVAServer` should be called just once in each process. From tech-talk:

Hi all,

I am at ESS running an IOC with 2 pva plugins both connected to the simDetector:

```
# Create a simDetector driver
# simDetectorConfig(const char *portName, int maxSizeX, int maxSizeY, int dataType,
#                   int maxBuffers, int maxMemory, int priority, int stackSize)
simDetectorConfig("DMSCSIMDET", 500, 500, 1, 0, 0)
dbLoadRecords("simDetector.template", "P=DMSC:, R=CAM-, PORT=DMSCSIMDET, ADDR=0, TIMEOUT=1")
#create pva plugin
NDPvaConfigure(DMSCPVA, 20, 0, "DMSCSIMDET", 0, "DMSC:IMAGE", 0)
VERSION : pvAccess Server v4.1.2
PROVIDER_NAMES : local
BEACON_ADDR_LIST :
AUTO_BEACON_ADDR_LIST : 1
BEACON_PERIOD : 15
BROADCAST_PORT : 5076
SERVER_PORT : 5075
RCV_BUFFER_SIZE : 16384
IGNORE_ADDR_LIST:
STATE : INITIALIZED
dbLoadRecords("NDPva.template", "P=DMSC:, R=PVA-, PORT=DMSCPVA, ADDR=0, TIMEOUT=1, NDARRAY_PORT=DMSCSIMDET")
#create pva plugin
NDPvaConfigure(DMSCPVA2, 20, 0, "DMSCSIMDET", 0, "DMSC:IMAGE2", 0)
2016-11-08T16:33:54.789 Using dynamically assigned TCP port 52363.
VERSION : pvAccess Server v4.1.2
PROVIDER_NAMES : local
BEACON_ADDR_LIST :
AUTO_BEACON_ADDR_LIST : 1
BEACON_PERIOD : 15
BROADCAST_PORT : 5076
SERVER_PORT : 52363
RCV_BUFFER_SIZE : 16384
IGNORE_ADDR_LIST:
STATE : INITIALIZED
dbLoadRecords("NDPva.template", "P=DMSC:, R=PVA2-, PORT=DMSCPVA2, ADDR=0, TIMEOUT=1, NDARRAY_PORT=DMSCSIMDET")
```

As you can see it start 2 pvAccess servers.

Monitoring one array sees multiple instance of the pv:
```
pvget -m DMSC:IMAGE2
[ChannelRequesterImpl] message(More than one channel with name 'DMSC:IMAGE2' detected, connected to: 194.47.240.7:52834, ignored: 194.47.240.7:5075, warning)
DMSC:IMAGE2
structure
    union value
        (none)
```

Listing the servers shows the 2 pvAccess servers (Which yo):
```
pvlist
GUID 0x7EE8215800000000799BA728, version 1: tcp@[x.x.x.x:5075]
GUID 0x7EE82158000000008AD62829, version 1: tcp@[x.x.x.x:52834]
```

Listing pvs on each server shows that both are publishing the same PVs.
```
pvlist 0x7EE8215800000000799BA728
DMSC:IMAGE
DMSC:IMAGE2

pvlist 0x7EE82158000000008AD62829
DMSC:IMAGE
DMSC:IMAGE2

```
Has anyone tried running 2 NDPluginPvas in the same IOC before? Are you supposed to start multiple pvAccess servers in the same IOC?

Note that we are using our own build system, and it might be an issue there.

Regards,
Alexander Söderqvist

brunoseivam, 2016-11-09:
As I stated in the tech-talk list, I thought of two solutions to this issue:

One solution is to have `startPVAServer` as a standalone function to be called after all instantiations of `NDPluginPva`, maybe using [1].

Another one, that I like more, is to have a static member variable in the `NDPluginPva` class that records if `startPVAServer` was already called. That works for multiple `NDPluginPva` instances but most likely won't work if other code not pertaining to the PVA Plugin also wants to start a server.

[1] https://github.com/epics-base/pvAccessCPP/blob/master/src/ioc/PVAServerRegister.cpp

brunoseivam, 2016-11-09:
OK so here are the two solutions. The second one is cleaner but requires changes to IOCs that depend on the old behavior. We should pick one of them, and currently I am leaning towards the second one, i.e., the user having to explicitly call `startPVAServer` after `iocInit()`.

MarkRivers, 2016-11-09:
Is it required that the call to startPVAServer be after the call to iocInit?  What if we want to test outside the context of an IOC?

mp49, 2016-11-14:
startPVAServer can be called before or after iocInit(). 

We've been using it in a standalone program for a while:
https://github.com/kasemir/EPICSV4Sandbox/blob/master/neutronsDemoServer/src/neutronServerMain.cpp

commit 134b22da (2016-11-22): startPVAServer won't be called by the plugin anymore. Fixes #219
commit 76a6e99e (2016-11-22): Merge pull request #222 from brunoseivam/fix-ndpva-startpvaserver2

startPVAServer won't be called by the plugin anymore. Fixes #219
closed 2016-11-22 by MarkRivers