areaDetector/ADCore #223: ADDriver should have additional parameters like SerialNumber and FirmwareVersion

MarkRivers, 2016-11-10:
Many drivers are implementing parameters like SerialNumber and FirmwareVersion independently.  A commonly used set of parameters should be added to ADDriver.  An initial list might be the following:

In ADDriver:
  - SDKVersion The version number of the vendor's SDK
  - FirmwareVersion The firmware version in the detector
  - SerialNumber Detector serial number

This should go in the asynNDArrayDriver base class
  - ADCoreVersion The version of ADCore that the driver or plugin was built with
  - DriverVersion The version of this driver or plugin

Please comment if there are other parameters that should be added to this list, or some deleted.  This is a relatively large job, since OPI screens need to be changed.  Existing drivers should be changed to use these new parameters, although that is not required and can be done as time permits.

MarkRivers, 2016-12-04:
I have completed this task.  Since this was quite straightforward I have committed to master.  Most of the changes involved adding new readonly parameters to asynNDArrayDriver and ADDriver.  The one change that involved a bit of code was adding NDPluginExecutionTime to NDPluginDriver, but this is also quite simple.

Here are the applicable notes from RELEASE.md

### NDPluginDriver, NDPluginBase.template, NDPluginBase.adl
* Added new parameter NDPluginExecutionTime and new ai record ExecutionTime_RBV.  This gives the execution time  in ms the last time the plugin ran.  It works both with BlockingCallbacks=Yes and No.  It is very convenient for measuring the performance of the plugin without having to run the detector at high frame rates.

### asynNDArrayDriver, NDArrayBase.template, NDPluginBase.adl, ADSetup.adl, all plugin adl files
* Added 2 new parameters: NDADCoreVersion, NDDriverVersion and new stringin records ADCoreVersion_RBV and DriverVersion_RBV.  These show the version of ADCore and of the driver or plugin that the IOC was built with.  Because NDPluginBase.adl grew larger all of the other plugin adl files have changed their layouts.

### ADDriver, ADBase.template, ADSetup.adl
* Added 3 new parameters: ADSerialNumber, ADFirmwareVersion, and ADSDKVersion and new stringin records  SerialNumber_RBV, FirmwareVersion_RBV, and SDKVersion_RBV. These show the serial number and firmware  version of the detector, and the version of the vendor SDK library that the IOC was built with. Because ADSetup.adl grew larger all driver adl files need to change their layouts.  This has been done for ADBase.adl in ADCore.  New releases of driver modules will have the changed layouts.

MarkRivers, 2016-12-04:
I have updated the driver code and medm screens for the following drivers to use the new parameters:
- ADSimDetector
- ADCSimDetector
- NDDriverStdArrays
- pvaDriver
- ADPointGrey
- ADProsilica

Other drivers still need to be done.

MarkRivers, 2016-12-04:
Things still to be done:
- New screen shots of all plugins in ADCore and all drivers with new version of ADSetup.adl

hinxx, 2016-12-05:
I can try to do it for ADAndor since we have one. Been rather busy with other stuff lately hence I hope I can do this for 2-6 before release.

MarkRivers, 2017-01-05:
I have done ADAndor and committed to Github.

MarkRivers, 2017-01-27:
I am closing this issue.  The work that needs to be done in ADCore is complete.  Many (but not all) of the detector drivers and detector medm screens have been updated to use the new features.  But drivers that have not been updated will still work, just with ugly medm screens and perhaps redundant parameters and records.

closed 2017-01-27 by MarkRivers