areaDetector/ADCore #224: Proposal to move pvaDriver into its own repository

MarkRivers, 2016-11-13:
When Bruno wrote pvaDriver we decided to put it in ADCore.  Its example IOC is in ADExample.  When I recently added the NDDriverStdArrays driver, I also originally put it in ADCore, and the example IOC in ADExample.  I then decided this was not a good idea, and I moved NDDriverStdArrays to its own repository.

I now think that all drivers should have their own repositories, complete with the example IOC.  Thus I propose to do the following:

- Create a new pvaDriver respository.  Move the pvaDriverSrc from ADCore to that new repository, preserving commit history.

- Move ADExample/iocs/pvaDriverIOC to the new repository.  Need to preserve commit history?

- Take the remaining ADExample repository and split it into 2 new repositories: ADSimDetector and ADCSimDetector.

This way all drivers have their own repositories, complete with the example IOC.  It simplifies ADCore and removes ADExample.

Comments?

ulrikpedersen, 2016-11-14:
I agree completely. In fact this is how we have done drivers and example IOCs internally at DLS for several years now. :+1: 

One question: wouldn't it be appropriate to keep the 2 simulation drivers together in a single repo?

mp49, 2016-11-14:
That sounds fine to me too.
Sounds fine to have the two Sim drivers in their own module.

brunoseivam, 2016-11-18:
Should NDPluginPva also be moved together with pvaDriver?

MarkRivers, 2016-11-18:
No, I think the plugin is fine where it is.  I want to keep dirvers in their own repositories with an example IOC, but commonly used plugins that don't depend on unusual 3'rd party software should be in ADCore.

MarkRivers, 2016-11-22:
I have done the following:

- Created a new pvaDriver repository.  This contains all of the pvaDriver code, databases, opi screens and documentation that were previously in ADCore.  It also contains the example IOC that was previously in ADExample.  Commit history for all important files has been preserved.  Tagged R1-0 of this repository.
- Removed all of the pvaDriver files from ADCore.
- Split ADExample into 2 new repositories: ADSimDetector and ADCSimDetector.  ADExample is now marked as obsolete.  Tagged each of these repositories R2-3 to continue the release numbering they had when they were in ADExample.

The remaining task is to add these new modules as submodules of the top-level areaDetector, change the top-level Makefile, top-level configure/EXAMPLE_RELEASE.local, and top-level documentation files.

MarkRivers, 2016-11-22:
I have completed the tasks for the top-level areaDetector directory, so I am closing this issue.

closed 2016-11-22 by MarkRivers

MarkRivers, 2016-11-22:
I found minor problems in the st.cmd files in ADSimDetector, ADCSimDetector and pvaDriver.  I fixed them and re-tagged with the tags listed above.