areaDetector/ADCore #227: Deficiency in AsynPortClientContainer class

MarkRivers, 2016-11-26:
There is a problem with ADApp/pluginTests/AsynPortClientContainer.  It hardcodes the asyn "addr" field to 0 in the write() and read() methods.  This means it cannot be used to fully test plugins like NDPluginAttribute, NDPluginOverlay, NDPluginROIStats, and NDPluginTimeSeries because these plugins support multiple addresses, but it can only write to the first address (0).

The AsynPortClient class contains an std::map based on the paramName to create a new asynInt32Client, etc.  However, it needs to have a map that is based on both the paramName and the addr, since a separate asynInt32Client is needed for each (paramName, addr) combination.

ulrikpedersen, 2016-12-02:
Yes that does seem reasonable.

CC @ajgdls (lets discuss next week)

ulrikpedersen, 2017-01-25:
@ajgdls  can you take a quick look at this please and see if its easy to solve?

ajgdls, 2017-01-25:
Hi @ulrikpedersen  @MarkRivers , I can suggest two ways of doing this:

1) The constructor takes the address number, and then all methods use that address number.  Multiple objects could be created if multiple addresses need to be accessed.

2) The class holds a vector of maps.  Each read/write method takes an address value (with default of 0 so no existing code fails).  When a new address is specified the map object is created and added to the vector, and parameters added to the map as they are currently.  Then the address is the vector index and the parameter is the map index.

I like option 2.  Any comments?

MarkRivers, 2017-01-27:
Fixed with #238

closed 2017-01-27 by MarkRivers