areaDetector/ADCore #245: NDPluginAttribute alternative

blaz-kranjc, 2017-02-14:
This is a proposal for including a plugin we (@dls-controls) wrote [1], I hope opening an issue is the right workflow.

The plugin is similar to the NDPluginAttribute but with a few different features we needed for our use case:
- Attributes are not selected by name but rather a list of first N attributes is extracted from the first NDArray of an acquisition;
- Circular buffer is used;
- The attribute values are written to the waveforms periodically (1Hz), decoupled from the callbacks

The current use case for the plugin here is plotting the data we receive from the Pandabox. The driver for Pandabox driver exposes data as attributes. The attributes are not know up front but can change between experiments. We wanted a simple plugin that would expose this attributes so we can plot them in an EDM screen [2].

Would you be interested in including this kind of plugin given the fact that NDPluginAttribute already exists?

[1] https://github.com/blaz-kranjc/NDAttrPlotPlugin
[2] https://raw.githubusercontent.com/blaz-kranjc/NDAttrPlotPlugin/master/Screenshot.png

mp49, 2017-02-14:
Does the Pandabox driver already expose the parameters as individual PVs, or do you only have the attributes to work with? If the PVs for each attribute are already available then using a set of compression database records might have been easier (or even just archived data). Or, do you need to somehow save this data in files? 

However, since it's already written this does look useful. Except I would suggest using the EPICS epicsRingPointer class instead of writing your own circular buffer.

blaz-kranjc, 2017-02-14:
This plugin was developed because the pandabox doesn't expose this data in PVs, the data is saved to HDF5 files via the HDF5 plugin. This plugin just enables live plotting for diagnostic purposes.

I'll replace my custom implementation with the standard ring buffer and create a pull request.

mp49, 2017-02-14:
That sounds fine. 

I guess a natural extension to this plugin (and NDPluginAttribute) is the ability to do some statistics on the data (like plotting the mean of every 10 data points). Maybe someone has already done that for plugins for ADC data though?

blaz-kranjc, 2017-02-14:
The epicsRingPointer's interface seems to be really restrictive in terms of accessing the underlying buffer (no accessors, only push/pop, no easy way to get a copy). Combined with the fact that the epicsRingPointer is not copy constructable that would mean that I would have to recreate the whole buffer on copy to epics waveform (which is definitely not something we should be doing).

I can move my circular buffer implementation in the plugin as a private class to not pollute the global namespace if that is the problem.

mp49, 2017-02-14:
I was more concerned about redundant code that we didn't need, but it sounds like it's appropriate in this case.

MarkRivers, 2017-05-07:
Closing because NDAttrPlot has been merged into master and is in R3-0.

closed 2017-05-07 by MarkRivers