areaDetector/ADCore #248: Install common*Makefile?

mdavidsaver, 2017-02-20:
In working on debian packaging for AD modules, I've found myself wishing that some additional files were "installed".  For example, at present the aravisGigE driver has

```
include $(ADCORE)/ADApp/commonDriverMakefile
```

Which is reaching into the source tree (ADApp/) for what is probably intended as an external interface.  For my immediate purposes I've included a copy under as ```$(ADCORE)/configure/commonDriverMakefile```, although I, and perhaps @anjohnson, would prefer a less generic file name.  Perhaps install copies as```configure/ADDriverMakefile``` and ```configure/ADLibraryMakefile```?

mdavidsaver, 2017-02-20:
Just to be clear.  I'm not proposing to chance the source file name, so this would be a compatible change.

MarkRivers, 2017-05-01:
I spoke to @anjohnson about this.  He suggested using the EPICS CFG file mechanism.  This installs files into $(INSTALL_DIR)/cfg.  This is better than installing into configure, because the installed files will be removed with "make uninstall", while they won't if they are installed into configure/.

I have added these lines in ADCore/ADApp/ADSrc:
```
# Install commonDriverMakefile and commonLibraryMakefile in top-level cfg directory.
# Drivers currently look in $(TOP)/ADApp for these files so we leave them there but
# need to install from this directory which requires using vpath.
vpath common%Makefile $(ADCORE)/ADApp/
CFG += commonLibraryMakefile
CFG += commonDriverMakefile
```
The vpath is needed because we want to leave the files where they are for backwards compatibility with existing drivers, and the CFG += lines must be in a source directory that uses RULES, not RULES_DIRS.  Drivers can gradually change from 
```include $(ADCORE)/ADApp/commonDriverMakefile```
to
```include include $(ADCORE)/cfg/commonDriverMakefile```

@anjohnson did not think it was necessary to change the names of the files.

@mdavidsaver let me know if you are OK with this.

MarkRivers, 2017-05-02:
This change is working fine for me with base 3.15.5.  @anjohnson looked at base and nothing significant with the CFG rules seems to have changed since 2013.  However, the ADCore Travis build is failing with this error:
```
make[2]: Leaving directory `/home/travis/build/areaDetector/ADCore/ADApp/Db'
make -C ./ADSrc install 
make[2]: Entering directory `/home/travis/build/areaDetector/ADCore/ADApp/ADSrc'
perl /usr/lib/epics/bin/linux-x86_64/makeMakefile.pl O.linux-x86_64 ../../..
mkdir O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. T_A=linux-x86_64 install
make[3]: Entering directory `/home/travis/build/areaDetector/ADCore/ADApp/ADSrc/O.linux-x86_64'
/etc/epics/configure/RULES_FILE_TYPE:45: ../../../cfg/commonDriverMakefile: No such file or directory
/etc/epics/configure/RULES_FILE_TYPE:45: ../../../cfg/commonLibraryMakefile: No such file or directory
make[3]: *** No rule to make target `../../../cfg/commonLibraryMakefile'.  Stop.
make[3]: Leaving directory `/home/travis/build/areaDetector/ADCore/ADApp/ADSrc/O.linux-x86_64'
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory `/home/travis/build/areaDetector/ADCore/ADApp/ADSrc'
make[1]: *** [ADSrc.install] Error 2
make[1]: Leaving directory `/home/travis/build/areaDetector/ADCore/ADApp'
make: *** [ADApp.install] Error 2
```
The Travis build is using the Debian epics-dev package.  @mdavidsaver do you understand this problem?

mdavidsaver, 2017-05-02:
> @mdavidsaver let me know if you are OK with this.

Yes.

> The Travis build is using the Debian epics-dev package. @mdavidsaver do you understand this problem?

This is from the nsls2 repo, which currently has 3.14.12.3-9.  At a guess, this needs one of these subsequent changes to work:  epics-base/epics-base@88ae947c84b5125778db04ffbca9b69988e75ce2   epics-base/epics-base@5610b0706867745df043b9cb93b029371889cd55

MarkRivers, 2017-05-02:
@mdavidsaver  how would you suggest we handle this?  Use epics-dev and apply patches you referenced as part of the Travis build?  Or should we just build a more recent version of base from scratch in Travis?

mdavidsaver, 2017-05-03:
I'd wait for feedback from @anj.  This may mean a different recipe, or at worst custom rules to install these files.

MarkRivers, 2017-05-03:
I just changes the Travis build configuration to build EPICS base from source rather than using the Debian packages.  I tried both 3.14.12.6 and 3.15.5 and I still get errors when trying to install the cfg directory.  This works fine with 3.15.5 on my own Linux system, so there is something different about the Travis configuration?

MarkRivers, 2017-05-03:
I figured out the problems:

- ADApp/ADSrc/Makefile was defining the locations of commonDriverMakefile and commonLibraryMakefile in the vpath command using the symbol $(ADCORE).  However this symbol was not being defined in a RELEASE file.
- I changed from using $(ADCORE) to using $(TOP).  That fixes the problems when building EPICS_BASE from source, and the build now succeeds.  I tested with 3.15.5, but I am quite sure that 3.14.12.6 will also work.
- However, using the Debian epics-dev package the build fails.  I am quite sure this is because it is using 3.14.12.3, and in that release the build always automatically includes all files in the cfg/ directory even if they don't have names starting with CONFIG or RULES.  That was fixed in 3.14.12.4.  That won't work because commonLibraryMakefile and commonDriverMakefile should only be used when explicitly included.
- I think it is a good idea for the Travis system to build EPICS_BASE from source anyway, since then we can easily test with different versions of base.
- I also plan to have it build and use all modules in ADSupport, so that we test building and using that as well.  That will increase the time for the build, but that is OK.

ulrikpedersen, 2017-05-03:
> I think it is a good idea for the Travis system to build EPICS_BASE from source anyway, since then we can easily test with different versions of base.

This is a good idea. The travis configuration is really due an overhaul - there is a very old and overly ambitious ticket #115 .

anjohnson, 2017-05-03:
I now understand why the `CFG` changes don't work with Base-3.14.12.3 or earlier; in those releases the base/configure/RULES_FILE_TYPE file contains:
```makefile
# Include existing and new $(INSTALL_CFG)/* definitions
#
TOP_CFG_FILES = $(sort $(wildcard $(INSTALL_CFG)/RULES*) \
                       $(wildcard $(INSTALL_CFG)/CONFIG*) \
                       $(addprefix $(INSTALL_CFG)/,$(CFG)))
ifneq ($(TOP_CFG_FILES),)
  include $(TOP_CFG_FILES)
endif
```
At first glance it looks like it only should include files from each RELEASE area's `cfg/` directory whose names start `CONFIG*` or `RULES*`, but that call to `addprefix` also causes it to include all files named in the `CFG` variable too. That part got removed in the modifications that actually made this stuff useful in Base-3.14.12.4 which Benjamin Franksen instigated (i.e. the first of the 2 commits that @mdavidsaver linked to above).

Please make sure your release notes document the new minimum Base requirement.

MarkRivers, 2017-05-03:
Thanks for tracking that down.   I have documented that EPICS base 3.14.12.4 or higher is required for ADCore R3-0.

MarkRivers, 2017-05-04:
I have solved the Travis problems.  The ci/install-packages.sh script can now be configured to build EPICS base or install from Debian (which fails with current Debian package).  It can also be configured to install HDF5, TIFF, XML2, etc. either from packages or use the code in ADSupport.  It is currently configured to use ADSupport for everything since that provides more testing.

closed 2017-05-04 by MarkRivers