areaDetector/ADCore #254: Implement NDPluginFile::readFile() for some file formats

MarkRivers, 2017-02-26:
The NDPluginFile base class has a readFile() method, but this has not yet been implemented for any file formats.  It would be useful to implement it for at least some file formats (e.g. TIFF) to be able to read an NDArray from a file.  This could be used, for example, to read flat field or dark current files in the NDPluginProcess plugin.  It can also be used to "replay" an experiment by looping through a number of files or multiple images in a single file (e.g. netCDF or HDF5).  HDF5 is likely to be quite complex, so may not be worth the effort.

ulrikpedersen, 2017-02-27:
Actually at Diamond we have already implemented an areaDetector driver (so not the readFile() method) which can read in HDF5 files and "replay" them at a user-configurable rate. It includes NDAttributes as well.

It has features to slice the dimensions in various ways - something I think we could only achieve by implementing a full driver rather than just using the readFile() method.

It's on github here: https://github.com/dls-controls/simHDF5Detector

mp49, 2017-02-27:
At the SNS we've also found replaying data very useful. It's the only way we can really test our data acquisition system without the accelerator running, and we can effectively re-do experiments with the real data. However, this doesn't use areaDetector written files so is SNS specific. But this new feature would be useful for our imaging beamlines where we write areaDetector TIFF files.

hinxx, 2017-03-14:
I would also find it useful for taking caget/camonitor or archived raw data that could be interpreted as detector data, for replay purposes and debugging without the actual detector running. 

I'm about to add a mode to simulator detector that will suck in camonitor output from a text file, and send it through the processing chain. 

Having this functionality as a plugin might be the right way to do it, instead.

MarkRivers, 2018-04-13:
I have implemented the readFile() method in the NDFileTIFF driver.  It currently has a number of limitations:
- It only works with mono data, not RGB
- It reads the timeStamp, uniqueID, and epicsTS attributes if they exist in the TIFF file, but it does not read any other attributes.
- It only reads a single file each time the ReadFile PV is written to, it does not support the Stream mode.

The new feature is useful for loading the background or flat fields in the NDPluginProcess plugin from TIFF files.  It can be done as follows:

-	In the Process plugin set the array port to the TIFF plugin, e.g. FileTIFF1.
-	In the TIFF plugin enable ArrayCallbacks
-	Set the path, file name, and file number to the name of your flat field or dark file.
-	Press the ReadFile button in the TIFF plugin.
-	This will load that array into the Process plugin
-	Press the SaveBackground or SaveFlatField buttons in the Process plugin
-	Switch the Process plugin back to getting its data from the detector or other upstream plugin

This mechanism is general which is nice, but it obviously is not terribly convenient for the user.  I will look into how to automate the steps in the processing plugin so it is simpler.

MarkRivers, 2018-04-16:
I have added sseq records to the NDPluginProcess database and medm screens.  This makes it very easy to load the background image or flafield image from a TIFF file.  They automate all of the steps outlined in the previous message, except for settings the path, filename, and file number of the flatfield or background file.

I am closing this issue since NDPluginFile and NDFileTIFF now support the readFile() function.  Other file plugins may add this capability in the future, but since TIFF files can store any datatype and are simple this may be sufficient.

closed 2018-04-16 by MarkRivers