areaDetector/ADCore #256: New plugins to support parallel processing

MarkRivers, 2017-02-26:
I propose to add 2 new plugins.

* NDPluginFanout (or NDPluginRoundRobin, or NDPluginScatter?)
This plugin will have a different implementation of doCallbacksGenericPointer() from that in the asynNDArrayDriver base class.  Rather than calling all registered callback client for each NDArray, it will only call one client for each NDArray.  On each subsequent callback it cycles through the list of registered callback clients.  This allows the plugin to spread the processing load over multiple plugins.  For example, the plugin could spread the load over 4 NDPluginProcess plugins, so each one only processes every 4'th array.  This will increase by a factor of 4 the number of NDArrays that can be processed in a given amount of time, assuming that spare cores are available.  The initial implementation might be a simple round-robin, but a more sophisticated might queue the next NDArray to the client with the fewest NDArrays in its input queue.

*NDPluginFanin (or NDPluginGather?)
All existing plugins register for callbacks from a single NDArray source (driver or other plugin).  This plugin will be different, it can register for callbacks from up to N different NDArray sources, where N is defined when the plugin is created.  This would allow the outputs from the 4 NDPluginProcess plugins in the above example to be recombined into a single stream that would then be sent to a file plugin.  The initial implementation will probably not attempt to re-order the NDArray output according to the UniqueId, but a more sophisticated implementation could use a buffer to do this.  It would need to handle the case when a particular UniqueId never arrives because it was dropped by a plugin.

ulrikpedersen, 2017-02-27:
This is an issue which can be divided up in so many different ways... :-)

Also consider a plugin with a generic pool of worker threads (scaled by how many cores are available). Each incoming NDArray is just passed asynchronously to the pool which picks out an idle worker threads and process() called on an instance of the plugin. I think this would require a slightly different plugin API - one which allow multiple "New Style" plugins to exist without overwriting the same asynParameters (i.e. plugins that don't inherit from asyn port driver - but can be called from asyn port drivers).

I find that often plugins can read relevant parameters at the start of the process() call - do the work using a latched copy of the parameters - and then post the results as parameter updates at the very end. Thus the 'real' processing part could be pulled out to an interface which just takes an NDArray const reference and a list or dictionary of key/value attributes and returns a new NDArray and an updated key/value attribute list.

This would require enough interface changes (although it could be done with backwards compatibility) to require a major release like 3-0.

MarkRivers, 2017-03-06:
> This is an issue which can be divided up in so many different ways... :-)
Also consider a plugin with a generic pool of worker threads (scaled by how many cores are available). Each incoming NDArray is just passed asynchronously to the pool which picks out an idle worker threads and process() called on an instance of the plugin. I think this would require a slightly different plugin API - one which allow multiple "New Style" plugins to exist without overwriting the same asynParameters (i.e. plugins that don't inherit from asyn port driver - but can be called from asyn port drivers).

I think both scatter/gather and a thread pool can be useful.
* ScatterGather pros:
  * The plugins that are being called in the round-robin are independent, so they don't need to have the same parameters.  In fact they don't even need to be the same plugin type.
* ScatterGather cons:
  * Must load N copies of the plugin, so for example 5 copies of the NDPluginProcess plugin to do parallel processing with that plugin

* ThreadPool pros:
  * Only need to load a single copy of the plugin.  We can add a new argument to the NDPluginDriver constructor which is the number of threads to create.
  * No need to configure the additional NDPluginScatter and NDPluginGather plugins
* ThreadPool cons:
  * The plugin must be written to be thread-safe against another thread in the same object.  This is not necessarily true today, but should not be hard to achieve.
  * The threads will all be using the same plugin parameters, cannot have some NDArrays processed with one set of parameters and some with another set of parameters.

I am thinking to also implement ThreadPool so there are 2 different ways to achieve parallel processing of a plugin.

MarkRivers, 2017-03-25:
I now have working implementations of both approaches.

- NDPluginScatter and NDPluginGather implement the ScatterGather method.
- NDPluginDriver has been enhanced to support multiple worker threads in a single plugin object.  This requires that the plugins be thread-safe against another thread in the same object.  The following plugins have now been changed so they meet this requirement.  The following plugins are now thread-safe and take a final maxThreads argument in their constructor to allow the use of multiple threads:
  - NDPluginColorConvert
  - NDPluginROI
  - NDPluginStats
  - NDPluginStdArrays
  - NDPluginTransform
- Additional plugins will be made thread-safe as time permits.

These changes are in pull request #261.

MarkRivers, 2017-04-08:
Closed via #261.

closed 2017-04-08 by MarkRivers