areaDetector/ADCore #258: Expose NDPluginStats profiles as NDArrays

alex-soderqvist, 2017-03-02:
I would like the option to hook up additional plugins that performs calculations on the profiles already calculated in NDStats plugin. 

I could already now use the NDDriverStdArrays, but it seems backwards. I would have to instantiate one driver per profile, instead of just setting address in the target plugin.

You could, if you don't mind breaking EPICS DB backwards compatibility, also remove the profile reasons and WF records and lets users instantiate their NDPluginStdArrays for the profiles needed to be exposed.

What is your take on this? Does it seem reasonable? Is it feasible?

MarkRivers, 2017-03-02:
The NDStats plugin could export the profiles as NDArrays using a different asyn "address".  It currently exports the original NDArray (possibly with added attributes) as address 0.  Higher addresses could be used for the time-series data, the X/Y profiles, and the histogram.  It might be nicer to use the asyn "reason" but I believe the interface currently hardcodes NDARRAY_DATA as the drvUser field for NDArray callbacks.

alex-soderqvist, 2017-03-02:
Reasons would be nicer from a user perspective.

Currently I am using the NDPluginROI to do this. But it puts quite some extra responsibility on the user, binning either X or Y to 1dim array, also having in mind scaling. It doesn't abstract very nice and is likely to involve extra support due to bad settings.

MarkRivers, 2018-06-25:
I have changed NDPluginStats so that it exports all of the statistics as an NDArray of dimension [23] each time there is a new input NDArray.  This is now exported to the NDPluginTimeSeries plugin.

The NDPluginTimeSeries plugin already does NDArrayCallbacks on the time-series.  It does 1-D NDArray callbacks with ADDR=0,numSignals-1 for the individual signals and a 2-D NDArray callback with all of the signals with ADDR=numSignals.

closed 2018-06-27 by MarkRivers