areaDetector/ADCore #266: NDFileHDF5 problems

MarkRivers, 2017-04-10:
The NDFileHDF5 plugin on the master branch is now generating errors like the following when saving NDArrays that have NDAttributes:
```

epics> HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5Pdcpl.c line 3225 in H5Pset_fill_value(): not a datatype
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5D.c line 123 in H5Dcreate2(): not a datatype ID
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 265 in H5Acreate2(): not a location
    major: Invalid arguments to routine
    minor: Inappropriate type
  #001: ../H5Gloc.c line 253 in H5G_loc(): invalid object ID
    major: Invalid arguments to routine
    minor: Bad value
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 634 in H5Awrite(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 1655 in H5Aclose(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 265 in H5Acreate2(): not a location
    major: Invalid arguments to routine
    minor: Inappropriate type
  #001: ../H5Gloc.c line 253 in H5G_loc(): invalid object ID
    major: Invalid arguments to routine
    minor: Bad value
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 634 in H5Awrite(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 1655 in H5Aclose(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 265 in H5Acreate2(): not a location
    major: Invalid arguments to routine
    minor: Inappropriate type
  #001: ../H5Gloc.c line 253 in H5G_loc(): invalid object ID
    major: Invalid arguments to routine
    minor: Bad value
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 634 in H5Awrite(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 1655 in H5Aclose(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 265 in H5Acreate2(): not a location
    major: Invalid arguments to routine
    minor: Inappropriate type
  #001: ../H5Gloc.c line 253 in H5G_loc(): invalid object ID
    major: Invalid arguments to routine
    minor: Bad value
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 634 in H5Awrite(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5A.c line 1655 in H5Aclose(): not an attribute
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5D.c line 908 in H5Dset_extent(): not a dataset
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5D.c line 375 in H5Dget_space(): not a dataset
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5Shyper.c line 6888 in H5Sselect_hyperslab(): not a data space
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5Dio.c line 225 in H5Dwrite(): not a dataset
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5S.c line 433 in H5Sclose(): not a dataspace
    major: Invalid arguments to routine
    minor: Inappropriate type
HDF5-DIAG: Error detected in HDF5 (1.10.0-patch1) thread 140588125533952:
  #000: ../H5D.c line 334 in H5Dclose(): not a dataset
    major: Invalid arguments to routine
    minor: Inappropriate type
```
If the NDArray does not contain NDAttributes then there are no error messages.  These errors happen in both "single" and "stream" modes.  They happen on both the master and swmr-fixes branches in ADSupport.

I don't know why these errors just started to occur, I will try to track it down.

MarkRivers, 2017-04-10:
I have isolated this problem.  It seems to be the result of 2 issues:

- simDetectorAttributes.xml defines a PV attribute to 13SIM1:cam1:AcquireTime.  That should be a valid local PV.  However, it appears that it is not connecting.  This is the output of asynReport:
```
epics> asynReport 20 FileHDF1
...
NDAttribute, address=0x7f09e0003430:
  name=AcquireTime
  description=Camera acquire time
  source type=2
  source type string=NDAttrSourceEPICSPV
  source=13SIM1:cam1:AcquireTime
  dataType=NDAttrUndefined
  PVAttribute
    dbrType=DBR_invalid
    chanId=(nil)
    eventId=(nil)
```

- NDFileHDF5 appears to have a problem if a PV attribute is not connected, producing the errors shown above.  If I remove this attribute from simDetectorAttributes.xml, but leave all of the other PV attributes in the file there are no error messages from NDFileHDF5.  I would be surprised if this issue has existed all along, because it seems we would have seen it before. 

I will investigate both issues.

MarkRivers, 2017-04-10:
I found and fixed the first issue with the PV attribute not connecting to a valid PV.  The problem was introduced in R2-6 where the following change was made to prevent multiple subscriptions when a PV disconnected and reconnected.

```
--- a/ADApp/ADSrc/PVAttribute.cpp
+++ b/ADApp/ADSrc/PVAttribute.cpp
@@ -73,6 +73,9 @@ PVAttribute::PVAttribute(const char *pName, const char *pDescription,
     /* Set connection callback on this PV */
     SEVCHK(ca_create_channel(pSource, connectCallbackC, this, 10 ,&this->chanId),
            "ca_create_channel");
+
+    connectedOnce = false;
+
 }

 /** Copy constructor for an EPICS PV attribute
@@ -220,6 +223,10 @@ void PVAttribute::connectCallback(struct connection_handler_args cha)
     epicsMutexLock(this->lock);
     if (chanId && (ca_state(chanId) == cs_conn)) {
         dbfType = ca_field_type(chanId);
+        if (connectedOnce) {
+          goto done;
+        }
+        connectedOnce = true;
         elementCount = ca_element_count(chanId);
```
This change has a race condition, because connectedOnce is set to false **after** the call to ca_create_channel.  What was happening is that for a local PV the callback was happening immediately, before connectedOnce was set to false.  The behavior at that point is unpredictable, since the callback references connectedOnce, which is an uninitialized variable.  I changed the code as follows:
```
--- a/ADApp/ADSrc/PVAttribute.cpp
+++ b/ADApp/ADSrc/PVAttribute.cpp
@@ -45,7 +45,7 @@ static asynUser *pasynUserSelf = NULL;
 PVAttribute::PVAttribute(const char *pName, const char *pDescription,
                          const char *pSource, chtype dbrType)
     : NDAttribute(pName, pDescription, NDAttrSourceEPICSPV, pSource, NDAttrUndefined, 0),
-    dbrType(dbrType), callbackString(0)
+    dbrType(dbrType), callbackString(0), connectedOnce(false)
 {
     static const char *functionName = "PVAttribute";

@@ -73,9 +73,6 @@ PVAttribute::PVAttribute(const char *pName, const char *pDescription,
     /* Set connection callback on this PV */
     SEVCHK(ca_create_channel(pSource, connectCallbackC, this, 10 ,&this->chanId),
            "ca_create_channel");
-
-    connectedOnce = false;
-
```
This fixes the problem, the PV is now connecting correctly.

I'll work on the second issue, NDFileHDF5 not handling a disconnected PV correctly next.

MarkRivers, 2017-04-10:
I fixed the problem with NDFileHDF5 not correctly handling a disconnected PV.  The changes are in commit c77dee0.  Previously the data type of a disconnected PVAttribute was set to -1, which is not a valid HDF5 datatype, and so the HDF5 library was printing error messages.  Now we set the datatype to H5T_NATIVE_FLOAT and the fill value to NAN.  The (undefined) data from the attribute is no longer written to the file, so the fill value is used.

The following is a dump of the PVAttribute dataset in the HDF5 file when the PV is connected.
```
corvette:~/scratch>h5dump -d /entry/instrument/NDAttributes/RingCurrent test_395.h5
HDF5 "test_395.h5" {
DATASET "/entry/instrument/NDAttributes/RingCurrent" {
   DATATYPE  H5T_IEEE_F64LE
   DATASPACE  SIMPLE { ( 100 ) / ( H5S_UNLIMITED ) }
   DATA {
   (0): 99.7196, 99.7196, 99.7196, 99.7196, 99.7196, 99.7196, 99.7196,
   (7): 99.7196, 99.7196, 99.7189, 99.7189, 99.7189, 99.7189, 99.7189,
   (14): 99.7189, 99.7189, 99.7189, 99.7189, 99.7189, 99.7157, 99.7157,
   (21): 99.7157, 99.7157, 99.7157, 99.7157, 99.7157, 99.7157, 99.7157,
   (28): 99.7157, 99.7166, 99.7166, 99.7166, 99.7166, 99.7166, 99.7166,
   (35): 99.7166, 99.7166, 99.7166, 99.7166, 99.7171, 99.7171, 99.7171,
   (42): 99.7171, 99.7171, 99.7171, 99.7171, 99.7171, 99.7171, 99.7166,
   (49): 99.7166, 99.7166, 99.7166, 99.7166, 99.7166, 99.7166, 99.7166,
   (56): 99.7166, 99.7166, 99.7165, 99.7165, 99.7165, 99.7165, 99.7165,
   (63): 99.7165, 99.7165, 99.7165, 99.7165, 99.7165, 99.7167, 99.7167,
   (70): 99.7167, 99.7167, 99.7167, 99.7167, 99.7167, 99.7167, 99.7167,
   (77): 99.7164, 99.7164, 99.7164, 99.7164, 99.7164, 99.7164, 99.7164,
   (84): 99.7164, 99.7164, 99.7164, 99.7165, 99.7165, 99.7165, 99.7165,
   (91): 99.7165, 99.7165, 99.7165, 99.7165, 99.7165, 99.7165, 99.7157,
   (98): 99.7157, 99.7157
   }
   ATTRIBUTE "NDAttrDescription" {
      DATATYPE  H5T_STRING {
         STRSIZE 20;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "Storage ring current"
      }
   }
   ATTRIBUTE "NDAttrName" {
      DATATYPE  H5T_STRING {
         STRSIZE 11;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "RingCurrent"
      }
   }
   ATTRIBUTE "NDAttrSource" {
      DATATYPE  H5T_STRING {
         STRSIZE 13;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "S:SRcurrentAI"
      }
   }
   ATTRIBUTE "NDAttrSourceType" {
      DATATYPE  H5T_STRING {
         STRSIZE 19;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "NDAttrSourceEPICSPV"
      }
   }
}
}
```
The following is a dump of the PVAttribute dataset in the HDF5 file when the PV is  not connected because the PV name was misspelled in the attributes XML file.

```

corvette:~/scratch>h5dump -d /entry/instrument/NDAttributes/RingCurrent test_393.h5
HDF5 "test_393.h5" {
DATASET "/entry/instrument/NDAttributes/RingCurrent" {
   DATATYPE  H5T_IEEE_F32LE
   DATASPACE  SIMPLE { ( 100 ) / ( H5S_UNLIMITED ) }
   DATA {
   (0): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (14): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (27): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (40): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (53): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (66): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (79): nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan,
   (92): nan, nan, nan, nan, nan, nan, nan, nan
   }
   ATTRIBUTE "NDAttrDescription" {
      DATATYPE  H5T_STRING {
         STRSIZE 20;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "Storage ring current"
      }
   }
   ATTRIBUTE "NDAttrName" {
      DATATYPE  H5T_STRING {
         STRSIZE 11;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "RingCurrent"
      }
   }
   ATTRIBUTE "NDAttrSource" {
      DATATYPE  H5T_STRING {
         STRSIZE 14;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "S:SRcurrentAIx"
      }
   }
   ATTRIBUTE "NDAttrSourceType" {
      DATATYPE  H5T_STRING {
         STRSIZE 19;
         STRPAD H5T_STR_NULLTERM;
         CSET H5T_CSET_ASCII;
         CTYPE H5T_C_S1;
      }
      DATASPACE  SCALAR
      DATA {
      (0): "NDAttrSourceEPICSPV"
      }
   }
}
}
```

kmpeters, 2017-04-10:
I see that this fix is in the release notes for R3-0:

https://github.com/areaDetector/ADCore/blob/master/RELEASE.md

Will there also be a R2-6-1?

MarkRivers, 2017-04-10:
I don't currently plan on releasing 2-6-1.  The changes that require updating to R3-0 are already on the master branch.  These changes break backwards compatibility, but only in a minor way.  The NDPluginDriver base class takes different arguments, but that is hidden from the user unless they have custom plugins.  The startup script arguments are different for a few plugins, but that is also hidden if EXAMPLE_commonPlugins.cmd is used.  

I expect to release R3-0 fairly soon.

MarkRivers, 2017-04-11:
Closing because both issues are now fixed.

closed 2017-04-11 by MarkRivers