areaDetector/ADCore #271: libxml2 does not build on vxWorks 5.5

MarkRivers, 2017-05-01:
Now that libxml2 is required for ADCore we need to it build on all supported architectures.  It was not building on vxWorks 5.5.

MarkRivers, 2017-05-01:
This is really an issue for ADSupport, not ADCore.  The issue can now be closed because I have modified https://github.com/areaDetector/ADSupport/blob/master/supportApp/xml2Src/os/vxWorks/config.h and https://github.com/areaDetector/ADSupport/blob/master/supportApp/xml2Src/os/default/libxml/dict.h to allow xml2Src to build successfully on vxWorks 5.5.

The APS Jenkins build server succeeded in building all of the detector libraries that are vxWorks compatible.    Note that it has not been tested on running IOC because I don't have vxWorks 5.5 IOCs available any more.  @ulrikpedersen can someone at Diamond test the simDetector on vxWorks 5.5?

closed 2017-05-01 by MarkRivers

ulrikpedersen, 2017-05-30:
Sorry I just noticed this when clearing out a long list of unread notifications...

We don't normally build anything areaDetector for VxWorks at DLS so we may have to do a bit of work to make that possible. I'll see what we can do.