areaDetector/ADCore #272: Add HTML docs for new NDPluginAttrPlot

ulrikpedersen, 2017-05-03:
Comment from @MarkRivers on the PR #247:

> However, I would like to request that @blaz-kranjc provide html documentation like all of the other plugins, including screen shots of some OPI (edm, CSS-Boy, etc.). Please also add an OPI screen in the source code. I won't hold up R3-0 release for these, but I would like to have them in a future release. Thanks.

blaz-kranjc, 2017-05-03:
@ulrikpedersen @GDYendell  I can do the documentation for the plugin over the weekend. As I no longer have access to the edm screens at DLS I'd like to ask you to create a pull request with the edm files from prod's NDPluginAttrPlot? I'll add documentation to that pull request (or I'll create a pull request with the html and we can add and fix the edm files afterwards).

ulrikpedersen, 2017-05-03:
Hi Blaz! That would certainly be appreciated as above and beyond! 👍 
Just create a PR with what you've got and we'll add edm screens and screenshots.

blaz-kranjc, 2017-05-07:
I've created PR #274

ulrikpedersen, 2017-05-16:
Thanks @blaz-kranjc 👍

commit 3d78c53a (2017-05-16): Merge pull request #274 from blaz-kranjc/attrplot-doc

HTML Documentation and GUI for NDPluginAttrPlot

Fixes #272
closed 2017-05-16 by ulrikpedersen