areaDetector/ADCore #279: Plugin queues full on IOC start up

GDYendell, 2017-06-09:
Having updated to 2-6, we have noticed that on start up the queues for all plugins display as full, but as soon as any frames are received they reset. As far as I can tell, this doesn't have any effect; the frames supposedly in the queue aren't processed if you run an acquisition and you still get the correct total frames through.

Does anyone else see this behaviour?

MarkRivers, 2017-06-09:
I see that on R2-6 and R3-0.  The problem was this commit
```
corvette:ADCore/ADApp/pluginSrc>git show 480dc337764ca6517497b93391103ed659f7ebbb
commit 480dc337764ca6517497b93391103ed659f7ebbb
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Wed Oct 5 10:05:20 2016 -0500

    Fixed error; constructor was calling setIntegerParam for queueSize twice; second call was changed to setIntegerParam for blockingCallbacks

diff --git a/ADApp/pluginSrc/NDPluginDriver.cpp b/ADApp/pluginSrc/NDPluginDriver.cpp
index 6f60ac7..a237577 100644
--- a/ADApp/pluginSrc/NDPluginDriver.cpp
+++ b/ADApp/pluginSrc/NDPluginDriver.cpp
@@ -547,7 +547,7 @@ NDPluginDriver::NDPluginDriver(const char *portName, int queueSize, int blocking
     setIntegerParam(NDPluginDriverArrayAddr, NDArrayAddr);
     setIntegerParam(NDPluginDriverDroppedArrays, 0);
     setIntegerParam(NDPluginDriverQueueSize, queueSize);
-    setIntegerParam(NDPluginDriverQueueFree, queueSize);
+    setIntegerParam(NDPluginDriverBlockingCallbacks, blockingCallbacks);
     unlock();
 }
```
This was done on Oct. 5, 2016 meaning it is in R2-5 and later.

It was a mistake.  I saw two calls to setIntegerParam both passing the value of queueSize and thought it was a duplicate line, not noticing that one is setting QueueSize and the other QueueFree.  That deleted line should be restored.  Can you change that and see if it fixes the problem?

GDYendell, 2017-06-09:
That fixes it for me @MarkRivers .

MarkRivers, 2017-06-09:
I found that not only does NDPluginDriverQueueFree need to be set there in the constructor, but it also needs to be set in NDPluginDriver::createCallbackThreads() because QueueSize can be changed in that function now.   Without this, the value of QueueFree would be wrong right after changing QueueSize, including when QueueSize is restored in autosave.  I have made these changes.

GDYendell, 2017-06-12:
Is the case where QueueSize is changed and QueueFree not updated just cosmetic, or will it actually affect the number of buffers available?

MarkRivers, 2017-06-12:
The problem was just cosmetic.  It was not setting QueueFree the value in the parameter library when the queue was resized, but it was setting it correctly once the next callback occured.  QueueSize is the parameter that matters, QueueFree is just a useful diagnostic.

closed 2017-06-12 by GDYendell